Klipper3d/klipper #4536: The klipper 3d.org contact link is broken.

Thunderrocks33, 2021-07-27:
i made a post here trying to get help with a klipper issue and got the not on github tag and a github automated replay telling me to go to https://www.klipper3d.org/Contact.html   that website/url is broken and gives a 404 not found error.  this is the contact url i found https://www.klipper3d.org/Contact/?h=con

KevinOConnor, 2021-07-27:
Yes - thanks.  We changed the site earlier today and we will need to either fix the links or update the notifications.  As indicated, the correct link is https://www.klipper3d.org/Contact

-Kevin

KevinOConnor, 2021-07-27:
I think this should be resolved now.  The original link of https://www.klipper3d.org/Contact.html should now work again.

-Kevin

closed 2021-08-04 by github-actions[bot]