Klipper3d/klipper #4538: Documentation anchors not working on Safari, macOS and iOS

blalor, 2021-07-28:
After the (fantastic) site update #4512, anchors aren't working correctly when loading the page on Safari on iOS (14.7) and macOS (11.4, Safari 14.1.1).  For example, if I open https://www.klipper3d.org/Config_Reference.html#heater_bed in a new tab, the page doesn't scroll to the anchor, but if I click it from the nav on the right, it does.  Firefox 89.0.2 does not show this behavior.  This means that linking between sections on different pages doesn't work.

damlobster, 2021-07-28:
Thanks @blalor for spotting this, I'll look into how to solve it.

KevinOConnor, 2021-07-28:
Thanks.  Hopefully this is resolved with the merge of #4539 .

-Kevin

blalor, 2021-07-28:
It is!  Thank you!

closed 2021-07-28 by blalor