Klipper3d/klipper #4549: Math Domain Error (Enhanced delta calibration)

joan-aluja, 2021-07-30:
Hello!

I recently went back to Klipper after messing around with RepRapFirmware and Marlin. I successfully installed Klipper (using KIAUH) and got started with the configuration. I successfully calibrated my delta printer using the basic delta calibration guide and started printing the calibrate_size.stl file. And afterwards I started inputting the measurements with the following commands.

DELTA_ANALYZE CENTER_DISTS=73.50,73.95,73.85,73.75,74.15,73.60
DELTA_ANALYZE OUTER_DISTS=73.50,73.55,73.95,73.90,73.45,74.15
DELTA_ANALYZE CENTER_PILLAR_WIDTHS=9.00,9.00,9.05
DELTA_ANALYZE OUTER_PILLAR_WIDTHS=9.00,9.00,9.00,9.00,9.00,9.15
DELTA_ANALYZE SCALE=1.0
DELTA_ANALYZE CALIBRATE=extended

After pressing Enter when sending "DELTA_ANALYZE CALIBRATE=extended" Klipper crashes and an error pops up "Math Domain Error". It seems the error is coming from klipper as it is shown on the log.

I have attached the log below.

[klippy_math-domain-error.log](https://github.com/KevinOConnor/klipper/files/6908894/klippy_math-domain-error.log)

I have checked in Fluidd and I have the latest version installed.

Let me know if you need anything else.

Thanks!

joan-aluja, 2021-08-03:
It seems the crash comes from

`z = -math.sqrt(radius2[0] - x**2 - y**2)` in line 107 mathutil.py

Somehow the value of `radius2[0] - x**2 - y**2` becomes negative and then spits out the error because it can't calculate the sqrt of a negative number.

I've printer another calibrate_size.stl because I had changed some settings in the slicer and I entered the following:

DELTA_ANALYZE CENTER_DISTS=73.90,74.35,74.10,74.00,74.55,74.00
DELTA_ANALYZE OUTER_DISTS=73.75,73.90,74.30,74.20,73.85,74.40
DELTA_ANALYZE CENTER_PILLAR_WIDTHS=9.25,9.20,9.40
DELTA_ANALYZE OUTER_PILLAR_WIDTHS=9.25,9.30,9.30,9.10,9.30,9.30
DELTA_ANALYZE SCALE=1.0
DELTA_ANALYZE CALIBRATE=extended

And it still gives the exact same error and output on the log

I'm going to open a topic on the forum to see if someone comes up with a solution

joan-aluja, 2021-09-07:
Issue seems to still be present. I'm going to try again and get more info as this issue seems to happen with different values

KevinOConnor, 2021-09-27:
Alas, I haven't had a chance to look at this.  Something definitely went wrong during the calibration.

-Kevin

joan-aluja, 2021-10-01:
Hi. I've tried to figure out what is causing the issue but I don't know much Python.
If you need me to check anything on my side please tell me!

closed 2021-11-13 by github-actions[bot]