Klipper3d/klipper #4552: Unknown pin chip name 'probe' - z-axis endstop

Daniel-engman, 2021-08-01:
[klippy.log](https://github.com/KevinOConnor/klipper/files/6912937/klippy.log)
[image]
Printer: Ender 3 max with added BL-touch
Running KlipperWrt snapshot using fluidd on a Creality wifi box
The config file: https://github.com/KevinOConnor/klipper/blob/master/config/printer-creality-ender3max-2021.cfg
**endstop_pin = probe:z_virtual_endstop**
It's seems to be that line that causes the issue, I unfortunately haven't found any solutions that eliminate this problem. 
Any way to correct this error?

Enlightnd, 2021-08-01:
Probably because you are using = instead of :?

Gilabite, 2021-08-01:
Remove the indented gcode macro section safe_z_home from the stepper_z  and bltouch areas (and the indentation from that as well). This isnt a github issue its a user error issue and belongs on discord, discourse.

Daniel-engman, 2021-08-01:
> Remove the indented gcode macro section safe_z_home and bed_mesh from the stepper_z and bltouch areas (and the indentation from that as well). This isnt a github issue its a user error issue and belongs on discord, discourse.

Thank you for your reply, though in the actual code, there is no such indentation. Only in the logs, therefore I don't exactly know how I am able to correct such a mistake.  Added extra newlines just to exaggerate.
[image]

Gilabite, 2021-08-01:
post your config and not the log so I can see it

Daniel-engman, 2021-08-01:
> Probably because you are using = instead of :?

I noticed I wrote the post wrong, old habit of writing declarations using equals. I am using : in the code itself. Thank you for noticing

Daniel-engman, 2021-08-01:
> post your config and not the log so I can see it

[printer.txt](https://github.com/KevinOConnor/klipper/files/6913207/printer.txt)
Saved as a text file this is the config

Gilabite, 2021-08-01:
you have a space in the config before those macros and bltouch section you moved down, there cant be any spaces before that or it will read it as indented

Gilabite, 2021-08-01:
its a single space

Gilabite, 2021-08-01:
you only indent gcode in your macros

Daniel-engman, 2021-08-01:
> you have a space in the config before those macros and bltouch section you moved down, there cant be any spaces or tabs before that or it will read it as indented

Alright I really have gone blind, thank you! That solved the issue, I'm gonna go get my eyes checked now.

closed 2021-08-01 by Daniel-engman