Klipper3d/klipper #4559: [output_pin (pwm)] only outputing 1 or 0 as initial value when 0.X is provided

TheAndi, 2021-08-04:
I specified my LED_pin as follows, and it works great (also dimming with `0.X` values)

> [output_pin LED_pin]
> pin: P2.4
> pwm: True
> value: 0.1  # The value to **initially** set the pin to during MCU configuration.
> shutdown_value: 0.01 #<-- this works
> hardware_pwm: True
> cycle_time: 0.001 #s

I'd like the LEDs to be dim at startup (~`0.1`). If I understand correctly, the `value` parameter should do that. But if I provide `0.1` (or any other `0.X`) the LED startup full brightness (=`1`). If I provide `0` they start with `0`.

The only way I can get the `0.1` from the config is by sending `RESTART` to restart the host. Then the "MCU configuration" (don't know much about that process) presumably gets executed. But shouldn't this not also take place at the initial startup?

Is there something I can do to make it work?

Thanks in advance!

[klippy.log](https://github.com/KevinOConnor/klipper/files/6934439/klippy.log)

KevinOConnor, 2021-08-12:
@1am3r - After a discussion with @TheAndi on Discord, it appears the symptoms of this issue are that lpc176x `gpio_pwm_write()` provides expected results, but a value provided to `gpio_pwm_setup()` on lpc176x does not seem to take effect.  Any ideas?

-Kevin

1am3r, 2021-08-13:
Hmm, could be the initialization of the PWM hardware is wrong or missing a step. I'll take a look.

PR #4605 (2021-08-24): lpc176x: Fix wrong inital value at PWM setup
This fixes a PWM going to full scale output when a initial value != 0 was
given. The output was on 100% until another update occurred.

This change enables the PWM counter before setting the channel values.

This fixes #4559
closed 2021-08-24 by KevinOConnor