Klipper3d/klipper #4560: [FR] Alternative resonance testing methods with the ADXL345

dmbutyugin, 2021-08-05:
For some time, I have been looking into ways of improving the current resonance test offered by Klipper, which runs vibrations of different frequencies and measures the printer response. I had some ringing on the X axis of my Ender 3 with the direct drive extruder, and the input shapers that were recommended as a result of the resonance testing measurements didn't remove the ringing very effectively. 

Overall, the current approach is quite decent and robust, but it may have some shortcomings:

- It is ultimately limited by the stepper motor resolution (microstepping only helps so much) at the high end of the spectrum.
- It may exaggerate the actual impact of the higher-frequency resonances.
- It finds all the resonances of the frame and the motion system, including some that are potentially not excited during the normal printing.
- Puts a lot of stress on the printer.
- It is not possible to properly test the effect of the input shaping on the printer kinematics and resonances, so it is not possible to compare the different input shaper configurations with the accelerometer and choose one that is the best.

So, in this FR I want to discuss and invite to test a different resonance testing procedure. I don't think it is ready for PR, but I did get some promising results on my printer, so it is worth further exploration and testing. If you want to check it out, the code is available in the branch
https://github.com/dmbutyugin/klipper/tree/resonance-test-methods

In order to enable the new test method, replace `[resonance_tester]` section in the `printer.cfg` with the following one:
```
[resonance_tester]
accel_chip: ....  # or separate accel_chip_x and accel_chip_y, this is also supported
method: pulses
max_accel: 10000
probe_points: ... # The usual probing point(s)
```
There are, in principle, more parameters, but they have reasonable defaults (I think), so no need to change them. `max_accel = 10000` works well for Ender 3, I suppose it will work for most printers too. If the printer has a too heavy bed, maybe `max_accel` here will need to be reduced to something like 5000, until the motor does not skip steps.

There is one more thing that must to be configured for the acceleromer(s). Previously, the ADXL345 axes orientation did not really matter. With this test, it matters a bit. Namely, one axis of the accelerometer must align with the tested axis, and you must configure the `axes_map` parameter for the accelerometer ([config reference](https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#adxl345)). To give you an idea, I have mounted the ADXL345 on the hotend in this manner:
[image]
As you can see, the Y axis of the accelerometer is aligned with the X axis of the printer, and the other 2 axes are not aligned with anything. Here's the relevant part of the accelerometer configuration:
```
[adxl345 hotend]
axes_map: -y,-x,z
```
X axis of the printer is mapped to -y axis of the accelerometer (this is important!), the other two axes are set somewhat arbitrarily, cause there is no strict alignment, and those axes don't matter. Note that in case of a CoreXY, Delta, and some Cartesian printers, 2 axes of the accelerometer must be aligned with the axes of the printer, and `axes_map` must correctly map those axes.

After the configuration is done, you can run the resonance test as usual:
```
TEST_RESONACES AXIS=X OUTPUT=raw_data,resonances
TEST_RESONACES AXIS=Y OUTPUT=raw_data,resonances
```
I'd ask you to write and keep the raw_data in case some debugging is needed, but to see resonances and calibrate the input shaper, use the `resonances` files:
```
~/klipper/scripts/graph_accelerometer.py /tmp/resonances_x_*.csv -o /tmp/resonances_x.png
~/klipper/scripts/graph_accelerometer.py /tmp/resonances_y_*.csv -o /tmp/resonances_y.png
```
(and similarly `calibrate_shaper.py` to try the automatic shaper calibration, as per docs).

The test runs some moves, records the acceleration data, computes the frequency response, subtracts the 'ideal' one, which for max_accel=10000 is
[image]
and finds the resonances where there is an excessive response.

If you want to test the input shaping, add `INPUT_SHAPING=1` parameter to the `TEST_RESONANCES` command.

If you test the new method, I'd also ask you to run the old test and post its results to see how the new test is different from the old one. Please note that I've made some changes to the code so the old .csv files are not really compatible with the new scripts; please re-run the old test instead. You can use the same branch, just put instead
```
[resonance_tester]
method: vibrations
```

I'll share my results in a separate post to keep this one at a reasonable limit.

dmbutyugin, 2021-08-05:
In my setup, this is what I get for X and Y axes respectively (the top is the old test, the bottom is the new test from the branch):

|[image]|[image]|
|:--:|:--:|
|[image]|[image]|

As you can see, the results for Y axis are almost the same with minimal differences. The X axis does look somewhat differently. With the new method, the resonance at 60 Hz is much smaller, and so is the resonance at 100-110 Hz. There is also now a strong resonance at ~52 Hz, which was previously either undetected, or got collapsed with the 60 Hz resonance.

I was also able to test how the input shaping affects resonances, and get a few other interesting results:

- Using the 3dB method, I estimated the damping_ratio for X axis to be ~0.038, this indeed showed some improvement in ringing.
- Apparently, the stepper motor current has an impact on the input shaping, at least when TMC Stealthchop mode is used. Low current may lead to less effective input shaping (because, I presume, the stepper motor does not have enough torque to follow the input shaper trajectory precisely).
- Recommended MZV shaper works quite OK, but the best results I got with 2hump_ei 51.9 Hz.

This chart shows the comparison of 2hump_ei @51.9 Hz and damping_ratio = 0.038 at different currents (0.4, 0.5 and 0.6 Amps):
[image]
And I was using 0.4 Amps for some time previously.

And this is the comparison of MZV and 2HUMP_EI input shapers:
[image]

BlackStump, 2021-08-06:
Old to New included all data in zips below

[image]
 [image]
[image]
[image]

[resonances_old.zip](https://github.com/KevinOConnor/klipper/files/6943555/resonances_old.zip)
[resonances_new.zip](https://github.com/KevinOConnor/klipper/files/6943560/resonances_new.zip)

dmbutyugin, 2021-08-06:
For those who are interested how the new algorithm works:

It is actually fairly simple. The new test creates back and forth moves at low velocity, and uses high acceleration to turn around. By default, the suggested max_accel = 10000, which is used throughout the test, and the moves velocity is max_accel / 1000 by default, so it is 10 mm/sec. This means that accelerations and decelerations are very short, 1 msec each. The frequency of the moves slowly changes from 1 Hz to 10 Hz by default. Because the accelerations and decelerations are very short, their spectra have many different frequencies, and so they can excite different resonances. This is how the ideal spectrogram looks like (if the toolhead followed the commanded moves exactly; the frequency response I posted earlier):
[image]

KevinOConnor, 2021-08-06:
Interesting.

Are you running the fft on both the real and simulated inputs and then subtracting the fft results?  Or, are you subtracting the raw acceleration and then feeding that into the fft?  As a very high-level idea, I was thinking of doing the latter.

I did notice that the new graphs show higher absolute peaks (eg, 1e5 vs 1e4).  Is that because these are "density" graphs and the energy is now more focused on a few points, or is the data actually showing greater peaks?

> Apparently, the stepper motor current has an impact on the input shaping, at least when TMC Stealthchop mode is used. Low current may lead to less effective input shaping (because, I presume, the stepper motor does not have enough torque to follow the input shaper trajectory precisely).

FWIW, this is one of the things I'm hoping the angle sensors can provide some insight on.

Cheers,
-Kevin

dmbutyugin, 2021-08-06:
@KevinOConnor 

> Interesting.
> 
> Are you running the fft on both the real and simulated inputs and then subtracting the fft results? Or, are you subtracting the raw acceleration and then feeding that into the fft? As a very high-level idea, I was thinking of doing the latter.

I run FFT on both and subtract the FFT results. I think this approach is more robust. But I don't have a strong opinion about the latter, and it would be interesting to  subtract the expected and actual accelerations too and look at the results. Though that would require reliable and accurate timings of the actual measurements. If I am not mistaken, right now the available timings are a bit approximate.

Anyway, I think there are 2 parts of it: first, come up with a reasonable alternative test, and then decide on its exact implementation. On a few previous attempts I wasn't successful at designing a robust test that provides useful results: either they were not providing stable results, or the signal-to-noise ratio was not satisfactory. The current proposal is more of a proof-of-a-concept. At least, the implementation is reasonable enough to run the tests, and the tests will show if it actually produces the useful signals. Then we can decide how to best integrate it. Also, perhaps by then the PR #4555 and the subsequent work will be integrated already.

> I did notice that the new graphs show higher absolute peaks (eg, 1e5 vs 1e4). Is that because these are "density" graphs and the energy is now more focused on a few points, or is the data actually showing greater peaks?

I'm afraid one cannot directly compare the magnitude of the peaks. It makes sense to compare the shape and the relative values within the chart, but not the values between the methods. First, in the current Klipper method we increase the acceleration with the frequency (we also divide the response, but not by frequency^2), and the new method uses the maximum acceleration for the whole duration. Also, the baseline frequency is very different between the methods, and so the spectral decomposition may produce a different energy density.

> FWIW, this is one of the things I'm hoping the angle sensors can provide some insight on.

Perhaps. It's a bit more rare hardware, but it would be great to see the support in Klipper. At least, right now I was able to see the effect of a different motor current this way through the resonance test. I suspect that many people don't realize that the current could have such an impact - it is usually a question, is the current enough that the stepper doesn't skip steps? At least, I often looked at it this way previously.

dmbutyugin, 2021-08-06:
@BlackStump Thanks for running the tests! At least in your case, both methods seem to agree for the most part. The resonance peaks are at a little bit different frequencies, and the higher frequency end of the charts in the new method have smaller values.

clborrello, 2021-08-13:
@dmbutyugin I completed the tests and generated the plots for X and Y at both 10k and 15k acceleration. I'm using an ender 3 pro and my square corner velocity for all tests was 8. You'll notice in the included spectrograms that I have a constant resonance at 130Hz. I'm guessing that's from my hotend cooling fan. This frequency is super amplified using the new pulses method, but not in the old vibrations method. 
[pulse resonances.zip](https://github.com/KevinOConnor/klipper/files/6984224/pulse.resonances.zip)
Pulses | Vibrations
------------ | -------------
[image] | [image]
[image] | [image]
[image] | [image] 
[image] | [image]

dmbutyugin, 2021-08-13:
@clborrello Thanks! Admittedly, loud unbalanced fans are a problem, even for the old algorithm. It can make it only somewhat better because it tends to have stronger peaks. So, ideally, any loud unbalanced fans should be off during resonance testing. I think I need to add this to the resonance testing docs, even for the current algorithm, because I saw people having problems with them.

That being said, I pushed a commit to my testing branch that can make the fan problem better to some degree. Feel free to test it with your current hardware and let me know how it goes then. From the data you've posted I see that it is still not perfect, especially with the new algorithm, but it does make the situation better.

clborrello, 2021-08-16:
@dmbutyugin I stopped the noisy fan and re-ran the test (before your last commit). I did 2 tests on the X axis: 

a=10k, SCV=5 | a=10k, SCV=8
------------ | -------------
[image] | [image]
[image] | [image]

Seems to be working well. I've noticed that when doing multiple resonance tests back to back, the results are slightly different each time. I'm not sure if that's mechanical, electrical, or computational. You can see that these resonance plots look a bit different than my last batch above.

dewi-ny-je, 2021-08-17:
But a fan isn't a source of vibrations which is uncorrelated with toolhead movement? shouldn't it be somehow filtered out as baseline noise?

KoenVanduffel, 2021-08-17:
I ran my set of tests comparing both methods. I had originally tuned input shaper using the printed model and was getting way lower frequencies than when measured with the adxl, especially on the X-axis. I am using a Creality CR-6 SE bed slinger with a direct drive setup (the direct drive setup as posted in my GitHub with the mount directly on the strain gauge).
[image]
Standard X-axis measurement

[image]
Pulses method X-axis

[image]
Standard Y-axis method

[image]
Pulses method Y-axis

What is striking is that the resonance frequencies are lower and resonances are stronger even though subjectively this is a much more gentle test. 
Resulting frequencies from the different methods:
                                  X                               Y
Pinted test:               42                             45
Original ADXL          79.4 (2hump_ei)        46.8 (2hump_ei)
Pulses method         52.2 (MZV)                44 (2hump_ei)

Visually the resulting prints using the original measurements were really poor so I reverted to the printed test (where I was using MZV for both X and Y). The prints using the Pulses  method look probably better than the ones tuned with the print but the difference is really small.

FYI I turned the hotend fan off during these tests to eliminate its influence. Rerunning with the fan on gave however no differences in my case (I am using a Sunon 3010 fan).

clborrello, 2021-08-18:
> shouldn't it be somehow filtered out as baseline noise?

@dmbutyugin could the first part of the test measure the baseline resonance before the pulses begin? The initial frequency response could then be subtracted from the final result.

rahsoft, 2021-08-18:
Can You include into main klipper updates to test it out (will be more people / tests) ?..

rahsoft, 2021-09-02:
And what about new "pulses" testing mode ?

matthewlloyd, 2021-09-20:
Super interesting to read about this work. Since this essentially an impulse response measurement and estimation problem, it maybe worth looking into the many standard techniques available, since they may require less measurement time, be more robust to noise, and be well grounded mathematically. In particular, for stimuli, PRBS (pseudo random binary sequences) comes to mind. May or may not be a useful direction, but thought I'd mention it.

dmbutyugin, 2021-09-28:
@rahsoft 
> Can You include into main klipper updates to test it out (will be more people / tests) ?..

I think that will have to wait until the next Klipper release is released.

> And what about new "pulses" testing mode ?

Sorry, I didn't understand the question. Anyway, this is another method that is available in [my test branch](https://github.com/dmbutyugin/klipper/tree/resonance-test-methods) that may or may not provide some better or more robust results (more testing needed, the initial results seem promising).

@matthewlloyd 
Thanks for the pointer, I'll definitely check out. If you have any further references for more established methods, please feel free to share them.

dmbutyugin, 2021-09-28:
@KevinOConnor 
Separately, I have updated the branch to the latest Klipper code. More importantly, the simulated adxl345 now uses Klipper internal system to dump trapq moves, and I also made the new method to subtract the raw acceleration readings from adxl345 and the expected acceleration per your suggestion. I'm actually very pleased with the results. The adxl345 readings align very well with the actual commanded moves, so it seems that the code which synchronizes the adxl345 readings with the print time works quite well. Interestingly, when the accelerometer is connected to the RPi (v3), there seems to be no lag between the data from the adxl345 vs the print time as reported by the trapq print_time:
[image]
This chart shows the actual adxl345 readings minus the acceleration computed from trapq moves. So, when you see a dip in acceleration, this means that there was a short large acceleration (when we subtract it, we get a negative value) and when there is a peak - there was a short large deceleration. You can see here that these short bursts of acceleration are not experienced by the toolhead right away, instead they start oscillations that spread out acceleration over a longer period of time and ultimately accelerate it to the expected speed.

However, when the adxl345 is connected to the MCU board (SKR v1.4 Turbo in my case), there seems to be a tiny amount of lag (~4-4.3 ms, pretty consistent across the whole duration of the test) of the data from the adxl345 vs the print time as reported by the trapq print_time:

[image]

Fortunately, this lag does not appear to be affecting the final results after the FFT processing. Still, I wonder where does it come from? It seems that perhaps the clock synchronization between the MCU and the RPi has some consistent lag, perhaps due to USB transmission delays or something similar? This lag may not exist when adxl345 is connected to the RPi because then both 'RPi MCU' and the host code use the same clock.

For those who wish to re-test, you can check out the aforementioned branch, and all you need to do is add the following to the config:
```
[resonance_tester]
method: vibrations  # new method
# other parameters should be the same, like accelerometers and probe_points
```

If ADXL345 axes X,Y,Z are not the same as printer axes X,Y,Z, do not forget to update the `axes_map` parameter, e.g.
```
[adxl345]
axes_map: -y,-x,z  # make sure to put the correct mapping, paying attention to the direction
```
Note that sign must match direction the direction of the accelerometer and the printer axis: if the positive direction of the accelerometer axis is in the same direction as a positive direction of a printer axis, the sign should be positive (no sign needed), if they are in the opposite direction - you should put '-'.

So far, I didn't figure out the automatic way of detecting incorrect axes mapping or inverted sign, but, on the other hand, I didn't think about it too much yet. So, maybe this would be possible in the next iteration. For now, if you mix up the sign, you will get this (vs the correct sign):
|[image]|[image]|
|:--:|:--:|

(blue line is correct sign in the axes_map, orange line - inverted sign).

So, you'll get some bogus peaks on the PSD spectrogram and generally the results will be distorted.

To run the test, use
```
TEST_RESONANCES AXIS=X OUTPUT=raw_data,resonances
```
Note that it generates 3 files: resonances_x, raw_data_x with the raw accelerometer readings, and raw_data_adjusted_x with the expected acceleration subtracted from adxl345 readings. You should use either resonances*.csv file or raw_data_adjusted*.csv files with the `calibrate_shaper.py` (also new version from my branch) to get the input shaper recommendations. And similarly for Y axis.

KevinOConnor, 2021-09-29:
Interesting.  I'm not surprised there would be some static time offsets, but 4-5ms seems very high.  Can you provide the Klipper log from the test?

> when the accelerometer is connected to the RPi (v3), there seems to be no lag between the data from the adxl345 vs the print time as reported by the trapq print_time

I'm not sure I understand - that chart does show two artificial peaks on  the y axis.  Or, are you saying that those peaks are of shorter duration?

> This lag may not exist when adxl345 is connected to the RPi because then both 'RPi MCU' and the host code use the same clock.

It's actually the other way around.  The "print_time" is the main mcu clock converted to seconds.  If the adxl345 and the steppers are on `[mcu]` then the time should be more accurate - only one clock is in use in that case.  When using multiple mcus, the host code attempts to predict the secondary mcus clock relative to the primary mcu clock.  So, if the adxl345 is attached to the rpi mcu, then the host code will attempt to correlate the clocks and that could lead to some offsets.  The klippy.log file has a bunch of stats on this that can be analyzed (`graphstats -f`).

> Fortunately, this lag does not appear to be affecting the final results after the FFT processing.

That's good.

> So far, I didn't figure out the automatic way of detecting incorrect axes mapping or inverted sign

FWIW, I had a rough idea of performing calibration via "integration".  Roughly: 1) wait for one second to gather freefall acceleration, 2) move the toolhead from position A to position B, 3) remove freefall acceleration from the sensor and then integrate the result, 4) the calibrated accelerometer `scale` is the measured integrated results divided by the actual distance moved.  I was also thinking this could also compensate for slight mounting angles.  Ultimately, it was easier for me to just align the sensor and configure it correctly though, so I didn't get a chance to see if this idea works.

-Kevin

dmbutyugin, 2021-09-29:
> Interesting. I'm not surprised there would be some static time offsets, but 4-5ms seems very high. Can you provide the Klipper log from the test?

I'll rerun both tests tomorrow and attach the logs. 
 
> I'm not sure I understand - that chart does show two artificial peaks on the y axis. Or, are you saying that those peaks are of shorter duration?

The test runs at 10K accel and small velocity 10 mm/sec (so the acceleration time is 2 msec: from +10mm/sec to -10mm/sec and vice versa at 10000 mm/sec^2). The scale on the charts as I generated was different for the tests on X and Y axis (to better demonstrate the offset on the X axis test). In both cases, the charts depict adxl345_accel(print_time) - toolhead_accel(print_time). Those peaks you see are when toolhead_accel is already at 10K (you can think of it as Klipper already commanding the toolhead to move at 10K accel), but the accelerometer does not pick up that acceleration yet (because the belts and frame start to deform first). Then we subtract ~0 and 10K acceleration and get large, but short peaks (2 msec each).

> It's actually the other way around. The "print_time" is the main mcu clock converted to seconds. If the adxl345 and the steppers are on `[mcu]` then the time should be more accurate - only one clock [...]

Hmm, it sounds like the log will indeed be very helpful. 

> FWIW, I had a rough idea of performing calibration via "integration". Roughly: 1) wait for one second to gather freefall acceleration, 2) move the toolhead from position A to position B, 3) remove [...]

Yes, I have a feeling that some automatic calibration (or at least some error reporting) can be implemented.

KevinOConnor, 2021-09-29:
> The scale on the charts as I generated was different for the tests on X and Y axis (to better demonstrate the offset on the X axis test).

Okay.  I'm still not sure I understand.  Are you pointing out that there is a brief "flat" section between the "dip" and "wave" on X while Y goes directly from "dip" to "wave"?

> I'll rerun both tests tomorrow and attach the logs.

If you rerun the tests, I'd be curious if you could simultaneously run `./scripts/motan/data_logger.py` and also provide the logs from that.

Cheers,
-Kevin

dmbutyugin, 2021-09-29:
@KevinOConnor 
> Okay. I'm still not sure I understand. Are you pointing out that there is a brief "flat" section between the "dip" and "wave" on X while Y goes directly from "dip" to "wave"?

Yes, that. Here's another view from the tests I did today, highlighting the lag:

[image]

Here are the raw files stacked together (I modified the graph_accelerometer.py script to support that). raw_data_x/y are the raw readings from accelerometers, raw_data_adjusted_x/y with the expected toolhead acceleration subtracted. X accelerometer is connected to the MCU, Y accelerometer to the RPi. As you can see, there is still some lag on Y accelerometer connected to the RPi, but it seems to be ~2x smaller than on X accelerometer. I also checked if there could be any play in the belts, etc. - but there isn't any I can manually detect.

> If you rerun the tests, I'd be curious if you could simultaneously run `./scripts/motan/data_logger.py` and also provide the logs from that.

OK, here are the full results with logs:

[test_data.zip](https://github.com/Klipper3d/klipper/files/7255005/test_data.zip)

dmbutyugin, 2021-09-29:
BTW, I also find it interesting to observe how the input shaping works:

|[image]|[image]|
|:--:|:--:|
|2HUMP_EI input shaper| MZV input shaper|

Here the blue line shows adxl345 acceleration - expected acceleration when no input shaper is enabled, the red line shows adxl345 acceleration - expected acceleration when input shaper is enabled, and the green line - true adxl345 acceleration when input shaper is enabled.

KevinOConnor, 2021-09-30:
Okay, I understand.  That's certainly odd.  I don't know why that would be.  It certainly looks like a coding issue with clock synchronization, but it's not immediately clear to me what would cause that.  In particular, your main mcu has ~1 millisecond (and as little as 158us) of round-trip-time anyway.  I'll try to review the code to see if I can find anything.

Would you be able to swap accelerometers and/or swap wiring to see if the problem moves with axis, accelerometer, or mcu?  Also, could the `spi_speed` setting on `[adxl345 hotend]` be contributing?

FYI, separately, it's possible to generate similar graphs using the motan tool - for example:
`../../scripts/motan/motan_graph.py test_x_motion -s 41.903 -d .05 -g '[["trapq(toolhead,x_accel)","adxl345(hotend,x)"],["deviation(adxl345(hotend,x),trapq(toolhead,x_accel))"]]'`
[image]

`../../scripts/motan/motan_graph.py test_x_motion -s 41.903 -d .05 -g '[["trapq(toolhead,x_accel)","adxl345(hotend,x)"],["deviation(adxl345(hotend,x),trapq(toolhead,x_accel))"]]'`
[image]

-Kevin

dmbutyugin, 2021-09-30:
> Okay, I understand. That's certainly odd. I don't know why that would be. It certainly looks like a coding issue with clock synchronization, but it's not immediately clear to me what would cause [...]

OK, this is very bizarre. First, I tried to increase the `spi_speed` parameter (comment it out, to be more precise, with the default being 5 MHz). This didn't help really (today 5MHz vs yesterday 2MHz):

[image]

But then I reconnected the Y accelerometer from RPi (where I didn't have problems) to the MCU (where the problems existed), ran the test, and the problem did not appear:
[image]

This is really strange, if the problem was with some timing calculation, or MCU <-> RPi communication, then I'd expect that when I connected the Y accelerometer to the MCU, the lag issue should appear, but this isn't the case. So, the problem must lie elsewhere.

Now, both accelerometers are the same china boards purchased at the same time from the same shop. I am not aware of any differences between them. Besides, it is not reasonable to expect any buffering happening on either of the boards besides its FIFO that Klipper is already accounting for. Both X and Y stepper motors are connected to the same MCU via the same TMC5160 stepper motor drivers. The stepper motors on X and Y axes are different, but I wouldn't expect the problem to come from there.

Unfortunately, it wasn't possible for me to connect the X axis accelerometer to the RPi to test this setup due to insufficient wiring length. Maybe I'll try to make an extender, or maybe I could just glue the X accelerometer to the bed and test as Y axis accelerometer to see if the problem is with the adxl345 board (but this is quite unlikely).

So, in the end, it might be a mechanical issue with the X axis. I'll need to invest more time into studying that possibility. At least, I don't have any other good ideas on how to reasonably explain that behavior.

> FYI, separately, it's possible to generate similar graphs using the motan tool 

That's good to know! It would also be interesting to add a possibility to plot the toolhead movement after input shaping. Anyway, for the resonance test data processing I still need the code to calculate that difference from within Klipper.

KevinOConnor, 2021-10-01:
> But then I reconnected the Y accelerometer from RPi (where I didn't have problems) to the MCU (where the problems existed), ran the test, and the problem did not appear

Interesting.

One thing I noted is that there isn't that much movement during the direction change.
`./scripts/motan/motan_graph.py test_x_motion -s 41.908 -d .01 -g '[["trapq(toolhead,x_accel)","adxl345(hotend,x)"],["trapq(toolhead,x_velocity)"],["stepq(stepper_x,0)"]]'`
[image]
Looking at that graph, the accelerometer starts to respond after around 7 steps in the reverse direction.  Since you're using 64 microsteps, that's not a lot of distance for the stepper motor - they aren't all that accurate in the microstepping range.  It might not be related though.

> It would also be interesting to add a possibility to plot the toolhead movement after input shaping.

You can - `stepq(stepper_x)` would show the raw steps sent to the mcu (post input_shaper), while `trapq(toolhead,x)` would show the requested position (pre input_shaper).

Running `./scripts/motan/motan_graph.py -l` shows the available transforms.  Admittedly, it's not particularly user-friendly.

-Kevin

wlhlm, 2021-10-02:
In case you are still looking for data, I ran the new resonance testing method on a Voron V2 (flying gantry coreXY) with the accelerometer attached directly to the hotend heater block below the nozzle.

|**Method**|**X-Axis**|**Y-Axis**|
|---|---|---|
|vibrations|[image]|[image]|
|pulses|[image]|[image]|
|pulses (stealthchop)|[image]|[image]|

[wlhlm-resonance-measurements.zip](https://github.com/Klipper3d/klipper/files/7271818/wlhlm-resonance-measurements.zip) (tar.xz in zip for better compression)

After running the first round of resonance measurements with the new pulses method I noticed the peak at 100Hz that doesn't show up with the vibrations method. My guess was that maybe this came from the Z-steppers holding position in spreadcycle mode, so I ran another test with StealthChop enabled (TMC2226 aka 2209). Looking at the results, the 100Hz resonances seem to be (mostly) diminished on the X-axis measurement, but are still visible on the Y-axis, so I'm not sure if spreadcycle was the culprit here.

matthewlloyd, 2021-10-03:
> Thanks for the pointer, I'll definitely check out. If you have any further references for more established methods, please feel free to share them.

A good reference is "System Identification: A Frequency Domain Approach, Second Edition" by Pintelon and Schoukens. Chapter 5, "Design of Excitation Signals", is particularly useful, and directly compares the efficiency of various test signals, including the swept sine wave chirp which Klipper currently uses (at least a close approximation thereof), and pseudo-random binary sequences.

A good excitation sequence has various properties:

* It should be as short as possible to measure the frequency response with a desired accuracy. This can be measured by the "time factor", which tells you how long the signal needs to run to guarantee a sufficiently high SNR (relative to the noise background) at each frequency of interest. Broadband excitation signals perform better than frequency sweeps on this measure, because they efficiently excite many frequencies at the same time.

* It should maximize the amount of spectral energy injected into the system, given the amplitude headroom available (in this case, related to max acceleration). One way to measure this is the "crest factor", which is essentially the ratio of the peak amplitude to the RMS amplitude. Sine wave sweeps are already quite good (~1.45) while random sequences are better (~1).

* It should have an even distribution of energy across frequencies, with no "dead spots". Frequency sweeps do well here almost by definition. Genuinely random sequences (e.g. white noise) fair poorly because by chance you may have some frequency with too little power. Artificially constructed pseudorandom sequences are designed to avoid this issue and make guarantees about the distribution of energy.

With a well designed PRBS, the measurement time could probably be reduced substantially from the current 90 seconds. PRBS are easy to implement using linear-feedback shift registers. Another advantage is the FFT of the measurements can be used directly as the response, since the frequency spectrum is flat.

dmbutyugin, 2021-10-05:
@KevinOConnor 
> > It would also be interesting to add a possibility to plot the toolhead movement after input shaping.
> 
> You can - `stepq(stepper_x)` would show the raw steps sent to the mcu (post input_shaper), while `trapq(toolhead,x)` would show the requested position (pre input_shaper).

True, but for Cartesian kinematics. For CoreXY/Delta/etc., stepper motion is only related to the toolhead motion.

Separately,
> Fortunately, this lag does not appear to be affecting the final results after the FFT processing.

It seems I was wrong. I added a possibility for the simulated adxl345 to output the measurements also with a lag, configured it as
```
[adxl345_simulated]
time_offset: 0.002

[resonance_tester]
accel_chip: adxl345_simulated
method: pulses
probe_points:
  100, 100, 20
```
and then ran `TEST_RESONANCES AXIS=X OUTPUT=raw_data` in the Klipper batch mode (for 1 msec, 2 msec and 4msec offsets). This is one pulse to demonstrate how the results look like:

[image]
[image]
[image]

So adxl345_simulated basically outputs the acceleration exactly as it should be, but with a configured time offset.

Then the spectrum of the signals looks like:
[image]

`raw_data_x_1msec_delay.csv`, `raw_data_x_2msec_delay.csv` and `raw_data_x_4msec_delay.csv` have exactly the same spectrum - the intensity of the vibrations at different frequencies injected into the system. `raw_data_adjusted_x_1msec_delay.csv`, `raw_data_adjusted_x_2msec_delay.csv` and `raw_data_adjusted_x_4msec_delay.csv` should ideally show a flat line near X axis (so PSD ~= 0), with maybe some high-frequency harmonics due to very rapid signal changes that we don't care about. Unfortunately, this is not the case: they show quite some low-frequency response below 100 Hz that we definitely do care about.

I need to look more into this, but as it is, it appears that a simple subtraction of the measured and expected acceleration data is not necessarily the best choice to find the resonances. It may (or may not) be the reason of quite a few cases I saw so far when the high-frequency response with the new method does not go to 0 at high-frequency end of the spectrum (unlike the vibrations method), but instead stays flat at some non-zero value (well below the resonance peak, but still).

@matthewlloyd 
Thanks for the pointer, I'll try to get my hands on it.

> Another advantage is the FFT of the measurements can be used directly as the response, since the frequency spectrum is flat.

I wanted to follow-up on this point a bit. As a general matter, we don't really need the frequency response by itself - we don't have a complicated motion control algorithm that requires obtaining the frequency response of the printer system. Instead, we need it to find resonances of the system, and ideally we want to give some input to the shaper auto-tuning algorithm.

The shaper tuning algorithm today is very simple. It just looks at the frequency response and tries to reduce it with an input shaper (to the level ~ peak_value / 20). However, the frequency response of an ideal oscillator looks like 
[image]
Before the resonance, it has the response >= 1 and after the resonances - decaying to 0. We don't really need or want to compensate for the response >= 1 before the resonance. So, we need to somehow identify in the shaper tuning algorithm that response >= 1 before the resonance is OK. However, we cannot just subtract 1 from the response because there could be subsequent resonances after the first one (e.g. because some other part oscillates). And because of response decay after the first resonance, the amplitude of that second resonance can get smaller than 1 and so the shaper tuning algorithm can completely miss it. As a result, some more advanced data processing algorithm, more advanced algorithm to tune the input shaper, or some other tricks are necessary to process the frequency response. That's the gist of a problem we have with the frequency response. The current trick with the vibrations test is the non-flat frequency input and ignoring the problem to some extent :) If you have any ideas how to improve the situation with frequency response processing (or any literature worth reading on the topic) - they are very welcome. In the pulses test the trick was supposed to be a bit different - subtract the measured and expected signal - but it seems it is not so easy as well.

KevinOConnor, 2021-10-06:
> True, but for Cartesian kinematics. For CoreXY/Delta/etc., stepper motion is only related to the toolhead motion.

FYI, if you want the nominal toolhead stepper motion on corexy it is possible to use `kin(stepper_x)`.  Admittedly, I only implemented this for corexy and cartesian, but it would not be difficult to extend to delta (see `scripts/motan/analyzers.py:GenKinematicPosition` for details).  It's also possible to go from raw corexy steppers back to an X position via `corexy(x,stepper_x,stepper_y)`.  See `./scripts/motan/motan_graph.py -l` for available transforms.

> I need to look more into this, but as it is, it appears that a simple subtraction of the measured and expected acceleration data is not necessarily the best choice to find the resonances.

Okay, thanks.  I guess, at a high-level, if there is a static offset between nominal impulse and actual impulse then the FFT would report that as a frequency.  FWIW, in your previous tests there didn't seem to be any actual overlap between nominal acceleration and measured acceleration, so there doesn't seem to be a gain in subtracting anyway.

-Kevin

KevinOConnor, 2021-10-06:
Just as a random thought - perhaps it would make more sense to integrate the accelerometer to a position and then subtract the nominal position from that.

From your earlier log - here's what that might look like for your hotend x data:
[image]
And your bed y data:
[image]

Looking at that a little closer - could the difference in delay between your y accelerometer response and x response actually be due to a difference in resonant frequency between the two axes?

FYI, I generated the above graphs using the `work-angle-20210722` branch - that has the experimental accelerometer integration code - which is, admittedly a bit "dodgy".  The first graph was generated with: `scripts/motan/motan_graph.py test_x_motion -s 41.600 -d .6 -g '[["trapq(toolhead,x_accel)","adxl345(hotend,x)"],["deviation(trapq(toolhead,x),integral(integral(adxl345(hotend,x),trapq(toolhead,x_velocity)),trapq(toolhead,x)))"]]'` and the second with `scripts/motan/motan_graph.py test_y_motion -s 43.1 -d 1 -g '[["trapq(toolhead,y_accel)","adxl345(bed,y)"],["deviation(integral(integral(adxl345(bed,y),trapq(toolhead,y_velocity)),trapq(toolhead,y)),trapq(toolhead,y))"]]'`.  I'm sure those commands were obvious though.  :smile:

-Kevin

KevinOConnor, 2021-10-06:
Separately, if we look at the Y response to an impulse:
[image]
we can see there is an initial response to the impulse and then the system settles into a wave with a 20ms frequency.

I suspect that 20ms frequency is due to the stepper motor full step distance.  With a 40mm rotation distance and 200 full steps per rotation, there is a full step every 0.200mm (40mm / 200).  At 10mm/s that is a full step every 0.200s (0.200mm / 10mm/s).

Here's a zoomed in graph along with the stepper microstep position:
[image]
Which indicates a correlation between stepper driver microstep position and acceleration.  So, I suspect systemic differences in stepper motor torque at microstep positions is introducing this acceleration - not a resonance from the initial impulse.

The X carriage shows a similar "full step frequency".

I suspect it may be challenging to separate the stepper motor induced frequencies from the impulse induced frequencies.

-Kevin

First graph is via: `scripts/motan/motan_graph.py test_y_motion -s 43.15 -d .267 -g '[["trapq(toolhead,y_accel)","adxl345(bed,y)"]]'`
Second graph is using latest work-angle-20210722 code and: `scripts/motan/motan_graph.py test_y_motion -s 43.2 -d .22 -g '[["trapq(toolhead,y_accel)","adxl345(bed,y)","step_phase(tmc5160 stepper_y,microstep)?color=grey&alpha=.3"]]'`

dmbutyugin, 2021-10-06:
Separately, I made a few tests to try to understand the nature of that 4msec delay in the acceleration propagation. I tried to see if the stepper momentum or the test speed has any impact on it. So, I ran the same pulse test with the default configuration accel = 10K mm/sec^2, velocity = 10 mm/sec, and stepper X current = 0.6A, and then 3 other combinations of current = 0.3 A (2x smaller) and velocity = 20 mm/sec (and accel = 20K mm/sec^2 to keep the acceleration time the same). It does not look like the delay is affected by either of this:
[image]

With velocity = 10 mm/sec the amplitude is smaller (because acceleration is smaller), but the rest looks more or less the same. I'd expect if there was any correlation with the test velocity (e.g. due to drag) or due to available stepper torque, then once the respective parameter is halved, then the delay decreases by some ~2x, but it doesn't seem to be happening.

Separately, the responses do look different (and not just modulo scaling):
[image]
And specgrams side by side
|[image]|[image]|
|:--:|:--:|
|[image]|[image]|

It seems that reducing stepper current increases the full stepping noise (50 Hz at 10mm/sec and 100 Hz at 20mm/sec), probably because the driver does not have enough current to compensate for stepper inaccuracies of movement and 'put it back on track', it is also possible that the stepper just works weirdly when under-powered and torque becomes generally less linear near 0.

For the reference, here's the frequency response with the current resonance test in Klipper with different stepper currents:
[image]

Note there is also 100 Hz resonance here. But this time, it definitely has a different origin (than the full stepping noise):
|[image]|[image]|
|:--:|:--:|

My hypothesis regarding this is as follows. If there is really some play somewhere in the system, if we move the toolhead in one direction, then in the opposite direction, if we time these movements right, we can excite the resonance a bit from that play, even if the system is otherwise relatively stiff at that frequency. However, 1 / 100 Hz gives 0.01 sec, or 5 msec for the movement in one direction. This is more than 4msec estimated previously, but it could be that the resonance increases the amplitude of a play and increases the time it takes to 'consume' the play. So, the 100 Hz peak in the vibrations test might be indicative of that hypothetical play.

KevinOConnor, 2021-10-07:
Interesting.  Thanks.

In case more data helps, I've run your code on my Voron Zero that has the angle sensors.

X vibrations:
[resonances_x_20211007_031221.csv.gz](https://github.com/Klipper3d/klipper/files/7298608/resonances_x_20211007_031221.csv.gz)
[raw_data_x_20211007_031221.csv.gz](https://github.com/Klipper3d/klipper/files/7298609/raw_data_x_20211007_031221.csv.gz)
[vibxlog.json.gz](https://github.com/Klipper3d/klipper/files/7298611/vibxlog.json.gz)
[vibxlog.index.gz](https://github.com/Klipper3d/klipper/files/7298613/vibxlog.index.gz)
Y vibrations:
[resonances_y_20211007_031516.csv.gz](https://github.com/Klipper3d/klipper/files/7298616/resonances_y_20211007_031516.csv.gz)
[raw_data_y_20211007_031516.csv.gz](https://github.com/Klipper3d/klipper/files/7298617/raw_data_y_20211007_031516.csv.gz)
[vibylog.json.gz](https://github.com/Klipper3d/klipper/files/7298618/vibylog.json.gz)
[vibylog.index.gz](https://github.com/Klipper3d/klipper/files/7298619/vibylog.index.gz)

Pulses X:
[pulsexlog.json.gz](https://github.com/Klipper3d/klipper/files/7298622/pulsexlog.json.gz)
[pulsexlog.index.gz](https://github.com/Klipper3d/klipper/files/7298623/pulsexlog.index.gz)
[resonances_x_20211007_032143.csv.gz](https://github.com/Klipper3d/klipper/files/7298624/resonances_x_20211007_032143.csv.gz)
[raw_data_x_20211007_032143.csv.gz](https://github.com/Klipper3d/klipper/files/7298625/raw_data_x_20211007_032143.csv.gz)
[raw_data_adjusted_x_20211007_032143.csv.gz](https://github.com/Klipper3d/klipper/files/7298626/raw_data_adjusted_x_20211007_032143.csv.gz)
Pulses Y:
[pulseylog.json.gz](https://github.com/Klipper3d/klipper/files/7298627/pulseylog.json.gz)
[pulseylog.index.gz](https://github.com/Klipper3d/klipper/files/7298628/pulseylog.index.gz)
[resonances_y_20211007_032539.csv.gz](https://github.com/Klipper3d/klipper/files/7298629/resonances_y_20211007_032539.csv.gz)
[raw_data_y_20211007_032539.csv.gz](https://github.com/Klipper3d/klipper/files/7298630/raw_data_y_20211007_032539.csv.gz)
[raw_data_adjusted_y_20211007_032539.csv.gz](https://github.com/Klipper3d/klipper/files/7298631/raw_data_adjusted_y_20211007_032539.csv.gz)

-Kevin

KevinOConnor, 2021-10-07:
FWIW, I also see an offset between pulse and acceleration on the Zero.

Interestingly, I also see an offset on the angle sensor response.

Here's a random example on X:
[image]
And a random example on Y:
[image]

So, I think that may indicate that either I've goofed on the timing of both the accelerometer and the angle sensors, or there is a microstepping torque lag in the stepper motor driver.

I'll have to take a closer look.

-Kevin

BlackStump, 2021-10-07:
would using full steps instead of microsteps help?

dmbutyugin, 2021-10-07:
@BlackStump I'm not sure I understand your suggestion. Do you mean, only for testing? In general, full steps have very low resolution (0.2 mm with GT2 and 20 teeth pulley). This will likely be insufficient for the test, because 0.2 mm / 10 mm/sec = 0.02 sec, far worse than the small deviations of 0.002 - 0.004 sec we observe today. And vibrations from the stepping and non-linearity of the motion could impact the quality of the resonance measurements, I suppose.

dmbutyugin, 2021-10-11:
Thanks, Kevin. The results we've collected give a lot of food for thought. At the very least:

- The commanded and the actual motion could be out of sync by a few milliseconds. It is not a mistake of measurements (at least, there are occurrences where it is not a result of measurement errors). Therefore, the test data processing must be robust to such offsets.
- The acceleration (its magnitude and offset) can depend on the test velocity and microstepping (and interpolation). E.g. with 16x microstepping and 0.2 mm full step and 10mm/sec speed, the stepper driver will notice the acceleration in 0.2 / 16 / 10 = 1.25 msec, and then depending on other factors may generate the acceleration only around 10 mm / 1.25 msec or maybe 2x that, so 20K accel may be not achievable in such test. It goes back to the design of the test and its results processing.
- Stepper noise from full steps may be noticeable during the test, so the test design and the results processing should account for that.
- Subtracting the measured and expected acceleration may not provide good results, not unless special care is taken to account for offsets and potentially smaller and smoother acceleration. It is likely better to explore other options (e.g. CSD, cross spectral density, among other things).

So, I'll be scratching my head in the next days and see if I can think of a good way around that, and read some literature what I can come across.

> Just for completeness, I also ran a "pulses" test on x with 1.5Amp, 128 microsteps, interpolation off, spreadcycle:

Thanks! If you have a chance, I would also ask you to run the default vibrations test at these settings, since the increased current appears to decrease the lag. I'm primarily interested if any of the resonances will change their frequency compared to the earlier setup. In other words, I'm curious to see if any of the resonances can be attributed to just the lag. I expect that it should be that way, but with your setup that changes the lag with the increased current, it is easier to demonstrate this effect.

rahsoft, 2021-11-03:
When if at all it will be in main Klipper ?..

dmbutyugin, 2021-12-10:
It is too early to close this. Though I do not have any concrete timeline, I've been doing some experiments in the past weeks.

sniperlucian, 2022-01-10:
@KevinOConnor 

can you elaborate what kind of angle sensor your use on your voron?  couldnt find any source here or with google.

Sineos, 2022-01-10:
I'd suppose https://github.com/KevinOConnor/nema14tle

KevinOConnor, 2022-01-10:
I've also started a topic on the angle sensor code at https://klipper.discourse.group/t/experiment-with-hall-effect-angle-sensors/1693

-Kevin

dmbutyugin, 2022-02-16:
Let's keep it open for now.

seancmonahan, 2022-03-05:
@matthewlloyd
> A good reference is "System Identification: A Frequency Domain Approach, Second Edition" by Pintelon and Schoukens. Chapter 5, "Design of Excitation Signals", is particularly useful, and directly [...]

Is the PDF version sold by Wiley decent, or should I try to track down a physical copy? (The PDF sample chapter seems to be a scan of a physical copy...)

I've got three ADXL345s hooked up, and I'm looking to play around with Klipper's resonance tuning, including this FR and the [accel-calibrate](https://github.com/dmbutyugin/klipper/tree/accel-calibrate) branch also in @dmbutyugin's fork.  Cheers!

closed 2022-04-17 by github-actions[bot]

dowjames, 2022-06-06:
This bot is really annoying

mz-design, 2022-08-08:
Hi, any news in thread? Just wondering this interesting research stopped..

BarsMonster, 2023-01-12:
Very interesting investigation!
My 5cents here is that stepper motor movement cannot physically immediately cause start of hotend acceleration. 

All mechanical  actions propagate approximately at the speed of sound in material. 
I would expect that in glass-fiber reinforced GT2 belt speed of sound is around 4.5km/sec. 
So for 250mm belt path one could expect ~0.055ms of delay, and up to 0.2ms in large printers (like Ender 5 plus).

Dmytry, 2023-05-11:
Some observation regarding stepper motor movements: the torque of the motor is zero when the rotor is at the commanded position. It is approximately equal to $` T_{max} sin{n pi \over 2N } `$ .  where N is number of microsteps per step, n is the difference between actual rotor position and commanded rotor position in microsteps. 

I think what would make the most sense is to obtain the response spectrum by dividing the PSD of real accelerations by PSD of commanded accelerations. That gives you response spectrum, in terms of accelerations, rather than position. I think it needs to be further scaled by $` f^{-1} `$  to obtain response spectrum in terms of position (a position vibration of magnitude x and frequency f results in ~ $` f^{2} `$ acceleration; as far as I can tell your PSD already produces squared magnitude, so you only need to scale by $` f^{-1} `$)

The advantage of doing this would be to allow comparisons to be made directly between graphs that were obtained using different motion patterns.

Another possible advantage is improving usefulness of the graphs (the squiggly lines themselves) shown to the users. As it is I think while the primary use of shaper recommendations is indeed selecting a shaper, on discord I see that a lot of people are using the graph to tune their mechanical system, often with confusion relating to the units.

> Before the resonance, it has the response >= 1 and after the resonances - decaying to 0. We don't really need or want to compensate for the response >= 1 before the resonance. So, we need to somehow [...]

It could be beneficial to express the current tricks in terms of the frequency response, to improve understanding of the existing solution. (It could be that <1 resonance peaks at above resonance frequency are not a significant practical problem.)

My understanding is that for frequencies above the resonance frequency of motor-toolhead system, there is a 90 to 180 degree phase shift, meaning that those frequencies probably show up as artifacts, even if they were faithfully reproduced at the ideal amplitude.

So to summarize, my understanding is that the weighting of squared magnitude should be $` f^{-1} `$ , and the shaping should probably bring the response close to a low pass filter with cut off near resonance frequency.