Klipper3d/klipper #4561: Update documentation for missed step

MattByczko, 2021-08-06:
Following the instructions at the following link appear to be missing steps.
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

#4409 mentions you need to perform `make menuconfig` then `make clean` then `make` for it to work on first-time setup.

It would be real handy to include it in the guide.

KevinOConnor, 2021-08-06:
A `make clean` is not necessary on a new install.  It's only necessary on an update, which is documented in the FAQ entry on updating.

-Kevin

crof2003, 2021-08-08:
Just adding that I installed Klipper for the first time this past week and also ran into the same issue.

Google pulled up [issues 3682](https://github.com/KevinOConnor/klipper/issues/3682) - which referenced doing a `make clean` to resolve an error with `field ‘i2c_config’ has incomplete type`

This allowed me to complete the Klipper build on my RaspberryPi 3B.

everettperiman, 2021-08-23:
I had the same issues on a Raspi 3B+ the make clean step was the fix for my field ‘i2c_config’ has incomplete type error.

CandyAngel, 2021-09-01:
I'm just setting up Klipper on a second Pi/printer (did the first on 2021-08-28) and `make clean` was required to get the firmware to build for me, as I got this when running `make` for the first time.

```
  Compiling out/src/i2ccmds.o
src/i2ccmds.c:14:23: error: field ‘i2c_config’ has incomplete type
     struct i2c_config i2c_config;
                       ^
src/i2ccmds.c: In function ‘command_config_i2c’:
src/i2ccmds.c:23:23: warning: implicit declaration of function ‘i2c_setup’ [-Wimplicit-function-declaration]
     i2c->i2c_config = i2c_setup(args[1], args[2], addr);
                       ^
src/i2ccmds.c: In function ‘command_i2c_write’:
src/i2ccmds.c:35:5: warning: implicit declaration of function ‘i2c_write’ [-Wimplicit-function-declaration]
     i2c_write(i2c->i2c_config, data_len, data);
     ^
src/i2ccmds.c: In function ‘command_i2c_read’:
src/i2ccmds.c:48:5: warning: implicit declaration of function ‘i2c_read’ [-Wimplicit-function-declaration]
     i2c_read(i2c->i2c_config, reg_len, reg, data_len, data);
     ^
make: *** [Makefile:64: out/src/i2ccmds.o] Error 1
```

closed 2021-09-23 by github-actions[bot]

everettperiman, 2021-09-23:
Was this closed with a fix in the documentation?

MattByczko, 2021-09-27:
@everettperiman it does not look like it:
https://github.com/Klipper3d/klipper/pull/4415 - closed/not merged.
https://github.com/Klipper3d/klipper/pull/4683 - closed/not merged (duplicate).

I feel it is worthy for an addition considering there has been three instances where it is necessary within a short time period. Those are just the commenters of this issue.

redfox05, 2021-11-26:
Just to note, I also came across this during a fresh install on octopi Raspberry Pi 3, which crashed during the `make` command from #4409. Running the clean instructions above allowed things to continue as normal.