Klipper3d/klipper #4563: New thermistor and my heater cartridge, getting negative reading on hotend. 100k HT-NTC B3950

makelele, 2021-08-06:
Hi, 
thank you so much for your hard work on Klipper. 
My issue is I have changed my thermistor and my heater cartridge and now I am having negative reading on my Sovol SV01. 
When I run PID_CALIBRATE I get an` Internal error on command:"PID_CALIBRATE"`
[klippy.log](https://github.com/KevinOConnor/klipper/files/6947120/klippy.log)

My thermistor is 100k HT-NTC B3950 and I've been trying to add it but I can't. I'm getting -75 (negative) on my nozzle temperature. 
I attached the log.

KevinOConnor, 2021-08-06:
Oh - the "internal error" is an area where Klipper could be improved.  However, the root cause of your issue is a "heater not heating at expected rate" - you should reach out to the Klipper community if you need help with that.

-Kevin

makelele, 2021-08-06:
ok, thank you.

ddacunha, 2021-08-20:
@makelele, I am in a similar situation. Btw, I get the same reading if I don't plug the thermistor. So maybe I have a loose connection, or I am not plugging it on the correct pin on the board.

Did you manage to find the cause of your issue?

ddacunha, 2021-08-20:
I confirm that my issue was due to faulty connection.

strayr, 2021-09-04:
I'm just gonna bring this up for @ddacunha and at @makelele https://github.com/KevinOConnor/klipper/issues/4054
These thermistors appear to be more accurate at printing temperatures using EPCOS 100K B57560G104F but you really need to test these.

@KevinOConnor does this want to be in the docs somewhere? I'll write it up and do a PR if it would be helpful. I've spent most of today finding ways to calibrate sensors and proper second guessing myself over sensor types.

closed 2021-10-17 by github-actions[bot]