Klipper3d/klipper #4576: Z Offset with BLTouch does not work when printing

jayzebra, 2021-08-11:
Hi,
I'm having this same issue on an Ender 3 Pro with a bltouch v3: [https://github.com/KevinOConnor/klipper/issues/3334](url)
By now, I calibrated everything according to the instructions several times and I can't get the nozzle to stay close to the bed. I even tried several values from 0 to 2 for the z_offset . The nozzle just hoovers around 1 to 2mm above the bed with the exact same issue the original poster described.

```
# This file contains common pin mappings for the 2018 Creality
# Ender 3. To use this config, the firmware should be compiled for the
# AVR atmega1284p.

# Note, a number of Melzi boards are shipped with a bootloader that
# requires the following command to flash the board:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
# If the above command does not work and "make flash" does not work
# then one may need to flash a bootloader to the board - see the
# Klipper docs/Bootloaders.md file for more information.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 30

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 30

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
position_max: 250
position_min: -1

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
microsteps: 16
rotation_distance: 31.917
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
min_temp: 0
max_temp: 255
#pressure_advance: 0.14

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
serial: /dev/ttyUSB0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0

[bed_screws]
screw1: 50,50
screw2: 200,50
screw3: 50,200
screw4: 200,200

# BLTouch probe. One may define this section (instead of a probe
# section) to enable a BLTouch probe. (Note! This bltouch module may
# not work correctly with some BLTouch "clones"!) A virtual
# "probe:z_virtual_endstop" pin is also created (see the "probe"
# section above for the details).
[bltouch]
sensor_pin: ^PB7
control_pin: PB5
#pin_up_touch_mode_reports_triggered: False
#pin_up_reports_not_triggered: False
#sensor_pin: ^PC4
#   Pin connected to the BLTouch sensor pin. This parameter must be
#   provided.
#control_pin: PA4
#   Pin connected to the BLTouch control pin. This parameter must be
#   provided.
#pin_move_time: 0.675
#   The amount of time (in seconds) to wait for the BLTouch pin to
#   move up or down. The default is 0.675 seconds.
#pin_up_reports_not_triggered: True
#   Set if the BLTouch consistently reports the probe in a "not
#   triggered" state after a successful "pin_up" command. This should
#   be True for a genuine BLTouch; some BLTouch clones may require
#   False.  The default is True.
#pin_up_touch_mode_reports_triggered: True
#   Set if the BLTouch consistently reports a "triggered" state after
#   the commands "pin_up" followed by "touch_mode". This should be
#   True for a genuine BLTouch; some BLTouch clones may require
#   False. The default is True.

# Measure the x and y distance between the tip of your nozzle and 
# your BLTouch tip
# x : + is to the right and - is to the left
# y : + is toward the back and - is toward the front
x_offset: -42 # 41mm left of the nozzle
y_offset: -9 # 13mm forward of the nozzle
#z_offset: 0.400

speed: 5.0
samples: 2
sample_retract_dist: 8.0
set_output_mode: 5V
probe_with_touch_mode: True 

[safe_z_home]
home_xy_position: 158,125 # Change coordinates to the center of your print bed
speed: 20
#z_hop: 10                 # Move up 10mm
z_hop_speed: 5

[bed_mesh]
speed: 80
horizontal_move_z: 5
probe_count: 5,5
mesh_min: 50,30
mesh_max: 179,200

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 23.239
#*# pid_ki = 1.201
#*# pid_kd = 112.419
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 74.536
#*# pid_ki = 1.656
#*# pid_kd = 838.534
#*#
#*# [bltouch]
#*# z_offset =  0.101

```

bugilt, 2021-08-11:
Make sure the gantry isn't running into something. Sometimes the old z sensor can be higher than the nozzle.

jayzebra, 2021-08-11:
Hi,

The old Z sensor was just barely attached to the axis but it was enough to disturb z_offset.
I have been looking in forums, discord, etc to no avail and this was driving me crazy!

Thank you so much for your help!

closed 2021-08-11 by jayzebra

bugilt, 2021-08-11:
It's a commons issue when people switch to probe based sensing. No problem, glad I could help.