Klipper3d/klipper #4581: TMC2209 (and possibly other trinamic driver) en_spreadcycle issue

mayukhnath, 2021-08-12:
My understanding is that "en_spreadcycle" bit of the GCONF register is determined by the "stealthchop_threshold" field in the config fie. It seems that if that is set to 0, then the en_spreadcycle bit bit also be set to 0; if it is set to a non-zero  value, then the en_spreadcycle bit is set to 1. 

My guess is that the intention is the only enable stealthchop when the hybrid threshold has a non-zero value. However, I am getting different behavior. I'm fairly certain this is related to the fact that I have my "spread" pins in logical high states, but in the off-chance that is a bug in the tmc configuration code, I am creating this issue.

It seems that to enable spreadcycle, the "en_spreadcycle" bit has to be high, but klipper seems to attempt to do the opposite. Below are my configuration  settings for my x an y axes.  I intentionally set a stealthchop threshold for x and did not for y. 

[tmc2209 stepper_x]
uart_pin = auxboard:PC5
interpolate = True
run_current = 0.8
sense_resistor = 0.110
stealthchop_threshold = 10
uart_address = 0
diag_pin = auxboard:PD4

[tmc2209 stepper_y]
uart_pin = auxboard:PC5
interpolate = True
run_current = 0.8
sense_resistor = 0.110
uart_address = 1
diag_pin = auxboard:PD7

Below are the dumps for the two axes:
If we look at "DRV_STATUS" we will see that that stealthchop (indicated by bit 30 being high) is enabled for stepper_y and is not for stepper_x
DUMP_TMC stepper_x
========== Write-only registers ==========
IHOLD_IRUN: 00081919 IHOLD=25 IRUN=25 IHOLDDELAY=8
TPWMTHRS:   000003aa TPWMTHRS=938
TPOWERDOWN: 00000014 TPOWERDOWN=20
SGTHRS:     00000000
========== Queried registers ========== 
GCONF:      000001c4 en_spreadCycle=1 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
GSTAT:      00000000
IFCNT:      00000063 IFCNT=99
OTP_READ:   0000000f OTP_FCLKTRIM=15
IOIN:       21000141 ENN=1 PDN_UART=1 SPREAD_EN=1 VERSION=0x21
FACTORY_CONF: 0000000f FCLKTRIM=15
TSTEP:      000fffff TSTEP=1048575
MSCNT:      000001f8 MSCNT=504
MSCURACT:   0108000a CUR_A=10 CUR_B=-248
CHOPCONF:   14030050 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
DRV_STATUS: 80190000 CS_ACTUAL=25 stst=1
PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
PWM_SCALE:  0000001d PWM_SCALE_SUM=29
PWM_AUTO:   000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14
SG_RESULT:  00000002 SG_RESULT=2

DUMP_TMC stepper_y
========== Write-only registers ==========
IHOLD_IRUN: 00081919 IHOLD=25 IRUN=25 IHOLDDELAY=8
TPOWERDOWN: 00000014 TPOWERDOWN=20
SGTHRS:     00000000
========== Queried registers ==========
GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
GSTAT:      00000000
IFCNT:      00000060 IFCNT=96
OTP_READ:   0000000e OTP_FCLKTRIM=14
IOIN:       21000145 ENN=1 MS1=1 PDN_UART=1 SPREAD_EN=1 VERSION=0x21
FACTORY_CONF: 0000000e FCLKTRIM=14
TSTEP:      000fffff TSTEP=1048575
MSCNT:      00000038 MSCNT=56
MSCURACT:   00e80053 CUR_A=83 CUR_B=232
CHOPCONF:   14030050 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
DRV_STATUS: c0190000 CS_ACTUAL=25 stealth=1 stst=1
PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
PWM_SCALE:  0000001d PWM_SCALE_SUM=29
PWM_AUTO:   000e0024 PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14
SG_RESULT:  00000002 SG_RESULT=2

To verify that I don't have my axes flipped, I have set stealthchop thresholds for all of them and none of them to make sure this behavior is consistent

.

This isn't a problem for me since I have no intention of using stealthchop, so I just enter a random stealthchop_threshold value to enable spreadcycle. However, it is still unexpected behaviour (I think).

The log may or may not be useful to you, but lines 2972-3016 and 2881-2896 are relevant 
[klippy (3).log](https://github.com/KevinOConnor/klipper/files/6972012/klippy.3.log)

attached is also a pdf of the custom board I am using (it is the tmc2226, which is identical to the tmc2209 besides the package)
[klipper_aux.pdf](https://github.com/KevinOConnor/klipper/files/6972088/klipper_aux.pdf)

jakep82, 2021-08-12:
> This isn't a problem for me since I have no intention of using stealthchop, so I just enter a random stealthchop_threshold value to enable spreadcycle. However, it is still unexpected behaviour (I think).

I think you misunderstand how this value works.  stealthchop_threshold is the speed at which the driver switches to spreadcycle.  If you're using a non-zero value, then you're using stealthchop at any speed below the value you've configured.  If you only want to use spreadcycle, you need to set stealthchop_threshold to 0.

mayukhnath, 2021-08-12:
I agree, that is the expected behavior, but I am getting the opposite. So, I am finding that stealthchop is enabler if I set the threshold to 0, and spread cycle is enabled of I set a non zero value. So since in my case I have stealthchop disabled and a threshold set, I think the threshold is ignored. What’s interesting is that if I set the value to zero, then stealthchop is enabled but I figured with a threshold of 0, it would still operate in spreadcycle mode, but instead I get reduced torque (which i am guessing indicates it is in stealthchop mode). Anyways, I hope this clarifies the situation.

mayukhnath, 2021-08-12:
> I think you misunderstand how this value works. stealthchop_threshold is the speed at which the driver switches to spreadcycle. If you're using a non-zero value, then you're using stealthchop at any [...]

I agree, that is the expected behavior, but I am getting the opposite. So, I am finding that stealthchop is enabler if I set the threshold to 0, and spread cycle is enabled of I set a non zero value. So since in my case I have stealthchop disabled and a threshold set, I think the threshold is ignored. What’s interesting is that if I set the value to zero, then stealthchop is enabled but I figured with a threshold of 0, it would still operate in spreadcycle mode, but instead I get reduced torque (which i am guessing indicates it is in stealthchop mode). Anyways, I hope this clarifies the situation.

KevinOConnor, 2021-08-13:
The Klipper code is not expecting the SPREAD pin to be high.  The tmc2209 behavior is quite strange in that mode - see #3766 .  If it's possible, you may want to investigate grounding that pin on the drivers.

Separately, on at least one set of stepsticks, the SPREAD pin was routed to an exposed pin.  On some boards, that pin could be shared with the UART pin - which led to complete havoc - as spreadcycle was then being randomly enabled/disabled during prints.

-Kevin

closed 2021-09-25 by github-actions[bot]