Klipper3d/klipper #4584: Stuttering during probing

avispa4291, 2021-08-14:
Since the last update multiple users on the Voron discord have reported a stuttering of z movement during probing sequences. In most cases it does not affect overall probing but in some cases it has caused probing sequence to fail.

JMTV250, 2021-08-14:
I have also noticed a strange z hop/stutter after completing probing of bed on Z after probing bed during Klicky z offset

jclloyd27, 2021-08-15:
I have also noted that is during extensive probing eg. Bed Mesh Calibration.
Per the OP I haven't noticed it creating outlier readings or causing a functional impact

Tallen27, 2021-08-15:
I have seen this happen with Klicky probe during Voron 2.4 Quad Gantry Level and Bed Mesh Calibration

closed 2021-08-18 by github-actions[bot]