Klipper3d/klipper #4594: stepper_z1 is not activated

MagnusThome, 2021-08-19:
Going from Marlin to Klipper I've run into one single problem. All else seem to work fine. I've googled this a lot but haven't been able to come up with what's wrong. I'm leaning towards some oddity with the TMC2208 in combination with the Trigorilla board but am lost...

Description:

Swapping pin definitions between stepper_z and stepper_z1 like shown below verifies that both drivers and steppers work. stepper_z works with both stepper pin definitions and both stepper motors. But stepper_z1 never works regardless pin definitions.

With this stepper_z runs the **right** stepper just fine but the left stepper is not moving
```
[stepper_z]
step_pin: PL3
dir_pin: PL1
enable_pin: !PK0
microsteps: 16
rotation_distance: 8
endstop_pin: ^!PD3
position_endstop: 0.0
position_max: 205
homing_speed: 5.0

[stepper_z1]
step_pin: PC1
dir_pin: PC3
enable_pin: !PC7
microsteps: 16
rotation_distance: 8
endstop_pin: ^!PL6

```

With this stepper_z runs the **left** stepper just fine but right stepper is not moving
```
[stepper_z]
step_pin: PC1
dir_pin: PC3
enable_pin: !PC7
microsteps: 16
rotation_distance: 8
endstop_pin: ^!PL6
position_endstop: 0.0
position_max: 205
homing_speed: 5.0

[stepper_z1]
step_pin: PL3
dir_pin: PL1
enable_pin: !PK0
microsteps: 16
rotation_distance: 8
endstop_pin: ^!PD3

```

The printer is an Anycubic Mega S with Trigorilla 1.4 modified with:
- TMC2208
- BLTouch (just wired up but not enabled yet in Klipper)

Have had this printer running a few years with Marlin. Even swapped back to Marlin to double check when running into this problem when trying to get Klipper working. Still works fine with Marlin.

[klippy.log](https://github.com/KevinOConnor/klipper/files/7012364/klippy.log)

MagnusThome, 2021-08-19:
To me it looks like a problem in Klipper since it refuses to control stuff under [stepper_z1] even when the hardware defined there is proven to work fine when under [stepper_z]. It's a very very basic hardware setup and very basic printer.cfg IMHO.

KevinOConnor, 2021-08-19:
Bug or not, someone needs to diagnose the root cause.  If you're actively working on diagnosing it then you can keep this issue open to share your progress.  If you're looking for help diagnosing it, you can reach out to the Klipper community for assistance.

-Kevin

MagnusThome, 2021-08-19:
More than happy to spend time on trying to find the issue but not super qualified to dive into the source code by myself. Very open to a team effort. But I absolutely 100% understand if it is outside the scope here.

closed 2021-08-21 by github-actions[bot]