Klipper3d/klipper #4597: Bed Mesh No Longer Functioning Properly

UrufuShinjiro, 2021-08-19:
Probing a mesh or Screws_Tilt_Calculate are no longer accurate. Several others are having this issue. With no mesh I can print a 150x150 first layer test perfectly, but screw tilt says I need to make adjustments and bed mesh shows a ski slope. This began in the last 10 days or so. See attachments

[image]
[klippy (1).log](https://github.com/KevinOConnor/klipper/files/7017457/klippy.1.log)

LobodPl, 2021-08-19:
I can also confirm inaccurate bed_mesh... Or bed mesh was don correctly but offset executing is wrong.
[image]
This is mesh i have made.
[image]
This is print without using bed mesh that confirms bed shape. (sides are higher than center)
[image]
And this is print after loading bed mesh.

demitrix, 2021-08-19:
I also experienced a similar happening with z_tilt and and was exacerbated by bed mesh. Triple Z CoreXY setup, z_tilt would align the three points within 0.005mm. Immediately moving the probe back to the left probe point and running probe accuracy would return the same z height. Moving nozzle to same location and to Z0.2 would result in a ~0.6 gap. If a bed_mesh was calibrated afterwards, and Z set to 0.2 the nozzle would collide with the right side of the bed, and be about 1mm above the bed on the left side.

JohnOCFII, 2021-08-19:
I recently moved from RRF to Klipper on my Railcore 300ZL which has a 3 lead screw leveling system.  The bed mesh results are _doing something_ but isn't smoothing out my first layer as I would have expected.  I'm happy to provide additional data, but not sure what would be helpful. 

I'm running Klipper version v0.9.1-694-g128a2f8c 

[image]
[image]

RomRider, 2021-08-20:
For me, bed mesh seems accurate, but during the print it seems that it's not used properly (offset not/wrongly applied?)

KevinOConnor, 2021-08-21:
It will be necessary for someone to determine root cause of the issue.  You can use the Klipper Discourse and/or Discord servers if you are looking for help tracking it down.

-Kevin

matthew-humphrey, 2021-08-21:
I am experiencing what I think it this issue. I have only recently converted my printer to Klipper, but mesh leveling was working using the stock firmware. I actually thought there was some mechanical problem causing it, so I have been spending all kinds of time trying to track that down.

Talha909, 2021-08-21:
I have very strange issue. I perfume the screw tile and bed mesh. My screw tile show the result accurate whenever I did the bed mesh after the screw tilt there is difference. There is problem in bed mesh. I did the screw tilt. It was good on screw title. But when I did bed mesh there is difference. 
klippy (1).log
printer.txt
.
First 
$ SCREWS_TILT_CALCULATE
// probe at 10.000,60.000 is z=0.409750 16:08:38
// probe at 10.000,270.000 is z=0.406625 16:09:22

Second

Bed Mesh
// probe at 10.000,60.000 is z=0.380375 16:14:31
// probe at 10.000,270.000 is z=0.398500 16:15:43

`16:08:05
$ G28
16:08:06
$ SCREWS_TILT_CALCULATE
16:08:35
// probe at 10.000,60.000 is z=0.409750 16:08:38
// probe at 10.000,60.000 is z=0.410375 16:08:40
// probe at 10.000,60.000 is z=0.408500 16:08:50
// probe at 260.000,60.000 is z=0.409750 16:08:52
// probe at 260.000,60.000 is z=0.411625 16:08:54
// probe at 260.000,60.000 is z=0.417250 16:09:03
// probe at 260.000,270.000 is z=0.412875 16:09:05
// probe at 260.000,270.000 is z=0.414125 16:09:08
// probe at 260.000,270.000 is z=0.416625 16:09:18
// probe at 10.000,270.000 is z=0.406625 16:09:20
// probe at 10.000,270.000 is z=0.407250 16:09:22
// probe at 10.000,270.000 is z=0.406625 16:09:22
// 01:20 means 1 full turn and 20 minutes, CW=clockwise, CCW=counter-clockwise
16:09:22
// front left screw (base) : x=10.0, y=60.0, z=0.40954 16:09:22
// front right screw : x=260.0, y=60.0, z=0.41287 : adjust CCW 00:00 16:09:22
// rear right screw : x=260.0, y=270.0, z=0.41454 : adjust CCW 00:0016:09:22
// read left screw : x=10.0, y=270.0, z=0.40683 : adjust CW 00:0016:09:25 16:14:29

Bed Mesh
// probe at 10.000,60.000 is z=0.380375 16:14:31
// probe at 10.000,60.000 is z=0.381625 16:14:34
// probe at 10.000,60.000 is z=0.382875 16:14:41
// probe at 135.000,60.000 is z=0.431625 16:14:43
// probe at 135.000,60.000 is z=0.434750 16:14:45
// probe at 135.000,60.000 is z=0.436000 16:14:53
// probe at 260.000,60.000 is z=0.403500 16:14:55
// probe at 260.000,60.000 is z=0.408500 16:14:57
// probe at 260.000,60.000 is z=0.401625 16:15:04
// probe at 260.000,165.000 is z=0.450375 16:15:06
// probe at 260.000,165.000 is z=0.452875 16:15:09
// probe at 260.000,165.000 is z=0.443500 16:15:16
// probe at 135.000,165.000 is z=0.497250 16:15:18
// probe at 135.000,165.000 is z=0.499750 16:15:21
// probe at 135.000,165.000 is z=0.498500 16:15:28
// probe at 10.000,165.000 is z=0.405375 16:15:30
// probe at 10.000,165.000 is z=0.405375 16:15:33
// probe at 10.000,165.000 is z=0.406625 16:15:38
// probe at 10.000,270.000 is z=0.398500 16:15:43
// probe at 10.000,270.000 is z=0.394750 16:15:44
// probe at 10.000,270.000 is z=0.397875 16:15:51
// probe at 135.000,270.000 is z=0.462250 16:15:53
// probe at 135.000,270.000 is z=0.479750 16:15:56
// probe at 135.000,270.000 is z=0.476625 16:16:03
// probe at 260.000,270.000 is z=0.409125 16:16:05
// probe at 260.000,270.000 is z=0.412250 16:16:07
// probe at 260.000,270.000 is z=0.418500 16:16:07
// Mesh Bed Leveling Complete
16:16:07
// Bed Mesh state has been saved to profile [default]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
16:16:08
// Bed Mesh state has been saved to profile [p1]
// for the current session. The SAVE_CONFIG command will
// update the printer config file and restart the printer.
16:17:22

Talha909, 2021-08-21:
Can someone perfume the screw tilt first and then bed mesh and match the result. I am doing these two test. there are always difference in the result of bed mesh and screw tilt.

closed 2021-08-23 by github-actions[bot]

marcandre83, 2021-08-24:
Does anyone know why this issue has been closed? For me the problem still persists.

mattborkin, 2022-02-18:
I'm having this EXACT same problem. Here's how I have been explaining it on the other forums: 
[image]
Everything I can think of is checking out properly and calibrating correctly - however, when the nozzle moves to the left half of the bed, my left z axis (using dual-z axis) is engaging and actively lowering that half just enough so the first layer won’t stick.I’m totally clueless now. 

My setup is:

Ender 5 Plus with an E3D Direct Drive Hemera with V6 hotend, BigTreeTech SKR 1.4 Turbo board with TMC2208 steppers modified for independent, dual Z-Axis control, running Klipper (Fluidd) on a Raspberry Pi 4 with a Raspberry Pi 7” touchscreen running KlipperScreen. I also have an ADXL345 accelerometer attached for Pressure Advance tuning. I have a textured PEI bed installed. 

I use z tilt adjust to level the left and right z axis. Then I use screws told adjust to tram the bed, then I do a 6x6 mesh probe and save it for default reference. I have also squared the entire machine and checked all wheels and even swapped the BLTouch probe. 

Even with the above steps, every time I print something, when the nozzle is on the left half of the bed, that left z stepper will actively drop the bed very slightly and then move back up when the nozzle gets to the right side of the bed. 

This is causing the right side of the bed to lay down layers perfectly as usual, but everything on the left side is not adhering as a result of the bed actively dropping.

It’s as if the mesh is telling the bed to lower on that side when it’s doesn’t need to. I have been swapped motors and rotated the bed. The results are the same - upper left part of bed drops during printing on the left side. Then raises as it goes back over to the right side. It’s like it’s being TOLD to do that for some reason. 

I have attached a link to my code as well as attached a pic of the mesh. 

I’m lost. 

https://pastebin.com/VTtz9Wwd

matthew-humphrey, 2022-02-18:
My issue turned out to be a mechanical one. With some careful testing with a set of feeler gauges, I was able to conclusively determine that the probe was triggering at a different height from the left hand side of the bed to the right. The issue turned out to be the Z axis bearing holders being loose. This was apparently allowing enough motion in the extruder body for the probe to tilt slightly, and apparently these inductive probes are very sensitive to their angle relative to the bed surface.

Klipper was and is working fine for me; the issue was entirely in my printer.

mattborkin, 2022-02-18:
> My issue turned out to be a mechanical one. With some careful testing with a set of feeler gauges, I was able to conclusively determine that the probe was triggering at a different height from the [...]

Do you think that was the case with the others in your thread as well - or was that fix unique to you?

matthew-humphrey, 2022-02-18:
> Do you think that was the case with the others in your thread as well - or was that fix unique to you?

It is really hard to say. My opinion is that there is not anything fundamentally wrong with Klipper's mesh bed leveling algorithm. The guy who wrote this is the same person that wrote Moonraker. He writes good, clean code, and takes bugs very seriously. 

There are so many variables with 3D printers, it's often really challenging to troubleshoot these things. I replaced all sorts of parts on my printer, rolled back to the original (Prusa) firmware, and even after I was able to determine that the probe was 100% giving consistently wrong readings, I didn't figure it out until some person on Discord told me to check those Z bearing holders. 

You've probably done all this already, but go through and grab every part of the frame and make sure there is not any wobble anywhere - especially around the extruder and the probe. Also, check that your frame is square, as I've heard that can cause similar issues.