Klipper3d/klipper #4600: v0.9.1-699-g23bb6fa1 causes unability to home of voron 2.4 with 4 z motors

Vassssko, 2021-08-20:
Hello.

I've just updated klipper and got an error during home routine:
"Homing instance has no attribute 'get_stepper_trigger.positions'", 'error': 'WebRequestError')

Rolling back to previous release fixes the issue instantly.
[Uploading klippy.log…]()
[image]

Vassssko, 2021-08-20:
Found out that klipper_z_calibration have reference to that issue: https://github.com/protoloft/klipper_z_calibration/issues/13 as well.

LeandroMarceddu, 2021-08-20:
Works fine here. Can you attach your printer.cfg?

LeandroMarceddu, 2021-08-20:
> Found out that klipper_z_calibration have reference to that issue: [protoloft/klipper_z_calibration#13](https://github.com/protoloft/klipper_z_calibration/issues/13) as well.

It looks like that module is to blame by using something that's not available any more. Disabling this module should get you up and running again.

Vassssko, 2021-08-20:
Found out that module was updated, and I was not fetched the update correctly. I have to doublecheck this, but it looks like the issue have that rootcause. Anyway for all of the folks having similar issue I would like to point to this: https://github.com/protoloft/klipper_z_calibration/issues/13#issuecomment-902526509

closed 2021-08-20 by Vassssko