Klipper3d/klipper #4602: Crash during print - RuntimeError: maximum recursion depth exceeded in cmp

gcormier, 2021-08-21:
Attached log. Was just printing, on the 2nd layer it just shutdown.

[klipper.log](https://github.com/KevinOConnor/klipper/files/7026078/klipper.log)

gcormier, 2021-08-22:
Again today.

1. Start print
2. Mesh Bed level begins
3. Levelling error's out due to out of tolerance sensor (this is what I'm trying to fix! Get a Klicky print going and replace it)
4. Sat for a bit and then crash.
[klippy.log](https://github.com/KevinOConnor/klipper/files/7027920/klippy.log)

Same error. Kevin on discord said watch for SDcard corruption and do a pull - this was a new pull today, updated to `v0.9.1-732-g7f704c47`

Will keep updating as things progress.

Arksine, 2021-08-22:
One of the clients connected to Moonraker is polling object state rather than subscribing.  It also appears to be making a bunch of redundant requests, so there is probably some kind of issue client side.  Somehow this must be causing an error in the query handler.

Aside from Fluidd/Mainsail what clients do you have connected?  If possible I'll see if I can reproduce.

gcormier, 2021-08-22:
I do have MoonCord installed - not sure if this could affect?

Another crash just now.

Arksine, 2021-08-22:
Possible.  I would suggest disabling Mooncord to see if it continues.  I'll see if I can get it to reproduce.

gcormier, 2021-08-23:
I disabled MoonCord, and so far have about 7 hours of printing no crashes.

I'm going to suspect if you think likely the culprit, we can close the issue. I will send the author of MoonCord to this issue.

eliteSchwein, 2021-08-23:
Hey Dev from MoonCord here, yeah i poll every half second or so instead of subscription, will change that!

Arksine, 2021-08-23:
@eliteSchwein I think the issue is duplicate requests.  To illustrate, below are the last 20 requests pulled from the log: 
```
Dumping 20 requests for client 3003945568
Received 1676.636580: {"id": 2785466576, "method": "objects/query", "params": {"objects": {"webhooks": null, "virtual_sdcard": null, "print_stats": null, "gcode_move": null, "system_stats": null, "display_status": null}}}
Received 1676.636580: {"id": 2785467024, "method": "objects/query", "params": {"objects": {"mcu rpi": null, "mcu": null}}}
Received 1676.636580: {"id": 2785467440, "method": "objects/query", "params": {"objects": {"print_stats": null}}}
Received 1676.636580: {"id": 2785468336, "method": "objects/query", "params": {"objects": {"webhooks": null, "virtual_sdcard": null, "print_stats": null, "gcode_move": null, "system_stats": null, "display_status": null}}}
Received 1676.636580: {"id": 2785517968, "method": "objects/query", "params": {"objects": {"mcu rpi": null, "mcu": null}}}
Received 1676.636580: {"id": 2785518384, "method": "objects/query", "params": {"objects": {"print_stats": null}}}
Received 1676.636580: {"id": 2785519280, "method": "objects/query", "params": {"objects": {"webhooks": null, "virtual_sdcard": null, "print_stats": null, "gcode_move": null, "system_stats": null, "display_status": null}}}
Received 1676.636580: {"id": 2785519728, "method": "objects/query", "params": {"objects": {"mcu rpi": null, "mcu": null}}}
Received 1676.636580: {"id": 2785520144, "method": "objects/query", "params": {"objects": {"print_stats": null}}}
Received 1676.636580: {"id": 2785521040, "method": "objects/query", "params": {"objects": {"webhooks": null, "virtual_sdcard": null, "print_stats": null, "gcode_move": null, "system_stats": null, "display_status": null}}}
Received 1676.636580: {"id": 2785521488, "method": "objects/query", "params": {"objects": {"mcu rpi": null, "mcu": null}}}
Received 1677.127723: {"id": 2796467568, "method": "objects/query", "params": {"objects": {"print_stats": null}}}
Received 1677.127723: {"id": 2796405328, "method": "objects/query", "params": {"objects": {"webhooks": null, "virtual_sdcard": null, "print_stats": null, "gcode_move": null, "system_stats": null, "display_status": null}}}
Received 1677.127723: {"id": 2796405776, "method": "objects/query", "params": {"objects": {"mcu rpi": null, "mcu": null}}}
Received 1677.127723: {"id": 2796406096, "method": "objects/query", "params": {"objects": {"print_stats": null}}}
Received 1677.127723: {"id": 2796406864, "method": "objects/query", "params": {"objects": {"webhooks": null, "virtual_sdcard": null, "print_stats": null, "gcode_move": null, "system_stats": null, "display_status": null}}}
Received 1677.127723: {"id": 2796407216, "method": "objects/query", "params": {"objects": {"mcu rpi": null, "mcu": null}}}
Received 1677.127723: {"id": 2796408464, "method": "objects/query", "params": {"objects": {"print_stats": null}}}
Received 1677.127723: {"id": 2796042640, "method": "objects/query", "params": {"objects": {"webhooks": null, "virtual_sdcard": null, "print_stats": null, "gcode_move": null, "system_stats": null, "display_status": null}}}
Received 1677.127723: {"id": 2825958000, "method": "objects/query", "params": {"objects": {"mcu rpi": null, "mcu": null}}}

````
As you can see, they are all "query" requests, and the timestamps show that duplicate requests are being issued.  Going to a subscription based model will alleviate the issue, so I concur that it would be a good choice.

I still would like to try to reproduce the underlying issue in Klipper as I'm not sure exactly what is causing the query handler to produce an error here.  I can't see how its possible that the recursion limit has been reached, by default it is set to 1000.

KevinOConnor, 2021-08-24:
> I can't see how its possible that the recursion limit has been reached, by default it is set to 1000.

That certainly is odd.  Indeed, the callstack on the error is only 8 deep.  As a wild guess, there may have been enough pending green threads in the reactor that the total callstacks across all of them exceeded 1000.

The `objects/query` method is never faster than `objects/subscribe`.  Perhaps we should add an error check to webhooks.py to return an error if a query is received while a query is pending from the same client.  That may prevent this type of error in the future.

-kevin