Klipper3d/klipper #4627: Unable to parse max_z_accel

jerryjr1980, 2021-08-28:
I'm new to Klipper and Octoprint, followed the guide to install Klipper on BTT Octopus board on an ender 3 v2, used both example files to edit printer.cfg and continue to get the same error
[klippy (2).log](https://github.com/KevinOConnor/klipper/files/7070515/klippy.2.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/7070516/printer.txt)

randellhodges, 2021-08-28:
The log file you attached indicates it cannot find/read the printer.config file. Perhaps you attached the wrong log file?

jerryjr1980, 2021-08-28:
Is there a way to generate a new log file? That was the only one in the folder, it was initially unable to access the printer.cfg but a chmod 777 fixed that issue.

randellhodges, 2021-08-28:
A restart after should create new log files. You might want to try the klipper discord for support, or maybe a tool like th33xitus/kiauh (on github) for installing and updating klipper.

jerryjr1980, 2021-08-28:
[klippy.log](https://github.com/KevinOConnor/klipper/files/7071041/klippy.log)
here is the latest klipper.log

closed 2021-08-30 by github-actions[bot]