Klipper3d/klipper #4628: gcode_button cannot set RPI GPIO active low with pullup

pacmac, 2021-08-28:
Version: v0.9.1-741-g8d4d16d5

I cannot seem to get the RPI GPIO to be active low. I have defined the pin active low with pullup: 

```
[gcode_button MY_BUTTON]
pin: ^!host:gpio24        ## Pin 18
press_gcode:
    M117 Pressed
release_gcode:
  M117
``` 

however it is still set as active high and no pullup: 

```
# gpioinfo | grep "GPIO24"
line  24:     "GPIO24"    "klipper"   input  active-high [used]
# raspi-gpio get 24
GPIO 24: level=0 fsel=0 func=INPUT pull=DOWN
```
What am I doing wrong, and yes I have done a firmware restart and even a reboot.

KevinOConnor, 2021-08-28:
Many versions of Linux don't support setting the pullups/pulldowns via the kernel interface Klipper uses (eg, #4207).  Alas, there isn't currently a good way to notify users of this.

Otherwise, if you need further info on this, you should follow up on one of the Klipper community servers (Discourse or Discord).

-Kevin

pacmac, 2021-08-28:
I have checked usin journalctl -f and the klippy log and I dont see any error being raises when the firmware is restarted ?

Where would I see this error, and is there a work-around ?

I.e. can I set the GPIO pullup and polarity outside of Klipper and klipper will still work / use it ?

closed 2021-08-30 by github-actions[bot]