Klipper3d/klipper #4629: [FR][Brainstorming] Automated Z-Offset Calibration With Physical Endstop and BLTouch

hawkeyexp, 2021-08-28:
Since a while i'm thinking about the possibility to make it obsolete to manually calibrate the z-offset by using the combination of a microswitch endstop and bltouch which allows additional the change the toolhead or nozzle without manually recalibration.

Setup: printer with BLTouch and microswitch endstop for z

Idea:

1) the printer detects the microsowitch endstop with the bltouch for first z-value
2) now we hit the microswitch endstop with the nozzle to get 2nd z-value
3) with the diff between 1st z-value and 2nd z-value we know the z-offset  (but we have to subtract the way the microswitrch needs to trigger to get the correct final z-offset value cause bltouch triggers on surface of endstop without pushing)
4) This way we get an automated calibration routine which would be very nice to have available.

How do you think about or possible have you tried something like this to include?

randellhodges, 2021-08-28:
https://github.com/KevinOConnor/klipper/pull/4614
https://github.com/protoloft/klipper_z_calibration

Something like that? Dunno about using a bltouch with that one, probably.

hawkeyexp, 2021-08-29:
looks good - i will give it a try...

Talha909, 2021-08-30:
can we use this on ender 3

KevinOConnor, 2021-08-30:
For discussions on features, it's probably best to use the Klipper Discourse server.  You're likely to find a wider audience there.  https://www.klipper3d.org/Contact.html

-Kevin

hawkeyexp, 2021-09-08:
> 
> 
> For discussions on features, it's probably best to use the Klipper Discourse server. You're likely to find a wider audience there. https://www.klipper3d.org/Contact.html
> 
> -Kevin

Sorry for posting here but i was interested on your opinion about this case :-)

closed 2021-09-25 by hawkeyexp