Klipper3d/klipper #4632: Quad_Gantry_level leaves bed tilted.

swatti87, 2021-08-29:
I have a problem, the system seems to probe all four corners nicely (genuine BLtouch) BUT it leaves the bed crooked.

I have FOUR motors to move my corners. I tried three but the issue remained, so i went four and its still crooking it.
500x500 bed. The bed tends to drop after power is cut so i need to level quite often.

I have this in my config:
[quad_gantry_level]
horizontal_move_z: 10
retry_tolerance: 0.009
speed: 250
max_adjust: 16
gantry_corners:
    10,-70   #Z  motor and rod
    10,550   #Z1 motor and rod
    550,470  #Z2 motor and rod
    450,-70  #Z3 motor and rod
points:
    90,90
    90,430              
    430,430 
    430,90

[bed_tilt]
horizontal_move_z: 10
speed: 2
points:
    50,50
    50,450              
    450,450 
    450,50
    
One question is, what is the difference between "bed_tilt" and "quad_gantry_level" as i can seem to have both... One that works would be fine by me. Bed_tilt is commented out, still same issue.   
If the bed is tilted, it does try to move it to compensate, its just not doing it right. I suspected Z123 were not in right corners but i have tested each by moving the motor via manual commands, corners are correct.

I also noticed an odd one running QGL: 
$ QUAD_GANTRY_LEVEL
// probe at 90.000,90.000 is z=2.834408
// probe at 90.000,430.000 is z=2.764408
// probe at 430.000,430.000 is z=3.584408
// probe at 430.000,90.000 is z=3.946908
// Gantry-relative probe points:
// 0: 7.165592 1: 7.235592 2: 6.415592 3: 6.053092
// Actuator Positions:
// z: 7.423132 z1: 7.426846 z2: 7.426846 z3: 7.423132
// Average: 7.424989
// Making the following Z adjustments:
// stepper_z = 0.001857
// stepper_z1 = -0.001857
// stepper_z2 = -0.001857
// stepper_z3 = 0.001857
​
Notice the stepper_z/1/2/3 numbers in the last segment. The number vary per each probe/QGL BUT all four numbers are always the same. Why? Is it adjusting always the same four numbers!?

Does anyone have a config-sample of proven and working machine that i could copy over and start a-new. I cant seem to make the sample-file in klipper docs to work... Something small is off but i cant ficure out what.

BlackStump, 2021-08-29:
I gather from what you said you have a moving bed not a moving gantry
[Did you read the docs before use Quad Gantry Level?](https://www.klipper3d.org/Config_Reference.html#quad_gantry_level)
exert from docs - _WARNING: Using this on a moving bed may lead to undesirable results._

swatti87, 2021-08-30:
I actually missed that... 
Should i be using "z-tilt" instead? What do i need to change, other then the name?

swatti87, 2021-08-30:
Ookkey... Swapped in "Z_tilt" and its couple settings and it looks like its working now.