Klipper3d/klipper #4641: AXDL

Talha909, 2021-08-31:
I have update the klipper with new version. After update I try to run TEST_RESONANCES AXIS=X. The system was shut down after that. I have to restart the system. But this was working good in before update. I have raspbery pi zero skr 1.4 turbo
[klippy.log](https://github.com/KevinOConnor/klipper/files/7083682/klippy.log)

bill-orange, 2021-08-31:
Ditto.  ACCELEROMETER_QUERY now gives the "not found" error.  I built and reloaded the secondary MCU firmware.  didn't help.

bill-orange, 2021-08-31:
** UPDATE**

@Talha909 

There was an update to Klipper while I was at lunch. I restored my old firmware.bin and rebooted the printer. I updated Klipper in Mainsail. I compiled a fresh RPI MCU and rebooted. When compiling, I added a 'make clean' before 'make' to the procedure. ACCELEROMETER_QUERY still did not find the sensor! I ran TEST_RESONANCES AXIS=X. Inexplicably ACCELEROMETER_QUERY now worked. So with a few oddities, all is now well. There remains one unknow. When I made a new MCU firmware.bin this morning, it would not work, I am running on my archive copy (this is why you make backups). I did not test the ability of the very latest Klippy to make MCU firmware for my SKR1.3 since everything now works. I am not pressing my luck. perhaps someone else can test this.

vladimir-poleh, 2021-08-31:
I had same issue with Klipper hangs after `TEST_RESONANCES`. It was caused by incorrectly wired accelerometer.

Talha909, 2021-09-01:
Its not like that my ACCELEROMETER works good before update. But after update when I run the TEST_RESONANCES AXIS=X command it started process and after 2 or 3 seconds  the klipper shutdown itself.  But before update I got the result and my ACCELEROMETER works good. And I also waited for 4 or 5 mintues. But there is no file created in the system.

bill-orange, 2021-09-01:
Did you rebuild and reinstall RPI MCU? If you did, did you run ‘make clean’ before ‘make flash’?

Talha909, 2021-09-02:
after update I rebuild and reinstall the RPI MCU. Like the document . Which query accelerometer it gives me the value.

Talha909, 2021-09-02:
> Did you rebuild and reinstall RPI MCU? If you did, did you run ‘make clean’ before ‘make flash’?
> […](#)
> Sent from my iPhone

I did not make clean before the  make flash. Please tell me how i do that.

unkn0wndude, 2021-09-02:
I am experiencing the same issue. Tried flashing pi and board again, but no change. Running latest klipper.

16:51:43
TEST_RESONANCES AXIS=Y
16:51:47
Disabled [input_shaper] for resonance testing
16:51:47
Testing frequency 5 Hz
16:51:47
Testing frequency 6 Hz
16:51:47
Testing frequency 7 Hz
16:51:47
Testing frequency 8 Hz
16:51:47
Testing frequency 9 Hz
16:51:50
Testing frequency 10 Hz
16:51:51
Testing frequency 11 Hz
16:51:52
Testing frequency 12 Hz
16:51:53
Testing frequency 13 Hz
16:51:54
Testing frequency 14 Hz
16:51:55
Testing frequency 15 Hz
16:51:56
Testing frequency 16 Hz
16:51:57
Testing frequency 17 Hz
16:51:58
Testing frequency 18 Hz
16:51:59
Testing frequency 19 Hz
16:52:00
Testing frequency 20 Hz
16:52:01
Testing frequency 21 Hz
16:52:02
Testing frequency 22 Hz
16:52:03
Testing frequency 23 Hz
16:52:04
Testing frequency 24 Hz
16:52:05
Testing frequency 25 Hz
16:52:06
Testing frequency 26 Hz
16:52:07
Testing frequency 27 Hz
16:52:08
Testing frequency 28 Hz
16:52:09
Testing frequency 29 Hz
16:52:10
Testing frequency 30 Hz
16:52:11
Testing frequency 31 Hz
16:52:12
Testing frequency 32 Hz
16:52:13
Testing frequency 33 Hz
16:52:14
Testing frequency 34 Hz
16:52:15
Testing frequency 35 Hz
16:52:16
Testing frequency 36 Hz
16:52:17
Testing frequency 37 Hz
16:52:18
Testing frequency 38 Hz
16:52:19
Testing frequency 39 Hz
16:52:20
Testing frequency 40 Hz
16:52:21
Testing frequency 41 Hz
16:52:22
Testing frequency 42 Hz
16:52:23
Testing frequency 43 Hz
16:52:24
Testing frequency 44 Hz
16:52:25
Testing frequency 45 Hz
16:52:26
Testing frequency 46 Hz
16:52:27
Testing frequency 47 Hz
16:52:28
Testing frequency 48 Hz
16:52:29
Testing frequency 49 Hz
16:52:30
Testing frequency 50 Hz
16:52:31
Testing frequency 51 Hz
16:52:32
Testing frequency 52 Hz
16:52:33
Testing frequency 53 Hz
16:52:34
Testing frequency 54 Hz
16:52:35
Testing frequency 55 Hz
16:52:36
Testing frequency 56 Hz
16:52:37
Testing frequency 57 Hz
16:52:38
Testing frequency 58 Hz
16:52:39
Testing frequency 59 Hz
16:52:40
Testing frequency 60 Hz
16:52:41
Testing frequency 61 Hz
16:52:42
Testing frequency 62 Hz
16:52:43
Testing frequency 63 Hz
16:52:44
Testing frequency 64 Hz
16:52:45
Testing frequency 65 Hz
16:52:46
Testing frequency 66 Hz
16:52:47
Testing frequency 67 Hz
16:52:48
Testing frequency 68 Hz
16:52:49
Testing frequency 69 Hz
16:52:50
Testing frequency 70 Hz
16:52:51
Testing frequency 71 Hz
16:52:52
Testing frequency 72 Hz
16:52:53
Testing frequency 73 Hz
16:52:54
Testing frequency 74 Hz
16:52:55
Testing frequency 75 Hz
16:52:56
Testing frequency 76 Hz
16:52:57
Testing frequency 77 Hz
16:52:58
Testing frequency 78 Hz
16:52:59
Testing frequency 79 Hz
16:53:00
Testing frequency 80 Hz
16:53:01
Testing frequency 81 Hz
16:53:02
Testing frequency 82 Hz
16:53:03
Testing frequency 83 Hz
16:53:04
Testing frequency 84 Hz
16:53:05
Testing frequency 85 Hz
16:53:06
Testing frequency 86 Hz
16:53:07
Testing frequency 87 Hz
16:53:08
Testing frequency 88 Hz
16:53:09
Testing frequency 89 Hz
16:53:10
Testing frequency 90 Hz
16:53:11
Testing frequency 91 Hz
16:53:12
Testing frequency 92 Hz
16:53:13
Testing frequency 93 Hz
16:53:14
Testing frequency 94 Hz
16:53:15
Testing frequency 95 Hz
16:53:16
Testing frequency 96 Hz
16:53:17
Testing frequency 97 Hz
16:53:18
Testing frequency 98 Hz
16:53:19
Testing frequency 99 Hz
16:53:20
Testing frequency 100 Hz
16:53:21
Testing frequency 101 Hz
16:53:22
Testing frequency 102 Hz
16:53:23
Testing frequency 103 Hz
16:53:24
Testing frequency 104 Hz
16:53:25
Testing frequency 105 Hz
16:53:26
Testing frequency 106 Hz
16:53:27
Testing frequency 107 Hz
16:53:28
Testing frequency 108 Hz
16:53:30
Testing frequency 109 Hz
16:53:30
Testing frequency 110 Hz
16:53:33
Testing frequency 111 Hz
16:53:34
Testing frequency 112 Hz
16:53:34
Testing frequency 113 Hz
16:53:34
Testing frequency 114 Hz
16:53:35
Testing frequency 115 Hz
16:53:37
Testing frequency 116 Hz
16:53:37
Testing frequency 117 Hz
16:53:38
Testing frequency 118 Hz
16:53:39
Testing frequency 119 Hz
16:53:41
Testing frequency 120 Hz
16:53:42
Testing frequency 121 Hz
16:53:43
Testing frequency 122 Hz
16:53:43
Testing frequency 123 Hz
16:53:44
Testing frequency 124 Hz
16:53:45
Testing frequency 125 Hz
16:53:46
Testing frequency 126 Hz
16:53:47
Testing frequency 127 Hz
16:53:48
Testing frequency 128 Hz
16:53:49
Testing frequency 129 Hz
16:53:50
Testing frequency 130 Hz
16:53:51
Testing frequency 131 Hz
16:53:52
Testing frequency 132 Hz
16:53:53
Testing frequency 133 Hz
16:53:54
Re-enabled [input_shaper]

ACCELEROMETER_QUERY gives no output. No files are written to /tmp/.

bill-orange, 2021-09-02:
Immediately before giving the command `make flash`, give the command `make clean`

unkn0wndude, 2021-09-02:
I managed to get this work by reverting back to commit (e03e0e9dac5694b35b81726bd2ed18f1a67b0a26) right before adxl changes. So i reflashed both board and pi. After that I got an error from ACCELEROMETER_QUERY - "Invalid adxl345 id (got 0 vs e5)". So I rebooted raspberry without touching anything. After rebooting the pi the ACCELEROMETER_QUERY gave me this output:
adxl345 values (x, y, z): -1376.853660, 1070.886180, 61346.479740

Good luck everyone

Talha909, 2021-09-04:
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/7109516/klippy.log)
I have raspberry Pi Zero. It was working good before the update. But after update my klipper shutdown. see screen shot

robthide37, 2021-09-04:
you have to lower the polling freq the zero cannot keep up

closed 2021-09-06 by github-actions[bot]

Talha909, 2021-09-06:
update. Today reinstall all the things but still got the error. how can i lower the polling freq .