Klipper3d/klipper #4643: Printer is not ready The klippy host software is attempting to connect. Please retry in a few moments.

arshillimbani123, 2021-08-31:
my printer is not connecting to klipper.
[image]
in ssh it shows I device but in a fluid it not connecting
[image]
what should I do ??
I tried changing USB cable and tried every USB port in raspberry pi
[klippy.log](https://github.com/KevinOConnor/klipper/files/7085727/klippy.log)

theopensourcerer, 2021-08-31:
Unfortunately Github is not used for this kind of issue. Please see:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

bill-orange, 2021-08-31:
Same here after latest update.

theophile, 2021-08-31:
You need to follow these instructions to set up your pi as a host mcu: https://www.klipper3d.org/RPi_microcontroller.html#install-the-rc-script

arshillimbani123, 2021-08-31:
> Same here after latest update.

I am also facing this problem after update.

bill-orange, 2021-08-31:
@theophile I did as directed in your link and it did not work.  I eventually got it to work.  In the successful compile, I added a 'make clean' before make.  That may be a necessary step in this rare case case.

theophile, 2021-08-31:
Yes, you should always do a `make clean` before recompiling for a different architecture.

closed 2021-09-01 by arshillimbani123