Klipper3d/klipper #4645: Bug: tmcuart_response/tmcuart_send broken UTF-8 output in klipper.log

dingyifei, 2021-09-01:
## Observed behavior: 

```gdb
Receive: 12 71128.353854 71128.346986 18: seq: 15, tmcuart_response oid=0 read='\n\xfa\xef-\xa0*\n( \xe6'
Receive: 13 71128.364091 71128.357921 18: seq: 17, tmcuart_response oid=0 read='\n\xfa/ \x80\x00\x02\x08\xa0\x89'
Receive: 14 71128.488720 71128.478855 15: seq: 1a, analog_in_state oid=14 next_clock=760889600 value=21777
Receive: 15 71128.523350 71128.516893 18: seq: 10, tmcuart_response oid=2 read='\n\xfa\xef-\xa0*\n( \xe6'
```

```gdb
Sent 97 71132.578838 71132.575450 30: seq: 1f, tmcuart_send oid=2 write='\xea\x03( \xc3' read=10, queue_step oid=9 interval=22279 count=5 add=1687, queue_step oid=9 interval=31761 count=3 add=8111
```

## Expected behavior:

The `read=` and `write=` should output binary/hex/unsigned int output that represents the data read through UART.

## Related files:

[tmc_uart.py](https://github.com/KevinOConnor/klipper/klippy/extras/tmc_uart.py)

[tmc_uart.c](https://github.com/KevinOConnor/klipper/blob/master/src/tmcuart.c)

The issue should be the `%*s` formatting operator, but I'm not confident in my C/Python for me to PR...

KevinOConnor, 2021-09-02:
I don't understand what is being reported here.

-Kevin

dingyifei, 2021-09-03:
the uart reads are being formatted as strings but utf8 is not the proper encoding for displaying the binary output from uart. The output is a mix of non existent Unicode like `\xfa` and random ymbols that was converted from binary to Unicode. 

`\n\xfa\xef-\xa0*\n( \xe6`

It's just the logging message being really weird and not printing in the correct way

closed 2021-09-13 by dingyifei