Klipper3d/klipper #4650: Issue Starting Klipper since updating

behold81, 2021-09-03:
Since doing an update this came up. I reflashed the MCU with no improvements. 

Whats the cause?

mcu 'rpi': Unknown command: adxl345_status

This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).

Known versions: host=v0.9.1-771-g8865f579, mcu=v0.9.1-771-g8865f579, rpi=v0.9.1-627-g121052ad

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

Gilabite, 2021-09-03:
the answer is in the error message plain as day.... you didnt update the pi too.

behold81, 2021-09-03:
> the answer is in the error message plain as day.... you didnt update the rpi too.

Unless I missed something obvious I did update my pi. It had 16+ updates to it.

Gilabite, 2021-09-03:
The 16 updates would have been the host OS. You also setup the pi as mcu for the adxl...

randellhodges, 2021-09-03:
It seems you following these instructions at some point:
https://www.klipper3d.org/RPi_microcontroller.html

And are using the ADXL connected to the PI. Updating klipper or the host doesn't update the "klipper mcu firmware" on the pi when using the pi as an MCU. You have to run thru the steps to make menuconfig and build the firmware, just like any other MCU.

behold81, 2021-09-03:
Right. That makes sense.  Thank you.  

I’ll try the MCU update. Thanks.

DaveWW00, 2021-09-04:
I get same error after updating everything, only way i could fix it was fully starting fresh reinstalling everything.  Would be nice if option was added to apply update needed to Pi also or at very least provide notice that you might break software when applying update.  Thanks.

closed 2021-09-06 by github-actions[bot]

usab, 2021-09-06:
Some problem here. 
I did not install any firmware manually, all was done automatically via vcore os and the update buttons.