Klipper3d/klipper #4657: PT1000 + Mosquito Hotend on CR-10S Pro V2 Issues in the Final Stretch

AlchemyCustoms, 2021-09-04:
I've made it all the way to the point that I can connect to the printer, but I get immediately disconnected due to my hotend reading 140375333.7°C... I think it's a bit cooler than that; so rather than force my way through that, I thought I'd ask if anyone has any experience or input on this.

Mainly I'm just trying to figure out where else I can tell the printer that I've changed the thermistor out for a PT1000... It looks to me like the Marlin firmware natively supports it, but it also looks like Klipper does too, so I'm confused on if I have a hardware error or a software problem!

I've been at this for HOURS now... losing my mind. Other things to note: Installation is based on KIAUH script inside of a Debian Buster Docker container on my Unraid server. Printer has the Mosquito hotend, and a Slice Engineering PT1000 and 50W heater. Wires all appear to be in good order from what I can best tell. Thanks in advance for the help, I appreciate it greatly!

[klippy.log](https://github.com/KevinOConnor/klipper/files/7109441/klippy.log)

[image]

closed 2021-09-04 by AlchemyCustoms

Zyorg12, 2021-11-18:
Hi, were you ever able to get this fixed? I have the exact same setup and issue and I've been pulling out my hair for two days now.

AlchemyCustoms, 2021-11-18:
Yep, I've got ya Zyorg12.

I've come quite a long way since this post so I'm feeling pretty confident I can help ya out. Message my business page on FB if you need me to just chat with you about it for a bit so we don't clog up the forums. That being said, I've attached my config and everything relevant for the CR-10S Pro V1/V2 (I have the V2) below.

Also, please note that I am using the TYHI Net Mosquito Mount made by TYHI3D on [Thingiverse](https://www.thingiverse.com/thing:4348224), so my offsets, mesh bed leveling points, etc. will all be in accordance with that. I will point out that some of them aren't super accurate, so if you use the same mount and decide to dial it in more, I'd love to know the perfect points but haven't had time to find them myself.

Remove the hash tags in front of the lines you are going to use when you actually do the tests for them. Input shaping and Pressure Advance are critical on these CR-10S Pros... Made an insane difference for my print quality. It does take a good bit of time and a decent amount of filament, but it is respectable when you realize how much benefit it brings for the time and overall money invested in filament.

```
[mcu]
#For serial, it will be something SIMILAR to the line below, but maybe not exactly. If you've got communication to the Klipper firmware already from the webUI, then you have nothing to worry about here.
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AG0JPIWH-if00-port0
pin_map: arduino

[virtual_sdcard]
path: ~/gcode_files

[printer]
kinematics: cartesian
max_velocity: 150 #Make sure to set your travel velocity to 150mm/s in Cura or whatever slicer you use as well
max_accel: 7000 #I typically set my printer to only run around 1400mm/s^2 for acceleration, but that's with a VERY well tuned machine and an extremely solid table. Please note in the next block that I am also using the Input Shaper. Almost a must-have at this acceleration on a CR-10S Pro V1/V2.
max_accel_to_decel: 7000 #Same For this line, except I run around 60mm/s^2 SLOWER than the acceleration value
max_z_velocity: 20
max_z_accel: 120

#[input_shaper]
#shaper_freq_x: 27.84 #These numbers WILL NOT be accurate for your machine; CALIBRATE ACCORDINGLY AND THOROUGHLY
#shaper_freq_y: 31.32
#shaper_type: mzv #Same with the shaper type, it may not be the same. Be patient and try several variations if budget allows.

[fan]
pin: ar9

#Offsets work for the TYHI Net Mosquito Mount
[bltouch]
sensor_pin: ^ar19
control_pin: ar11
stow_on_each_sample: False
probe_with_touch_mode: True
x_offset: -27
y_offset: -2
z_offset: 5.150 #Set this to your actual Z-offset. This is the one thing I never could get just right out of the gate accurate.
speed: 3.0 #Can decrease by up to half but not really necessary... Could increase a bit but eh, stability
samples: 2 #Change this to 1  for faster probing or 3 for more probes per point / more accuracy
sample_retract_dist: 3

#I'm a little lax here in the hysteresis, same for the bed as well...
#Not the safest, but I have some MAD temp fluctuations throughout the day at my shop
[verify_heater extruder]
heating_gain: 2
check_gain_time:60
hysteresis: 10
max_error: 160

[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
#Calibrate this VERY carefully and get it as absolutely spot-on as possible
#Most likely ALL of your other quality issues will stem from this if all else is failing.
rotation_distance: 22.801
microsteps: 16
full_steps_per_rotation: 200
nozzle_diameter: 0.400
filament_diameter: 1.747 #Be VERY specific if you have some calipers and can measure multiple parts of the filament
max_extrude_only_distance: 500.0
max_extrude_only_velocity: 100.0
max_extrude_only_accel: 1000.0
heater_pin: ar10 #This is where the 50W heater is plugged in
sensor_type: PT1000 #This is where the PT1000 gets specified
[... 201 lines cut ...]
#   for more details on using this feature.
#min_freq: 5
#   Minimum frequency to test for resonances. The default is 5 Hz.
#max_freq: 120
#   Maximum frequency to test for resonances. The default is 120 Hz.
#accel_per_hz: 75
#   This parameter is used to determine which acceleration to use to
#   test a specific frequency: accel = accel_per_hz * freq. Higher the
#   value, the higher is the energy of the oscillations. Can be set to
#   a lower than the default value if the resonances get too strong on
#   the printer. However, lower values make measurements of
#   high-frequency resonances less precise. The default value is 75
#   (mm/sec).
#hz_per_sec: 1
#   Determines the speed of the test. When testing all frequencies in
#   range [min_freq, max_freq], each second the frequency increases by
#   hz_per_sec. Small values make the test slow, and the large values
#   will decrease the precision of the test. The default value is 1.0
#   (Hz/sec == sec^-2).

[save_variables]
filename: ~/variables.cfg

[include macros/*.cfg]
```