Klipper3d/klipper #4660: Another "Move exceeds maximum extrusion (2.037mm^2 vs 2.000mm^2)"

vert-rouge, 2021-09-05:
Hi,

I’ve read the other issues, and mine does not seems to be related to the slicer.
I’ve done the print multiple time. Sometimes it works really well, and sometimes, it doesn’t.

example in the klippy, where the interesting stuff starts at line #22114
line 22114 : first unsuccessful print
line 22206 : second unsuccessful print.
line 22292 : reboot
line 23191 : successful print

the print is actually not finished.
I have the gcode if it’s useful.

I’m using Fluidd, and use the exact same file.

strayr, 2021-09-07:
this looks like a support request? 
Is this your purge line? Is it really fat? 
Log doesn't look like it's attatched properly
See
```
[extruder]
max_extrude_cross_section: 1 #set this to something sane
```
you should probably take this to the discourse, community discord or r/klippers
interesting that it's only sometimes over the limit though i think that's some odd floating point rounding behaviour in the iterative solver? I had similar behaviour in a second print of a fat purge line failed i thing becasue it was using accel and PA from the last print not the accel from the config file and it was enough to push the floating point rounding over the edge.

vert-rouge, 2021-09-08:
Hi,
I come from the discourse.
I really think it is a bug as I made section calculation by hand.
[klippy.zip](https://github.com/KevinOConnor/klipper/files/7127564/klippy.zip)

My guess is there is some problems with pressure advance. I looked at the code, but I didn’t manage to get debug log working.

closed 2021-09-16 by github-actions[bot]

vert-rouge, 2021-09-16:
I don't think this is a request for help.

There is a problem if a code produce different results, and the problem keep happening.

watzon, 2021-09-17:
I agree. I'm having this same issue both with Cura and with gcode generated by https://teachingtechyt.github.io.

In both cases there is no pressure advance settings in play, and even increasing the max_extrude_cross_section property to an insane number still gives issues.

RyushoYosei, 2021-12-30:
THere very much is a problem here, I have encountered it myself With profiles that were working flawlessly without issue for months suddenly now having this issue.