Klipper3d/klipper #4662: Resonance compensation X Axis no csv outpuz

Dreamdancer100, 2021-09-06:
Good morning have some big problem when I made resonance compensation with adxl345  2 weeks ago all is fine but after klipper update I made new test but X axis won’t give csv output for Y axis test all is fine. Made all Firmware new update all to latest but X axis won’t give output. When test run to 133 hz klipper freeze and nothing happened Input shaper was startet but no csv file I must restart to work with Printer again! If I testing after 133 hz I see temp sensors are freeze too what can I do ?

Talha909, 2021-09-06:
I have the same error message. adxl works good before the update. After update my adxl disconnect. If you find any thing please let me know.  Thanks. 
[klippy (1).log](https://github.com/KevinOConnor/klipper/files/7116170/klippy.1.log)
 and data

Dreamdancer100, 2021-09-06:
Yeah when I’ve some answer I told you the result!
The hard way is to setup new pi means reinstall  mainsail OS

Talha909, 2021-09-06:
> Yeah when I’ve some answer I told you the result!
> The hard way is to setup new pi means reinstall mainsail OS

I reinstall the pi with fresh fluidd. But the result was same. So it does not change any thing when reinstall

Dreamdancer100, 2021-09-08:
I’ve used one new adxl345 but same problem 
I think the Problem comes out klipper update

Dreamdancer100, 2021-09-08:
Mainsail OS have some update I installed and in evening I test resonance X axis again if that problem fixed

Dreamdancer100, 2021-09-08:
I‘m test after System Update no result 
Resonance test X axis no csv output

Dreamdancer100, 2021-09-11:
Anyone some idea

Dreamdancer100, 2021-09-14:
OK thanks a lot
i will contact right one

closed 2021-09-16 by github-actions[bot]