Klipper3d/klipper #4668: Confirm axdl345 not working after update of klipper

Talha909, 2021-09-07:
I had update the klipper with the new version. After that I try to check TEST_RESONANCES AXIS=X .But this command did not work. It shut down the mcu after 4 or  5 sec. 

After that I reflash the raspbery with the fluidd: v1.16.2 and this version has the fallowing klipper version v0.9.1-616-g28f60f7e -> v0.9.1-773. I test the TEST_RESONANCES AXIS=X command and its works. And gave me the result.  What's the problem in new version.  And I did not update any thing.

luigi20000, 2021-09-07:
hello does it give Unable to obtain 'spi_transfer_response' response

Talha909, 2021-09-07:
> hello does it give Unable to obtain 'spi_transfer_response' response

No

glabifrons, 2021-09-08:
Since there's already an issue for this problem, I'll add some data.

Example error:
```
Send: ACCELEROMETER_QUERY
Recv: !! No adxl345 measurements found
Changing monitoring state from "Operational" to "Error"
```

I've verified my hardware still works by moving it from one printer to another (reflashing with matching firmware prior to each test).

Working correctly:

- Ender 5 (stock board for primary mcu)
- Raspberry Pi 3
- commit be122411b588ce95785c49dbdf9e4b31a214242e

Not working:

- Sapphire Pro (stock board for primary mcu)
- Raspberry Pi 4
- commit c84956befe88daeeb9512acaa9fa82395665df16

Sorry I don't have a narrower timeframe.

Note that I'm using a Raspberry Pi Pico for the ADXL345 communication (so I can move it from machine to machine) as written up here: https://www.reddit.com/r/klippers/comments/owqvo2/raspberry_pi_pico_as_a_secondary_mcu_for/
(Thank you very much for adding support for the rp2040!)

Log attached:
[klippy.log](https://github.com/KevinOConnor/klipper/files/7126418/klippy.log)

Talha909, 2021-09-09:
One more thing when I gave command of TEST_RESONANCES AXIS=X the klipper load goes up to 450%. And it shut down itself.

gregmatic, 2021-09-09:
i have the same problem

Anti51, 2021-09-12:
After updating klipper you might have to rebuild and reflash both the printer and the linux mcu. That worked for me.

glabifrons, 2021-09-13:
> After updating klipper you might have to rebuild and reflash both the printer and the linux mcu firmware. That worked for me.

That's my SOP for Klipper.
I had to reflash the Pico to with the older Klipper firmware on the Ender 5 to get it to communicate with it. 
Reflashing again back to current on the Sapphire results in the above stated problem.
I ended up copying the old installation from the Ender 5 to the Sapphire's Pi  and swapping in the .config file from the Sapphire, rebuilt and reflashed both the Robin Nano and the Pi Pico, and it worked fine. After tuning resonance compensation, I swapped back in the Sapphire's previous (more current) firmware and removed the Pico as I no longer needed the accelerometer. Not everyone is going to be lucky enough to have this setup though.

KevinOConnor, 2021-09-14:
Not sure why that would be.  You'll need to reach out to the community for assistance though - as we don't use github to help diagnose problems with a printer.

-Kevin

glabifrons, 2021-09-14:
> Not sure why that would be. You'll need to reach out to the community for assistance though - as we don't use github to help diagnose problems with a printer.
> 
> -Kevin

I'm not sure about the other one, by my issue is definitely versions-specific and not printer-specific (as shown in my previous posts). I might be able to narrow it down to the exact patch (instead of the range I gave), but it'll take me some time (I'm not strong with git).

closed 2021-09-16 by github-actions[bot]