Klipper3d/klipper #4669: Bl touch on toolboard

robthide37, 2021-09-07:
I don't have a log for this since klipper does not know their is anything wrong. My issue is I am wiring a bl touch to 2 endstop pins which the endstop triggers and it reads the pin position when set properly but the servo does not have control. I have 5v on power but afaik the stm32 is using 3.3 logic for the pwm signal. I have tried switching the pins and inverting etc to no avail. I bought a logic shifter and will test but some tell me that the bl touch doesn't need the 5v pwm and I know it can operate open drain where it supports 5v or 3.3v logic but is that only on the endstop pin not on the pwm servo pin? Or do I need to simply power the the bl touch with 3.3v and then it will see the 3.3v signal ? I would think it needs the 5v to power the bl touch. My last thought is it is something with the code as it is just merged for multi mcu homing which is why I am making this post here just to see if their is nay chance of this or if it's purely hardware issue. Thanks in advance I just don't want to go crazy if it's just something that isn't feasible yet for the bl touch and I have to stay inductive or mechanical for now

jakep82, 2021-09-07:
Help requests should be posted on the Discord or Discourse servers. 

That said, many endstop inputs have hardware pull-ups that won't work for controlling a servo.

robthide37, 2021-09-08:
Only reason I posted here is I wanted to verify that it was not a limitation of the multi mcu implementation or if their was a software work around. And the v.61 huvud is bit released yet but was designed for bl touch. I want to make sure it is nit hardware before we mass produce them

KevinOConnor, 2021-09-14:
Alas, you are unlikely to find a useful response here - many people with similar hardware don't monitor this issue tracker.  Best to raise on Discord, Discourse, or Patreon.

-Kevin

mraction, 2021-09-16:
bl touch is working perfectly with huvud. You have to remove R19 and connect the Servo to Pin3 from J7. In my case, I soldered a 0Ohm resistor to R36 but I think it would work without.

Alternativ, you use Pin1, 5 or 6 from J2 (Debug connector)...

robthide37, 2021-09-17:
I was just trying to figure this out so thankyou. Oddly enough I used it
with huvud v.5 it's the beta v.61 boards that are giving me props. You
think their is a downside to this for production?  Like would reg endstop
still work or would it make one pin servo only basically

robthide37, 2021-09-17:
Excellent with the debug connector actually cause then we can still have 3
endstops thankyou very much for that. I thought I had tried it on the new
ones but I was honestly going nuts so I lost track of what I tried

closed 2021-10-30 by github-actions[bot]