Klipper3d/klipper #4674: SAVE_CONFIG doesn't work with extruders defined in [include]d files

charlespick, 2021-09-10:
Running `SAVE_CONFIG` after a pid tune for an extruder defined in a file included from the printer.cfg gives this error:
```
!! SAVE_CONFIG section 'extruder' option 'control' conflicts with included value
```

When loading a config file with a save_config section that contains PID info for an extruder defined in an included file you get this error:
```
Option 'control' in section 'extruder' must be specified
```

Nothing is logged to the klippy.log file during either of these events

My main printer.cfg looks like this:
```
# Updates are managed remotely with moonraker now
[include KlipperConf/Octo Creator Pro/*.cfg]
```

All the other files are at https://github.com/charlespick/KlipperConf/tree/master/Octo%20Creator%20Pro

charlespick, 2021-09-10:
I think it is worth noting that this issue does not occur with endstop phase calibration

charlespick, 2021-09-10:
scratch that, this is the log file
[klippy-2.log](https://github.com/KevinOConnor/klipper/files/7141087/klippy-2.log)

KevinOConnor, 2021-09-14:
This is the intended behavior.  SAVE_CONFIG does not support and is not intended to support parameters saved in included config files.

-Kevin

charlespick, 2021-09-15:
is there a particular reason for that? Also why does it work with other save_config circumstances?

charlespick, 2021-10-22:
I still want to know the answer to why this is the way it is￼

charlespick, 2021-11-27:
@KevinOConnor i still want to know why this is...

I'm using git to manage deployment of config updates to 20+ printers. The only way I've found to do this without having a different repo for different models of printer is to include the files from the printer.cfg. There's also use cases such as different extruders as @benjjyman mentioned above, temporary accessories, etc

closed 2022-01-10 by github-actions[bot]