Klipper3d/klipper #4675: Losing communication with MCU during resonance testing

charlespick, 2021-09-10:
I'm getting a number of issues running the resonance tester after the recent adxl345 commits. I rolled back to 6999ff6 and it began working again. By X axis tests fine but my Y axis always crashes somewhere around 80-100hz. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/7140552/klippy.log)
[image]
[image]

Talha909, 2021-09-10:
now a days every has the same issue. May be it will solved in coming days.  Which type or raspbery pi u r using.

charlespick, 2021-09-10:
rpi 4

KevinOConnor, 2021-09-14:
That's unfortunate and I don't know why that would be.  We don't use github to help diagnose problems with a printer though.  Best to see if someone on Discord or Discourse can assist.

-Kevin

charlespick, 2021-09-15:
@KevinOConnor I don't think this is a problem with my printer. This is only happening after the recent changes to adxl345. Rolling back and changing nothing about the printer solves the problem.

closed 2021-09-17 by github-actions[bot]

charlespick, 2021-09-18:
thanks, bot

dmbutyugin, 2021-09-19:
@KevinOConnor, today I've finally upgraded my installation to the the latest ADXL345 measuring code, and it seems that we are maybe on to something here. I checked the performance and the system load at the current head (59bd51d207747efcdd37b5577f817cd95ff5edbb) and before the changes to the adxl345 code (6999ff6256514474d2a309c76cef18e26acf2969). I ran the same resonance testing sequence
```
G28
TEST_RESONANCES AXIS=Y OUTPUT=raw_data,resonances NAME=...
M112 ; after all test processing is finished
```
after rebuilding and reflashing the MCU code for all MCUs, deleting `klippy.log` and restarting Klipper with the aforementioned versions of the code. The ADXL345 is connected to the Raspberry Pi 3 Model B+ (printer is Ender 3). Then I used the `scripts/graphstats.py -s` command to generate the sytem load charts, which are:

[image] | [image] |
|:--:|:--:|
|before (6999ff6256514474d2a309c76cef18e26acf2969) | after (59bd51d207747efcdd37b5577f817cd95ff5edbb) |

(I believe the title of the charts generated by the [script](https://github.com/KevinOConnor/klipper/blob/master/scripts/graphstats.py#L164) are incorrect with `-s` flag, btw). The charts demonstrate the higher CPU usage with the latest code (and a bit higher system load, though I'm not 100% sure what really counts there).

So, I think there is a real performance regression with the new adxl345 code. I looked at the code to see why this could be the case. At the very least, I suspect the change to `ADXL345._extract_samples` method at
https://github.com/KevinOConnor/klipper/blob/59bd51d207747efcdd37b5577f817cd95ff5edbb/klippy/extras/adxl345.py#L274-L307
or rather, how it is used now. Previously, the samples decoding code was located all in `ADXL345Results.decode_samples` method, which was called at the end of the measurements. This was done on purpose: that code contains relatively heavy calculations, which are not very efficient in Python. Running the vibrations test requires generating quite a lot of moves per second, e.g. at 100 Hz the test generates 400 moves per second, which is, probably, already quite a lot of calculations. Now that samples decoding (which are generated at the rate of 3200 s/s) is running in parallel with that (my understanding is that [`_api_update`](https://github.com/KevinOConnor/klipper/blob/59bd51d207747efcdd37b5577f817cd95ff5edbb/klippy/extras/adxl345.py#L386) is now called every 0.1 seconds) may be a bit too much for some setups. Though I'm surprised that @charlespick has performance issues on RPi4, it is possible that, combined with extra sysload like webcam streaming, the new code just tips off the balance and results in the inability of Klipper to deliver the required stepper steps at some point during the test.

So, Kevin, what is your take on this? Is this something worth fixing, in your opinion? It does not look like a terrible performance regression, but something that could affect some user setups that previously worked fine. I think it might be possible to split `_extract_samples` into 2 parts: one that generates the timing for the samples, which must be done online, and the actual samples decoding, which could be moved back to `ADXL345QueryHelper.decode_samples`.

Separately, I think there is an error reporting regression with the new `APIDumpHelper`: now any errors with the ADXL345 (e.g. adxl345 not connected or any other connection problems) seems to be silently 'eaten' without any feedback on the console here:
https://github.com/KevinOConnor/klipper/blob/59bd51d207747efcdd37b5577f817cd95ff5edbb/klippy/extras/motion_report.py#L40-L43
The command `ACCELEROMETER_QUERY` generates a message 'No adxl345 measurements found', `ACCELEROMETER_MEASURE` generates an empty output file, and `TEST_RESONANCES` [generates](https://github.com/KevinOConnor/klipper/blob/59bd51d207747efcdd37b5577f817cd95ff5edbb/klippy/extras/resonance_tester.py#L189) 'x-axis accelerometer measured no data'. The errors are, well, technically correct, but not particularly helpful. But this is really for another bug report, probably.

I am also attaching the klippy.log files for the before and after tests, in case some additional helpful information could be extracted from them:
[klippy_vibrations_old.log](https://github.com/KevinOConnor/klipper/files/7192831/klippy_vibrations_old.log)
[klippy_vibrations_new.log](https://github.com/KevinOConnor/klipper/files/7192832/klippy_vibrations_new.log)

Talha909, 2021-09-20:
so how i correct this now

KevinOConnor, 2021-09-21:
Hi @dmbutyugin,

Yes, I agree the new code can introduce slightly higher host cpu load (for the reasons you cite).  However, the additional load should be small.  Indeed, your graphs tend to indicate that the code went from ~35% of one core to ~40% of one core (with some peeks to 50%)?  That doesn't sound like an issue to me - in order for the code to miss a movement deadline I would think it would need to be really overloaded (eg, 100+% of a core for multiple seconds).  It's possible some "micro bursts" are occurring - for example during Python garbage collection - but it's really hard to say from the stats.  The tests I ran on my printer don't show an issue (rpi3), and it seems your printer doesn't show an issue either.  I suspect someone with a machine that is actually failing will need to do the work necessary to track down what it is about that particular environment that is causing an issue.

> The errors are, well, technically correct, but not particularly helpful.

Okay, but I'm not sure how best to make the user facing error more descriptive.  The Klipper log should have a detailed description of the issue (the `logging.exception()` call in the code you cite should write out the details to the log).  The reason it can be hard to get the data back to the console is that the capture may not have been started from the console.  Specifically, it is now possible to start the capture from the API Server.  It's also possible to run multiple tools at the same time (eg, capture and analysis).  Let me know if you have suggestions on how this can be improved.

Cheers,
-Kevin

charlespick, 2021-09-22:
@KevinOConnor Is it possible that the new code has higher MCU load as well? I can't think of what else could be causing this about my setup.

dmbutyugin, 2021-09-23:
@KevinOConnor 

> Yes, I agree the new code can introduce slightly higher host cpu load (for the reasons you cite). However, the additional load should be small.

It was hard to estimate the additional load for me precisely. And profiling will likely prohibitively slow down to make profiling practical. I agree that it didn't become prohibitively expensive, but definitely creates more load. I just saw an abnormal spike in the tickets around the time the changes were made, so I suspected that some number of users were affected one way or the other. In general, I can believe that some number of configurations were operating near the edge for whatever reason, and with these changes their config became too slow. For example, I have a camera, but I don't use it during testing (I don't really use it that much in general), but I wonder if I was?

> It's possible some "micro bursts" are occurring - for example during Python garbage collection - but it's really hard to say from the stats. The tests I ran on my printer don't show an issue (rpi3), and it seems your printer doesn't show an issue either.

Yes, I did not run into this issue (not yet, anyway). I have 2 adxl345, one connected to the MCU, and the test for the corresponding axis also succeeded. I will also try to repeat the test on the delta on the weekend, but there I use RPi4, so I don't really expect to run into the issues.
 
> Okay, but I'm not sure how best to make the user facing error more descriptive. The Klipper log should have a detailed description of the issue (the `logging.exception()` call in the code you cite [...]

It's just that connectivity of adxl345 to the board is a pain point to many users (judging from Discord, at least), so the error messaging suggesting the problem _with the connection_ should definitely be more helpful than that no measurements were performed. I see the gist of the problem with reporting the errors now though. But it could be possible to capture the error status within ADXL345QueryHelper in some form and then report that status instead (or in addition to) the generic error message. One possibility to capture it would be to add a new method like `set_error_status` to the InternalDumpClient/web_request in addition to the `send` method and call it in APIDumpHelper as appropriate. Then ADXL345QueryHelper could expose the error status in via its own method. Perhaps the API server could also benefit from that?

@charlespick 
> @KevinOConnor Is it possible that the new code has higher MCU load as well? I can't think of what else could be causing this about my setup.

I think it does, but to a much lesser degree (and that small increase could only happen if your adxl345 is connected to the MCU). You need to think what could be causing an additional load on your system. Perhaps streaming a video from webcam during the test? Something else? To get a bit of an idea, you could SSH to your RPi, run `top` in the shell, start the resonance testing, and see which processes consume CPU (and how much).

KevinOConnor, 2021-09-24:
> Is it possible that the new code has higher MCU load as well?

Not that I know of.  If anything, it should have lower MCU load due to being more bandwidth efficient.

> I can't think of what else could be causing this about my setup.

I reviewed your logs and it looks to me that you've run out of bandwidth between your "arduino mcu" and the host.  This leads to dropped messages and eventually the host reports a failure.  As a guess, this used to "work" before as a quirk of how messages get dropped.

Try setting `rate: 1600` in the adxl345 printer.cfg section.  That will notably reduce the bandwidth (at the cost of less accelerometer samples).

If you want to experiment, you could increase `tx_bufsize[96]` to `tx_bufsize[192]` at the top of src/generic/serial_irq.c (then do a recompile and flash).  That may help improve results when using the default 3200 rate.  But note that you're likely better off with a rate of 1600 then having lots of drops.

-Kevin

KevinOConnor, 2021-09-24:
> I agree that it didn't become prohibitively expensive, but definitely creates more load.

FWIW, it's actually less load overall - in the sense that it avoids the huge processing spike at the end of the test.  That is, the system load should be lower overall.  It should also be less load on the message reading threads in Klipper.  It is true that it is slightly more load on the Klipper main thread.

> It's just that connectivity of adxl345 to the board is a pain point to many users (judging from Discord, at least), so the error messaging suggesting the problem with the connection should definitely be more helpful than that no measurements were performed.

Ah, yeah, I missed that - all errors are now going only to the log.  I agree this should be changed.

-Kevin

charlespick, 2021-09-24:
interesting, I'll do some bandwidth testing on the cable and try a different board. how much bandwidth is necessary? it's a separate Arduino UNO entirely so it's entire usb 2.0 bus should be dedicated to the accelerometer

KevinOConnor, 2021-09-24:
A 250000 baud serial link has 25000 bytes per second (in both directions).  At 3200 samples per second, the old `Klipper` adxl345 code will generate ~23500Bps from mcu to host (the exact amount varies by the actual sampling rate of the accelerometer which can vary += a few percent).  The new Klipper mcu code will generate ~19500Bps.

Your log shows retransmits and invalid_bytes - which should not occur on a clean connection (irrespective of speed).  Could be line noise between usb-to-serial chip and atmega328p.  Could also be that the usb-to-serial chip on the board is overloaded at these speeds.  Seems unlikely to be an issue between host and usb-to-serial chip.

-Kevin

dmbutyugin, 2021-09-26:
> FWIW, it's actually less load overall - in the sense that it avoids the huge processing spike at the end of the test. That is, the system load should be lower overall. It should also be less load on [...]

Yes, that's actually a very good thing. I thought that it may be a bit more work now to pack the data, but I guess these are all bit operations and are not expensive at all.

> Ah, yeah, I missed that - all errors are now going only to the log. I agree this should be changed.

BTW, changing the error handling could also allow us to bail out in `TEST_RESONANCES` earlier upon detection of an error - right now on the connection failure it will run the test only later to conclude that the data was not measured. I want to fix that too, but I wanted to reach some agreement on how to deal with the communication errors.

KevinOConnor, 2021-09-26:
> BTW, changing the error handling could also allow us to bail out in TEST_RESONANCES earlier upon detection of an error - right now on the connection failure it will run the test only later to [...]

Oops - it was intended to work that way originally, but there was a minor error in the code preventing errors from propagating.  I've committed a fix (commit 1eecd412).

Hopefully that will make errors more transparent.

-Kevin