Klipper3d/klipper #4682: Wrong automatic PID calibration?

robustini, 2021-09-11:
I have found that unfortunately the automatic calculation of the PID is wrong in many situations, or rather it gives values that do not maintain an accurate hotend temperature.
These are the PIDs found by the calibration:

pid_kp = 18.441
pid_ki = 0.640
pid_kd = 132.774

With these values at 200° I have an excursion of +/- 0.7 °.
These are PIDs that I entered manually instead:

pid_kp = 10
pid_ki = 0.7
pid_kd = 100

With these manual values at 200° I have an excursion of +/- 0.1 °.
The automatic calibration at least here fails with a 24V heating cartridge.
I tried to play with max power but the results after calibration are always inaccurate. 

[klippy.zip](https://github.com/KevinOConnor/klipper/files/7148378/klippy.zip)

robustini, 2021-09-12:
The pictures of the charts show the huge difference.
To interpret them correctly look at the temperature scaling on the left. 

PID detected with calibration

[image]

PID entered manually

[image]

keleticsaba, 2021-09-13:
As new user, i can agree. After marlin, starting of the print takes minutes to target temperature. Also pid values differ marlin pid tuning values. Increasing the smoothing time to 2 or 3 seconds helps a bit. However what I get is overshooting the target temperature, not fluctuating over target. 24V system, with mega2560.

keleticsaba, 2021-09-14:
Update:
Using lower target temperature with the calibration made a good results. (The Klipper tutorial uses 170C.)
PID_CALIBRATE HEATER=extruder TARGET=120 >>> Now my printjob at 245C starts instantly, and temperature is rock solid.

robustini, 2021-09-14:
Already tried, 

> 
> 
> Update:
> Using lower target temperature with the calibration made a good results. (The Klipper tutorial uses 170C.)
> PID_CALIBRATE HEATER=extruder TARGET=120 >>> Now my printjob at 245C starts instantly, and temperature is rock solid.

Already ready several times, the described problem remains.

KevinOConnor, 2021-09-14:
You should upgrade to the most recent version of the code, as changes to smooth_time have improved this behaviour.  Otherwise, you'll need to reach out the to the Klipper community via Discord or Discourse to see if someone can help.

-Kevin

closed 2021-09-16 by github-actions[bot]