Klipper3d/klipper #4685: Unable to add controller_fan when extruder is not defined

JVdbKtr, 2021-09-13:
I recently ran into the issue where I was building a config for a custom machine during which I have completed setting up the steppers and fans, but had not yet started to work on extruder config which throws the following error: "Unknown heater 'extruder'".
This appears to be the case since controller_fan.py adds 'extruder' to the heaters list without checking if it exists or not.

My current understanding is that there is no requirement to have an extruder defined, so I believe controller_fan.py should validate the existence of heaters from the config if none are defined in the fan config.

I can see two situations where this could be relevant:

- People building configs and setting up fans before they set up extruders
- CNC or Laser machines which do not have heaters but do potentially have controller fans

Note: klippy.log contains errors from many iterations of trying to work around this issue in config - always unsuccessfull.
[klippy.log](https://github.com/KevinOConnor/klipper/files/7152109/klippy.log)

VIXIVIXIV, 2021-09-13:
I may have a github commit that solves this, though I'm not used to contributing to the main repo. Essentially I added an extra config option to just bypass monitoring the heaters

VIXIVIXIV, 2021-09-16:
It's an optional extra parameter, iirc you can still specify steppers with use_heater set as false