Klipper3d/klipper #4690: Homing - "MCU 'mcu' shutdown: Timer too close" - CTC13 (Anet A8 Board)

stevencole2305, 2021-09-14:
When trying to perform a homing action the Z axis moves 10mm then throws a Timer too close error

Klipper is running on a Raspberry Pi 3B

Unfortunately unable to use printer due to this issue, tried homing both a single axis and all axis but with the same result

Any help would be appreciated

[klippy.log](https://github.com/KevinOConnor/klipper/files/7160307/klippy.log)

Koenzo, 2021-09-14:
Hey, 

I have to same problem on my Voron with BTT octupus TMC2209 drivers(all axis). reflashed mainsail and made a new firmware.bin file and changed a main power(24v) to a lab PSU but without any results. 

The problem occurs in a lot of different ways as shown on the the topics but this one from stevencole2305 is exactly the same as mine. 

After z-hopping the error occurs when the XY has to home.

Thanks in advance

Koenzo, 2021-09-14:
Okay, I got mine fixed after some hours. (fix was found on discord (same problem as kulaan). 

The problem was human. *as always.

The solution is to reflash the firmware.bin with the correct clock frequency

The fault occurs because the default is 8 Mhz and can be set if you go deeper in the setting:

[image]
[image]

@KevinOConnor  maby its wise to adress this(that its default on 8Mhz).

greetz

closed 2021-09-16 by github-actions[bot]