Klipper3d/klipper #4697: 'manual_stepper' [STOP_ON_ENDSTOP] causes kipper to remain "busy" after hitting endstop

Nickperillo, 2021-09-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/7178226/klippy.log)

When I send this command:
`MANUAL_STEPPER STEPPER=clay_ram_stepper ENABLE=1 SET_POSITION=100 SPEED=0.5 ACCEL=10 MOVE=0 STOP_ON_ENDSTOP=1`
It works fine and the motor stops on the endstop, but then Klipper remains busy and I can't issue any more commands without restarting. 

Also when I try to `SET_POSITION` higher than 100 Klipper shuts down with this error: `MCU 'eboard' shutdown: Timer too close`
I want to set the position to the maximum travel of the axis which is 500.

I'm trying to set up an external ram extruder for printing clay. It needs to be able to home itself for refilling the tank. This is why I'm attempting to set it up as a `manual_stepper` so I can easily write a macro. 
I would prefer to use `extruder_stepper` but this doesn't allow for homing and I haven't figured out a workaround.

Any ideas? it seems like a fairly basic command so I'm not sure where I'm going wrong.

EtteGit, 2021-09-24:
The manual steppers homing does indeed "last" for whole move duration. It's probably due to how it's in sync with the toolhead moves (not an expert on that at all). Not sure how easy it is to fix that.

Note that : 
- you don't need to do the "ENABLE = 1", manual steppers will be turned on automatically if you ask them to move.
- I would do the SET_POSITION once the homing move is done. Something like that : 
     1 - MANUAL_STEPPER STEPPER=clay_ram_stepper SET_POSITION=0 
     2 - MANUAL_STEPPER STEPPER=clay_ram_stepper SPEED=0.5 ACCEL=10 MOVE=-600 STOP_ON_ENDSTOP=1
     3 - MANUAL_STEPPER STEPPER=clay_ram_stepper SET_POSITION=600

closed 2021-11-06 by github-actions[bot]

mkutscher, 2022-02-22:
@Nickperillo I have the same issue with homing a height ajustable wiper...
have you found a solution or a workaround?

Nickperillo, 2022-02-22:
> @Nickperillo I have the same issue with homing a height adjustable wiper... have you found a solution or a workaround?

@mkutscher No I haven't yet. I currently manually jog my motor close to the position and then run a macro to home it from there.
It's not a nice solution.