Klipper3d/klipper #4699: Renaming ACTIVATE_EXTRUDER results in an internal error

Deutherius, 2021-09-17:
If I use the `rename_existing` block to override a gcode command, say `BED_MESH_CALIBRATE`, everything works fine, i.e.

```
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: WHATEVER
gcode:
  M117 Nothing should happen
```
Calling `BED_MESH_CALIBRATE` now writes to display. Calling `WHATEVER` takes a bed mesh.

If I rename `ACTIVATE_EXTRUDER`, however, with the same code, i.e.
```
[gcode_macro ACTIVATE_EXTRUDER]
rename_existing: WHATEVER
gcode:
  M117 Nothing should happen
```
calling `WHATEVER` shuts klipper down with an internal error instead of activating an extruder (or complaining that no EXTRUDER was specified). Replicated the error on 2 independent machines, one klippy log included.

[klippy.log](https://github.com/KevinOConnor/klipper/files/7187140/klippy.log)

KevinOConnor, 2021-09-21:
This looks like a dup of #3497.

-Kevin

Deutherius, 2021-09-21:
Correct, no idea how I missed it.

closed 2021-09-21 by Deutherius