Klipper3d/klipper #4703: Help installing on Neptune 2

aSocialMenace, 2021-09-17:
First off if like to commend the developers on the excellent documentation they have provided! I found it very useful and informative in trying to solve my problem. However I was unsuccessful say here's what I'm dealing with..

I'm having trouble installing klipper on my elegoo Neptune 2 with the stock zpn Robin nano 1.2 board. 

I compiled the firmware using the recommend settings in the example file as there is such a file for the Neptune 2. And I've read various posts from others online who have used these settings successfully. But when I load the sd card with the elegoo.bin file into the printer it boots up with a TFT status bar that reaches 100% quickly but never goes away. After I reboot the printer the screen just says booting. And that never goes away. 

Now if I try to connect klipper while the TFT status bar is up I get nothing. but after I power cycle the ptinter the screen says booting.. I get the messages 

connected to host via /temp/printer @115200kbs 

and 

Firmware version v0.9.***

But the printer doesn't respond to any gcode and if I hit the get status button I get the messages 
mcu 'mcu': Unable to connect 

and 

mcu 'mcu'  Unable to connect Once the underlying issue is corrected, use the "FIRMWARE_REATART" command to restart firmware, reload the config, and restart host software. Error configuring printer Klipper state: Not ready. 

So I've tried recompiling the firmware to match the baud rate octoprint is telling me the printer is connected to (115200bps) and that didn't work. I've also tried disconnecting the touchscreen as I've heard that can cause issues installing klipper on some printers. However without the touchscreen the printer wouldn't even install the firmware which I could tell because it didn't change the name of the file to elegoo.cur . I've also tried using a brand new USB cable and nothing has worked. 

I kinda wonder if the firmware isn't installing correctly on the printer. As I just don't think the TFT loading screen and the booting screen should be stuck on. That's just not the behavior I expected. Also I never had to power cycle either of my ender 3s to get klipper to connect after the firmware was installed.. (oh yeah.. this isn't my first time installing klipper 😎)  But then again I also suspect that perhaps the example config file has something wrong with it. perhaps the make config instructions. Specifically something with the communication configuration.. or perhaps something in the code. like the code to disable the screen. But again.. I've seen posts from other users in various forms and on Reddit that claim these settings work for them on their Neptune 2... 

[klippy.txt](https://github.com/KevinOConnor/klipper/files/7188460/klippy.txt)

[image]

Kinda stumped here.. any thoughts?

aSocialMenace, 2021-09-17:
Sorry.. I probably should explain I created a simple http server while shelled into my octoprint server to download the kilppy.log file to my phone as that what I used to post here.. I needed to open the file on my phone to make sure I downloaded it correctly but I didn't have any apps that would open a .log file so I had to rename it .txt so klippy.txt is kkippy.log

aSocialMenace, 2021-09-17:
Ok.. sorry for being a space case.. here is my config file.. this time I remembered to re re name it back to printer cfg

closed 2021-09-20 by github-actions[bot]

jonathanrobichaud4, 2022-01-02:
@aSocialMenace did you ever figure out the issue?? Im stuck too