Klipper3d/klipper #4706: Move out of range: 250.012 0.000 2.000 [0.000]

RaoulFabisch, 2021-09-19:
When starting a print I often get this error: Move out of range: 250.012 0.000 2.000 [0.000] 

Sometimes it starts when restarting the print often enough, but I guess that is not how it should behave. Hope that there is a quick fix for this problem.

[klippy.txt.txt](https://github.com/KevinOConnor/klipper/files/7191515/klippy.txt.txt)

dmbutyugin, 2021-09-19:
When exactly does this happen? My 2 cents: in the config you have
```
[stepper_x]
endstop_pin = tmc2209_stepper_x:virtual_endstop
position_endstop = 250
position_max = 250
```
The toolhead appears to be outside of the allowed range of [0, 250] on X axis at 250.012. But I assume it is not how your gcode commands it? Does it happen after homing, or bed meshing? If it happens just mid-print, I'd double check gcode.

florian-senn, 2021-09-20:
Maybe forgot to G21 set metric values?

KevinOConnor, 2021-09-21:
Best to raise this on Klipper Discord or Discourse.

-Kevin

schnello, 2021-09-22:
Hello.

[250.012 0.000 2.000 [0.000]]

This happens if you use sensorless homing and you start the homing process if the toolhead is parked directly at the frame. 

Regards

embedded-iot4me, 2021-09-22:
Hello, 

after updating to the latest FW (v0.9.1-780-g7a02cde4), I got the **move out of range -0.013 -0.013 15.000 [0.000]** message by sending a "simple" G28 (home) command. I was able to mitigate this behavior by changing the value for position_endstop: 0 to 1 (below: stepper_x / stepper_y in printer.cfg)

best regards,
Ulf

closed 2021-09-24 by github-actions[bot]