Klipper3d/klipper #4710: Heater extruder not heating at expected rate

eboston, 2021-09-20:
I am having a problem with a print failing with "Heater extruder not heating at expected rate" at the same point in the print.  I have done a PID calibration with no luck.  Checked that the part cooling fan does not effect the temperature.  Resliced the file with the print stopping at the same point.  Because it happens at the same space, I don't think it is an intermitted hardware problem.

What else can I try to fix this problem.  I think I attached the log file.

eboston, 2021-09-20:
[klippy (3).log](https://github.com/KevinOConnor/klipper/files/7199034/klippy.3.log)

Attempt 2

behold81, 2021-09-22:
I just upgraded to 
klipper
v0.9.1-784-g0b918b35
and
moonraker
v0.7.1-56-g2447cca

I get the same error. I have traced the issue to Klipper not enabling 24V to the HeaterBed from the main board itself. 

I am trying to identify if the board is actually a fault but I believe its not.

eboston, 2021-09-22:
> I just upgraded to
> klipper
> v0.9.1-784-g0b918b35
> and
> moonraker
> v0.7.1-56-g2447cca
> 
> I get the same error. I have traced the issue to Klipper not enabling 24V to the HeaterBed from the main board itself.
> 
> I am trying to identify if the board is actually a fault but I believe its not.

I am running klipper v0.9.1-779-g59bd51d2 and moonraker v0.7.1-48-ge92da4c but am showing the upgrades are available.  So it looks like it happened earlier.  Is there a way to go back to earlier versions of klipper?

closed 2021-09-24 by github-actions[bot]

eboston, 2021-09-24:
So where are we suppose to report bugs with Klipper?

Gilabite, 2021-09-24:
Why is it a bug and not user error? Is your thermistor a slice engineering thermistor? If it is try using SliceEngineering 450 if its a 450c version or ATC Semitec 104GT-2 for the 300c version for the sensor_type instead of your custom table and see if that helps.