Klipper3d/klipper #4712: ADXL345 on Raspberry Pi Pico stopped after specific commit

glabifrons, 2021-09-21:
Using an ADXL345 on a Raspberry Pi Pico via USB worked up until (and including) commit e03e0e9dac5694b35b81726bd2ed18f1a67b0a26
This combination fails specifically on commit 8cfaed441d1ec295c8c4be8e0733099fe3f3395a and all later commits that I tried (including the one from 9/16/2021 and the ones from 9/4/2021).

I triggered the failure by sending "ACCELEROMETER_QUERY".
The error in the klippy.log is "Invalid adxl345 id (got f2 vs e5)".

For each test, I rebuilt the firmware for and reflashed both the primary MCU and the Pi Pico, then rebooted the host Raspberry Pi to ensure everything was on the exact same version of firmware.

Please find attached klippy log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/7200481/klippy.log)

If needed, I can revert to the functional commit and upload the working klippy.log file as well.
I can also repeat the same sequence of tests on a different printer (I have several running Klipper, all different models and styles).
Please let me know if you need any more information from me.

KevinOConnor, 2021-09-21:
That's unfortunate and I don't know why that would be.  If you're working to find out why you can use this issue to relay your status.  If you're looking for someone else to help with this then you should reach out to the Klipper community on either Klipper Discord or Klipper Discourse.

-Kevin

glabifrons, 2021-09-22:
I finally have it figured out.

Prior to commit 8cfaed441d1ec295c8c4be8e0733099fe3f3395a you could use ACCELEROMETER_QUERY by itself and it would find the ADXL345 chip (per https://github.com/KevinOConnor/klipper/blob/master/docs/Measuring_Resonances.md ).
After commit 8cfaed441d1ec295c8c4be8e0733099fe3f3395a you **must** add "CHIP=default" for ACCELEROMETER_QUERY to not fail.
After commit 404b64fd627e04f68f64db4fef605fae1a18a747 the parameter is changed to "CHIP=adxl345", but _appears to no longer be mandatory_.

However, even in current versions ACCELEROMETER_QUERY gives an error the **first time** you execute it after a firmware restart, but succeeds every time after (verified on multiple printers with completely different hardware, but the same versions of Klipper).

```
Send: ACCELEROMETER_QUERY
Recv: !! No adxl345 measurements found
Send: ACCELEROMETER_QUERY
Recv: // adxl345 values (x, y, z): 764.918700, 35033.276460, -15221.882130
```

[klippy.log](https://github.com/KevinOConnor/klipper/files/7207085/klippy.log)

closed 2021-09-24 by github-actions[bot]