Klipper3d/klipper #4714: Feature suggestion: Faster multiple probe sampling

danijoo, 2021-09-22:
When the stdev between multiple probing samples is to high, klipper discards all previous samples and reruns the whole probing (controlled via `samples` and  `sample_tolerance`). Marlin uses a different approach: it has the option `EXTRA_PROBING` to discard outliers and do additional probing runs when the samples show a strong deviation. This is much faster than discarding all previous samples. Therefore, I suggest to adapt a similar workflow for multiple probing, i.e.:

1) Probe `N` samples
2) if tolerance > `samples_tolerance`, discard the one with the biggest deviation from the mean and sample a single new point (instead of `N` new samples).
3) Redo step 2 until `samples_tolerance_retries` is exceeded or the tolerance criterion is met.

I am willing to implement this and create a PR, but wanted to test the waters first. What do you think?

KevinOConnor, 2021-09-26:
FYI, there was a discussion on this at https://github.com/Klipper3d/klipper/pull/3828

-Kevin

danijoo, 2021-10-02:
@KevinOConnor so are you open for a PR regarding this?

I think my approach does not have the same problems you discussed in #3828 since it does not depend on the order.

closed 2021-11-13 by github-actions[bot]