Klipper3d/klipper #4729: 3DTouch on Creality 4.2.2 Board always reports open

XoBoR, 2021-09-27:
Hi,

I'm trying to get my 3D Touch (BLTouch Clone) to work with Klipper, with no success.

I followed the Instructions on BLTouch, but QUERY_PROBE always returns open.

The only time it reports Triggered is when the probe is in Error mode (blinking red).

I validated that the 3DTouch is working by flashing back my previous Marlin setup, and it works great there.

this is the current configuration, but I think i've tried most if not all possible combinations:

[bltouch]                             # enable for BLTouch - fast-mode
sensor_pin: ^PB1
control_pin: PB0
#pin_move_time: 0.680
#stow_on_each_sample: True
probe_with_touch_mode: True
#set_output_mode: 5V
#pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: True
x_offset: -41                          # modify as needed for bltouch location
y_offset: -3                          # modify as needed for bltouch location
z_offset: 0.0                         # modify as needed for bltouch or run PROBE_CALIBRATE
speed: 10
samples: 3
sample_retract_dist: 5.0              # Can be set lower, example 2.5 depending on height of bltouch from bed
lift_speed: 40
samples_tolerance_retries: 3
speed: 10
samples: 2

any hints would be appreciated, since I really ran out of ideas what could be wrong

XoBoR, 2021-09-29:
Thanks to JamesH from the Discord community I just tried homing and found out, that the Probe actually Works.

It's just the Test that doesn't work for the 3D Touch (https://www.klipper3d.org/BLTouch.html#initial-tests this one).

I spend hours on this, and didnt even think about just trying to home, since i thought I will just crash my nozzle into my bed, so i really think either the test should be corrected to work with Clones, or there should be a note that states that it doesnt work with Clones.

closed 2021-10-02 by github-actions[bot]

grantemsley, 2021-12-13:
Please reopen this - updating the documentation as XoBoR noted would be greatly appreciated. I too wasted several hours on this. 

> The next step is to confirm that the sensor pin is working correctly. Run BLTOUCH_DEBUG COMMAND=pin_down, verify that the pin moves down, run BLTOUCH_DEBUG COMMAND=touch_mode, run QUERY_PROBE, and verify that command reports "probe: open". Then while gently pushing the pin up slightly with the nail of your finger run QUERY_PROBE again. Verify the command reports "probe: TRIGGERED". If either query does not report the correct message then check your wiring and configuration again. At the completion of this test run BLTOUCH_DEBUG COMMAND=pin_up and verify that the pin moves up.

My clone touch sensor never reports probe: TRIGGERED, but homing works perfectly. Adding a note there that some clones will not report triggered, and to cautiously move on to the "homing" by touching it with your finger test should save others a few hours too.