Klipper3d/klipper #4745: "Extrude below min temp"-Error without a reason

ElAndree, 2021-10-01:
Hello,

im experiencing a extrude below min temp error without a reason.
The extrude_below_mintemp is set to 170.
My printer is randomly stopping printing with the error, but keeps the heater on and as you can see in the screenshot the temperature curve does not show any anomaly.
I've also attached my log files.

Thank you!

- Andre

[image]
[klippy-error.txt](https://github.com/Klipper3d/klipper/files/7265160/klippy-error.txt)
[moonraker.txt](https://github.com/Klipper3d/klipper/files/7265164/moonraker.txt)

ElAndree, 2021-10-03:
Hello @KevinOConnor,

I think this is a misunderstanding.
I don't need any help with my printer etc. but I think I have a clipper bug (there are also several errors in the log) that lead to a print abort and strange behaviour as described above.
A bug is also on the topic of I2C and BME280 - a similar issue is also open, the one also has similar if a bit different problems with print aborts. Maybe you could look at it again, I think I'm in the right place, it's really about a code bug.
Thank you very much in advance.

Translated with www.DeepL.com/Translator (free version)

KevinOConnor, 2021-10-03:
Bug or not, someone needs to investigate, determine root cause, and implement a solution.  If you're actively doing that work then you can keep this ticket open to share the results of your work.  If you're looking for help with it, you'll need to reach out to the Klipper community.

-Kevin

closed 2021-10-03 by ElAndree

fleetofgreen, 2022-01-31:
Did you ever find a fix for this? I am having the same issue.