Klipper3d/klipper #4777: Problem with klipper today 2021/10/07

kapoira, 2021-10-07:
This morning i see that klipper have update a new release (i'm using moonraker)

i have update klipper reset the printer and all seems ok

Now i was redoing the first layer and need to do the paper test 

PROBE_CALIBRATE

i do the ACCEPT when i think that the paper is near of

 the fusor

but i get and error when i have to 

SAVE_CONFIG

[image]
(i have atached the messaged that moonraker is saying) 

yesterday i don't have this problem (any idea?) (sorry for my english not my first language)

Dragi2k, 2021-10-07:
i have the same problem after updating

E8geni, 2021-10-08:
There is a problem
`Internal error on command:"SAVE_CONFIG"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 245, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 415, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 200, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 120, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/configfile.py", line 376, in cmd_SAVE_CONFIG
    data = regular_data.rstrip() + autosave_data
UnicodeDecodeError: 'ascii' codec can't decode byte 0xd0 in position 2901: ordinal not in range(128)
MCU 'z' shutdown: Command request`
[image]

Sineos, 2021-10-09:
See here for a work around: https://klipper.discourse.group/t/klipper-error-ascii-codec-cant-decode/1049/14?u=sineos

E8geni, 2021-10-09:
> See here for a work around: https://klipper.discourse.group/t/klipper-error-ascii-codec-cant-decode/1049/14?u=sineos

Thanks!
I returned to version v0.9.1-616-g28f60f7e
Everything works and nothing needs to be fixed.