Klipper3d/klipper #4779: UART w TMC2208, 2209 on Mks Robin Nano v1.2 doesn't work because of too much TMC_BAUD_RATE

strobe, 2021-10-07:
I have  MKS Robin Nano v1.2 with all 5 TMC2209 installed with soldered jumper wires (as in this video: https://www.youtube.com/watch?v=7ShcFKXrVHo) (fifth slot previously has TMC2208, I switched to TMC2209 in attempt to understand the issue). 
I used that hardware configuration with Marlin and it works well. But after I switched to Klipper I got  
`' Unable to read tmc uart 'extruder1' register IFCNT'` errors any time then I tried to activate fifth slot.
Then I found this PR https://github.com/Klipper3d/klipper/pull/4505 where @katodo mention that some Pins on this board like PA5 (which I'm using) can't work with that baud 40000 rate.
So I tried to change `TMC_BAUD_RATE` in `klippy/extras/tmc_uart.py` to 19200 and after re-flashing firmware my second 
extruder motor start working as expected. So it looks like extract my problem.

I don't know how this issue should properly addressed - should Baud Rate be configurable or is it something else is possible to do, might be additional information in error string or in documentation?

Just in case, here pins definitions from Marlin:

from _pins_MKS_ROBIN_NANO.h_:
```
//
// TMC 2209 UART MODE via WIFI pins
//

#if HAS_TMC_UART
#define X_SERIAL_TX_PIN PA10
#define X_SERIAL_RX_PIN PA10

#define Y_SERIAL_TX_PIN PA9
#define Y_SERIAL_RX_PIN PA9

#define Z_SERIAL_TX_PIN PC7
#define Z_SERIAL_RX_PIN PC7

#define E0_SERIAL_TX_PIN PC13
#define E0_SERIAL_RX_PIN PC13

#define E1_SERIAL_TX_PIN PA5
#define E1_SERIAL_RX_PIN PA5

#define TMC_BAUD_RATE 19200
#endif // TMC2208 || TMC2209

```

from pins_MKS_ROBIN_NANO_common.h:
```
//
// Steppers
//
#define X_ENABLE_PIN                        PE4
#define X_STEP_PIN                          PE3
#define X_DIR_PIN                           PE2

#define Y_ENABLE_PIN                        PE1
#define Y_STEP_PIN                          PE0
#define Y_DIR_PIN                           PB9

#define Z_ENABLE_PIN                        PB8
#define Z_STEP_PIN                          PB5
#define Z_DIR_PIN                           PB4

#define E0_ENABLE_PIN                       PB3
#define E0_STEP_PIN                         PD6
#define E0_DIR_PIN                          PD3

#define E1_ENABLE_PIN                       PA3
#define E1_STEP_PIN                         PA6
#define E1_DIR_PIN                          PA1
```

th0m4sek, 2021-10-07:
I,m using PA13 without issues.
It's located on pin 1 of J11 programming header (SWD) and has pull up resistor.

closed 2021-11-19 by github-actions[bot]

SpaceEggs24, 2021-12-26:
> I,m using PA13 without issues. It's located on pin 1 of J11 programming header (SWD) and has pull up resistor.

Sorry to open this thread back up but can you tell me where this pin is located exactly? I can't seem to make sense out of the schematic..

SpaceEggs24, 2021-12-26:
> @th0m4sek thanks that works for me.
> 
> @KevinOConnor does it wort to add note about it to [TMC_Drivers Common Questions](https://github.com/Klipper3d/klipper/blob/master/docs/TMC_Drivers.md#i-keep-getting-unable-to-read-tmc-uart-stepper_x-register-ifcnt-errors) ?

[image]

Is it supposed to be this one?

th0m4sek, 2021-12-26:
Yes

alessandro111297, 2022-05-08:
@th0m4sek Sorry, can you explain the procedure with PA13? You soldered UART pin of TMC2209 and plugged on pin 1 of J11, and in Klipper "uart_pin: PA13", right?

th0m4sek, 2022-05-08:
`[tmc2209 stepper_x]
uart_pin: PA13
run_current: 0.8
uart_address: 0

[tmc2209 stepper_y]
uart_pin: PA13
run_current: 0.8
uart_address: 1

[tmc2209 stepper_z]
uart_pin: PA13
run_current: 0.8
uart_address: 2

[tmc2209 extruder]
uart_pin: PA13
run_current: 0.8
uart_address: 3`

and remember to set adresses using ms1 ms2 ms3 jumpers

alessandro111297, 2022-05-08:
@th0m4sek I have only one tmc2209 for extruder, I soldered UART pins of TMC2209 and wired in pin 1 of J11