Klipper3d/klipper #4790: SD Card menu option on display causes unhandled exception

bonjipoo, 2021-10-09:
Ticket recommended to be opened by timmit99.  Possible Klipper module error.

Ive fresh installed Klipper on a new printer and all is working. Configured a V0 display for it which is also working however when trying to use the SD Card menu option I get an unhandled exception. See log and screenshot

[image]
.
[klippy-12.log](https://github.com/Klipper3d/klipper/files/7316127/klippy-12.log)

quwahiro, 2021-10-10:
Pls check dir name and file name. Do it include 2byte characters in filename or not?

bonjipoo, 2021-10-10:
So these are the files contained in in the folder and the virtual_sdcard printer config.  Is the ~ symbol the issue?  This is the base configuration from mainsail.  Assume it is correct.

[image]
[image]

bonjipoo, 2021-10-10:
Have you fixed this do you mean?

> I had same problem but update from this morning .....0-39 fix SD card in menus

bonjipoo, 2021-10-10:
Latest update via mainsail fixes this issue.