Klipper3d/klipper #4794: Internal error on command:"SDCARD_PRINT_FILE"

michi6383, 2021-10-11:
Internal error on command:"SDCARD_PRINT_FILE"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 120, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 144, in cmd_SDCARD_PRINT_FILE
    self._load_file(gcmd, filename, check_subdirs=True)
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 187, in _load_file
    gcmd.respond_raw("File opened:%s Size:%d" % (filename, fsize))
  File "/home/pi/klipper/klippy/gcode.py", line 208, in respond_raw
    cb(msg)
  File "/home/pi/klipper/klippy/gcode.py", line 430, in _respond_raw
    os.write(self.fd, (msg+"\n").encode())
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 46: ordinal not in range(128)
Transition to shutdown state: Internal error on command:"SDCARD_PRINT_FILE"

michi6383, 2021-10-11:
Ok. I found a workaround. The problem is the "µ" in the filename. 
Ultiron_Fiberlogy_PETG_Ultiron_120µm_Angled_Foregrip.gcode

If you replace or remove the character it works fine.

drak42, 2021-10-13:
Glad I found this...  Renaming file worked

michi6383, 2021-11-23:
Still existing...

vladimir-poleh, 2021-11-25:
There are two options to work with non-ASCII characters:
1. Use release version v0.10
2. Switch to Python 3

soxhi8, 2021-11-25:
> There are two options to work with non-ASCII characters:
> 
>     1. Use release version v0.10
> 
>     2. Switch to Python 3

Is Klipper fully supported on Python 3 yet? According to the author of KIAUH it does not really sound like it:
https://github.com/th33xitus/kiauh/issues/136#issuecomment-963480182

mvoss96, 2022-01-15:
Can confirm this error still persists.
I wanted to print a file containing the german letter "ä" and it crashed the printer

Sineos, 2022-01-15:
As long as the issue is not tagged as "resolved" it will continue to exist :wink:

CircuitGuy, 2022-02-25:
This issue got marked as a duplicate of #4959, but I think that's incorrect.

I have #4959 exactly. The same stack trace, etc. Which is not the stack trace of this issue.

Klipper seemingly won't print any files from the LCD with a space in it. For example ```Calibration Cube.gcode``` won't print, but ```Calibration_Cube.gcode``` will.

tbb98, 2022-05-13:
Strange. It has happened to me multiple times.
Spaces in folder or filenames and non-ASCII characters (íöüóőúéáű in my case) are working for weeks/months and after a while, it stops working and I have to rename them. (Moonraker + Mainsail)

[image]

RPhilbo, 2022-09-14:
> Strange. It has happened to me multiple times. Spaces in folder or filenames and non-ASCII characters (íöüóőúéáű in my case) are working for weeks/months and after a while, it stops working and I have to rename them. (Moonraker + Mainsail)
> 
> [image]

Today, for me this happend also again with an german "ü".

Everything up to date.

webhive, 2022-10-01:
Same thing. All software is up to date.

jperrin72, 2022-10-06:
got the issue today with a French word containing accentued characters (é)

Piezoid, 2022-10-07:
I'll said it: this issue is waiting for python2 support to be removed.

If you want to keep a legacy python2 installation you need to stay on a stable release that support it. (v0.10)

While trying to have a look I realized that I can no longer easily install python2 on my dev machine and pi. This is probably the reason that no one will tackle this, even if [the fix might be easy to implement](https://github.com/Klipper3d/klipper/commit/adeb869f560ed876ef44f4c950e5798090948276#commitcomment-58632845).

I've been running klipper on python3 since April 2021 with minimal modifications. Since then, support has been added to klipper master in the wake of the v0.10 release. As of October 4, 2021, klipper runs without modification on python3. I believe it will be made official with the v0.11 release.
MainsailOS is defaulting to Python3 since August 2021 and KIAUH added experimental support in v4. By now, it should be easy to use a python3 installation. I think that this should be recommended when following the *development* updates of klipper.

DX94-Quas, 2023-01-16:
I got a failure when using uppercase letters, got this on the log:
```
Traceback (most recent call last):
  File "~/klipper/klippy/extras/virtual_sdcard.py", line 178, in _load_file
    fname = files_by_lower[fname.lower()]
KeyError: 'ce3_polea mod 5.gcode'
Unable to open file
```
the original name of the file is `CE3_polea mod 5.gcode`, I am using KIAUH to install, with python3

mbravorus, 2023-01-24:
Same occurrence - sliced an STL from a file which had Chinese characters in the name, and this happens:

```
Internal error on command:"SDCARD_PRINT_FILE"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 150, in cmd_SDCARD_PRINT_FILE
    self._load_file(gcmd, filename, check_subdirs=True)
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 187, in _load_file
    gcmd.respond_raw("File opened:%s Size:%d" % (filename, fsize))
  File "/home/pi/klipper/klippy/gcode.py", line 224, in respond_raw
    cb(msg)
  File "/home/pi/klipper/klippy/gcode.py", line 446, in _respond_raw
    os.write(self.fd, (msg+"\n").encode())
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe6 in position 12: ordinal not in range(128)
Transition to shutdown state: Internal error on command:"SDCARD_PRINT_FILE"
Dumping gcode input 0 blocks
```

in fact, this is not the only place where things break - even after changing the gcode file name on the rPi host, it still refused to print, bailing out like this:

```
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 425, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 216, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 198, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 135, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 150, in cmd_SDCARD_PRINT_FILE
    self._load_file(gcmd, filename, check_subdirs=True)
  File "/home/pi/klipper/klippy/extras/virtual_sdcard.py", line 187, in _load_file
    gcmd.respond_raw("File opened:%s Size:%d" % (filename, fsize))
  File "/home/pi/klipper/klippy/gcode.py", line 224, in respond_raw
    cb(msg)
  File "/home/pi/klipper/klippy/gcode.py", line 446, in _respond_raw
    os.write(self.fd, (msg+"\n").encode())
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe6 in position 12: ordinal not in range(128)
Requested toolhead position at shutdown time 77291.848183: (5.0, 170.0, 109.68693354800938)
```

which from Fluidd looks like file is accepted and starts printing, but nothing is actually happening on the printer side

this is with

klipper v0.11.0-66-g4671cf2d
fluidd v1.23.0
moonraker v0.7.1-806-ga154c5f

```
 $ cat /etc/debian_version
10.13
$ uname -a
Linux octopi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
```

the system has fully functional python3, but the default one (and the one in klippy-env) is still python 2.7.16

Neko-vecter, 2023-04-05:
I think if the file name that contains the characters that not in `ascii` will cause this issue.
Currently possible solutions is migration Klipper to python3.

Murat65536, 2023-05-06:
How do I change from Klipper Python 2 to Klipper Python 3?

Piezoid, 2023-05-07:
> How do I change from Klipper Python 2 to Klipper Python 3?

@Murat65536 
Since you're asking, the smoothest course of action for you would likely be reinstalling Klipper using [KIAUH](https://github.com/th33xitus/kiauh) and select Python 3 when asked:
`###### Select Python version:`

The manual way is to recreate Klipper's python virtual environment. Something in the lines of:
```bash
cd
mv klippy-env klippy-env-py2
python3 -m venv klippy-env
./klippy-env/bin/pip install -r ./klipper/scripts/klippy-requirements.txt
sudo systemctl restart klipper
```
This relies on `/etc/systemd/system/klipper.service` invoking the new python interpreter from the environment (`$HOME/klippy-env/bin/python`)
This is highly dependent on how Klipper was installed on your system.

Murat65536, 2023-05-07:
@Piezoid
I attempted to preform the actions but the error is still there. Even when there are no abnormal characters, the error does not go away. Is there anything that I can do?

Murat65536, 2023-05-09:
> > How do I change from Klipper Python 2 to Klipper Python 3?
> 
> @Murat65536 Since you're asking, the smoothest course of action for you would likely be reinstalling Klipper using [KIAUH](https://github.com/th33xitus/kiauh) and select Python 3 when asked: `###### Select Python version:`
> 
> The manual way is to recreate Klipper's python virtual environment. Something in the lines of:
> 
> ```shell
> cd
> mv klippy-env klippy-env-py2
> python3 -m venv klippy-env
> ./klippy-env/bin/pip install -r ./klipper/scripts/klippy-requirements.txt
> sudo systemctl restart klipper
> ```
> 
> This relies on `/etc/systemd/system/klipper.service` invoking the new python interpreter from the environment (`$HOME/klippy-env/bin/python`) This is highly dependent on how Klipper was installed on your system.

@Piezoid
I attempted to preform the actions but the error is still there. Even when there are no abnormal characters, the error does not go away. Is there anything that I can do?

iqubik, 2023-07-16:
Cant print file with russian Exclude object names.
[123_0.6n_PLA_29s.zip](https://github.com/Klipper3d/klipper/files/12064294/123_0.6n_PLA_29s.zip)
Print is stop in start on delta flsun q5.

StefanIGit, 2023-12-29:
Same here with german and named objects from Prusaslicer 2.7.1

```
EXCLUDE_OBJECT_DEFINE NAME=Körper1_stl CENTER=117.500,117.500 POLYGON=[[132.500,132.500],[102.500,132.500],[102.500,102.500],[132.500,102.500]]
```
error from klippy 
```
Starting SD card print (position 0)
virtual_sdcard dispatch
Traceback (most recent call last):
  File "/home/stefan/klipper/klippy/extras/virtual_sdcard.py", line 264, in work_handler
    self.gcode.run_script(line)
  File "/home/stefan/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/stefan/klipper/klippy/gcode.py", line 213, in _process_commands
    self._respond_error(str(e))
UnicodeEncodeError: 'ascii' codec can't encode character u'\xf6' in position 47: ordinal not in range(128)
Exiting SD card print (position 55053)
```
[Klippy.log-and-gcode.zip](https://github.com/Klipper3d/klipper/files/13795561/Klippy.log-and-gcode.zip)

Neko-vecter, 2024-04-21:
This problem occurs if characters that are not in ACSII are used. It can able fix by updating Klipper python version to python3.

tinyboxxx, 2024-05-31:
I think it's solved after reinstalling Klipper using KIAUH and select Python 3.

reitermarkus, 2024-12-11:
Just got the same error while calibrating pressure advance using OrcaSlicer, which generates an object called `Würfel` in its German translation. (See https://github.com/SoftFever/OrcaSlicer/issues/3907).

Can confirm switching to Python 3 fixes this.

KevinOConnor, 2025-02-02:
I'm going to close this as we no longer use the Klipper issue tracker. It is believed this issue is fixed on python3 and there is no plan to fix this on python2.

-Kevin

closed 2025-02-02 by KevinOConnor