Klipper3d/klipper #4812: [BUG] Klipper doesn't move to a valid position, gives error instead

p-v-n, 2021-10-17:
Here is a valid sequence of commands for a 230x230 bed, isn't it?

```
G29
G1 X65 Y115 Z0.03
```

However, I'm getting the following error:

`!! Move out of range: 65.000 115.000 -0.022 [0.000]`

G29 is defined as:

```
[gcode_macro G29]
gcode:
 G28
 BED_MESH_CALIBRATE
 BED_MESH_PROFILE SAVE=p1
 G1 X0 Y0 Z5 F4000
```

Here is the probed mesh (reformatted for better readability):

```
{
  "mesh_max": [215.0, 215.0],
  "z_positions": [
    [ 0.2625,  0.2325,  0.185,  0.125,  0.1025 ],
    [ 0.0425,  0.0625,  0.0675, 0.0825, 0.15   ],
    [-0.0675, **-0.0525**, -0.0175, 0.0575, 0.1825 ],
    [-0.0075, -0.0225,  0.005,  0.0675, 0.195  ],
    [ 0.19,    0.0725,  0.035,  0.04,   0.1425 ]
  ],
  "mesh_min": [15.0, 15.0]
}
```

Notice that the value (in bold above) corresponding to X65 Y115 is negative and it's absolute value is greater that the Z offset requested in the G1 command.
I suspect that there is a check somewhere in code, which ignores the mesh, and therefore giving an error because it thinks that the move would put the nozzle bellow the bed.

p-v-n, 2021-10-18:
Looks like _check_endstops() doesn't account for mesh.

p-v-n, 2021-10-18:
@Gilabite, the default value of position_min is 0. The gcode above commands a move to Z=0.02. How is this a user error?

Gilabite, 2021-10-18:
Its in the documentation https://www.klipper3d.org/Bed_Level.html?h=position_min#choose-the-appropriate-calibration-mechanism

p-v-n, 2021-10-18:
How is that a solution? How is the error correct? This is clearly a bug.

The command was `G1 X65 Y115 Z0.02`. It's clearly valid.

What's the expected behavior?
The head should move to position X=65 and Y=115 and put the nozzle 0.02mm above the bed, correct?
The actual behavior was to give an error - that's not expected and thus this is a bug.

Gcode has no way of knowing how warped the bed is, and it doesn't have to - that's the responsibility of the firmware to use probed mesh and make necessary adjustments, and it seems that the adjustment was calculated correctly, but validation was not performed correctly.

Setting position_min to a negative number will workaround this error, but it won't prevent the nozzle from ramming into the bed at a different point, where the bed level is higher.

KevinOConnor, 2021-10-18:
We only use github to share the results of work done to improve Klipper.  If you're looking for help - even help diagnosing a bug - you'll need to reach out to the Klipper community.  It looks like this issue is a dup of your #4808 - please refrain from opening additional tickets.

-Kevin

Black6spdZ, 2021-11-20:
I am also having this exact issue with a less than perfect bed. Never had this issue with Marlin so I agree with p-v-n, the way Klipper handles this is NOT a feature, it is a bug.