Klipper3d/klipper #4861: Communication timeout during homing probe (multi-mcu probing)

simonkuehling, 2021-11-02:
I am running multi-mcu probing in a setup with a couple of separate SAMD21 mcu boards - and occasionally I get a "Communication timeout during homing probe" error during a probe run. USB communication is otherwise super stable during printing and all operations around that, and since mutli-mcu homing/probing is relatively new, I thought this might be worth taking a look at to decide if its a hardware or a software issue.

The probe-mcu currently is a completely separate SAMD21 board with just this one probe GPIO in use - no other tasks run on here. 

A fresh klippy.log is attached including a M112 shutdown for debugging output.
[klippy.log](https://github.com/Klipper3d/klipper/files/7459354/klippy.2.log)

simonkuehling, 2021-11-04:
Digging through the code a bit more, I am having trouble exactly understanding why the timeout is happening. Since I do not see any retransmits in the M112 output (correct me if I'm wrong!), I would assume the serial communication itself is fine... Also the bandwith and MCU loads are super low.

Is there any case where the overall system load (on the Raspi CPU) could mess the synchronisation up so that the data sent by the probe-mcu isn't read in time for the klippy-homing-process to recognize it?

simonkuehling, 2021-11-04:
Ok, I guess I figured it out by trial and error finally.

For reference, something to look out for: Besides the 5 Klipper MCUs there was a touchscreen and a wireless keyboard dongle attached to a USB3.0 port on a Rapsberry Pi 4. Only when I moved all peripherals (touchscreen, keyboard dongle) to the USB2.0 ports and reserved the USB3.0 only for all of the Klipper MCUs, communication timeouts are gone.
Seems like the USB3.0 and USB2.0 sockets are sufficiently isolated in the Pi's USB chip so that all of the serial communication can flow smoothly and uninterrupted now.

The important observation was that touchscreen and keyboard input was showing a little lag occasionally during a PROBE_ACCURACY run with 100 samples and 10mm retract distance (=a lot of communication for trigger checks).

Closing this issue for now...

DroneMang, 2021-12-28:
I was getting this error during a bed mesh calibration.  I figured out that I needed to make the points farther apart (increase time between probes) and the error went away.  This is on a HermitCrab toolchanger setup with separate mcu at toolhead.  So it seems that after a probe, klipper needs some time to sync up the timing or do some calculations between probing.  What is strange is I can do a probe calibrate command with 50 probes and do timeout, but if lots of probes during a bed mesh then timeout.  I will try with pps set to 0,0 so no interpolation, then maybe there will be no error?

simonkuehling, 2021-12-29:
Interesting.. Can you show your [probe] config section to better understand how fast you were probing there?
Also, what host computer do you use - is it a Raspberry Pi as well? And if that, which version?

mrwilby, 2022-01-12:
I have a Pi 3b and a multi-MCU setup and I'm also seeing this (toolboard at the toolhead). I see this during Voron 2 quad gantry leveling sometimes, but it seems to require some specific action:

- I need to be heating the bed and/or extruder heater (not sure which)
- I need to have HDMI output enabled (and various HDMI-related settings in my /boot/config.txt enabled)

If I remove the HDMI settings from /boot/config.txt then I have so far not seen the issue in some hours of testing. Also, this problem all started to arise after I added an HDMI touchscreen to my printer. Before that, things were very stable.

I'm using CAN and a Waveshare CAN device to communicate between Pi & the toolboard. 

It doesn't matter whether the HDMI device (or USB touch screen) are plugged in or not. The problem still occurs (at random) when the HDMI & USB touch device are unplugged, providing the /boot/config.txt options are enabled).

Very puzzling, but I was wondering if my 3b is too underpowered (CPU-wise) given all the USB ports are in use on my machine (I also have a USB camera and am using USB to communicate to the main mcu, an STM32F446).

Gilabite, 2022-01-12:
@mrwilby have you tried using a powered USB hub? That would take the load off of the PI and put it on the hub.

mrwilby, 2022-01-12:
Thank you @Gilabite for the thoughts... I haven't yet, but I am thinking of making up a cable to power the waveshare AMOLED (5.5") display from a 5v power supply I have inside the printer. 

That said, as I mentioned, even if the HDMI screen is _not_ plugged in (but various HDMI settings *are* defined in /boot/config.txt) I am able to reproduce the issue (sometimes, it can take a bit of trying as it's not 100% deterministic). Again, if I remove those HDMI settings, it appears as though it "solves" the issue. Very strange.

mrwilby, 2022-01-12:
I switch to a RPi 4 but that didn't help - same exact behavior, stable until I added the HDMI-related settings to config.txt (did not plug in the display in any way).

I increased the SPI frequency of my CANBus adaptor and that now seems to have helped - need to test further to see if it's actually a 'fix' or not, but so far, seems promising.

simonkuehling, 2022-01-13:
I‘ll reopen the issue for the moment, looks like there is more to it…

I have read that at least the Raspi 4 seems to have interference issues coming from the HDMI port (EMI crosstalk) on high screen resolutions (1920px and up) which can lead to unstable wifi on channel 1 (2.4ghz band).
My screen does run at 1920x1080, so there‘s that. Maybe not only wifi but also USB could be affected at specific throughputs or something?

@mrwilby, since you are able to reliably reproduce the problem at the moment - can you try if reducing your screen resolution help?

jakep82, 2022-01-13:
I'm not seeing how this is a Klipper issue. If disabling HDMI solves it, shouldn't this be raised as an issue with the OS?

simonkuehling, 2022-01-13:
> I'm not seeing how this is a Klipper issue. If disabling HDMI solves it, shouldn't this be raised as an issue with the OS?

I agree that this could potentially be a hardware or OS issue - but since raspberry pis are probably the most common used platform that Klipper is run on, it might be worth collecting a few more hints or work-arounds to possibly add to the documentation or FAQ if confirmed... If it is a board design issue, we're stuck dealing with it until a Pi 5 is released anyway (with hopefully improved design in this regard).

The HDMI/Wifi RF interference issue was reported back in 2019 I think - although only for the Raspi 4. It was concluded that HDMI-adapters and low quality HDMI cables (bad shielding) were at least part of the problem. But as @mrwilby reported this with a 3B as well, it does not quite match, but some comments on a [hackaday article](https://hackaday.com/2019/11/28/raspberry-pi-4-hdmi-is-jamming-its-own-wifi/) from back then tell similar issues for a 3B, too.

mrwilby, 2022-01-13:
I also do not know if this is really a klipper issue. It might be some quirk of my hardware setting. FYI, I am using:

- Fysetc spider v1.0 (STM32F445 IIRC)
- Huvud-like CAN bus toolboard
- Waveshare RS485 CAN HAT
- Waveshare 5.5" AMOLED HDMI display - this uses 2 x USB connectors (power, touch) and 1 x HDMI connector (display)

The probe issue started to manifest once I enabled the required Waveshare HDMI settings (as said, I don't even need to plug in any USB or HDMI).

With this /boot/config.txt configuration I am able to (periodically, not 100% every time) reproduce the issue:

```
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
# enable raspicam
start_x=1
gpu_mem=128

#MainsailOS Specific
#Enables UART Communication by default
enable_uart=1
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=12000000,interrupt=25,spimaxfrequency=1000000,disable-bt

max_framebuffer_height=1920
max_usb_current=1
config_hdmi_boost=10
hdmi_group=2
hdmi_force_hotplug=1
hdmi_mode=87
hdmi_timings=1080 1 26 4 50 1920 1 8 2 6 0 0 0 60 0 135580000 3
```

If I either:

a) Comment out all the HDMI-related settings at the end of the file OR
b) Change the dtoverlay entry to have a higher frequency for SPI (spimaxfrequency=5000000) then 

... I cannot (so far) reproduce the problem anymore.

mrwilby, 2022-01-13:
@simonkuehling I'm not exactly sure how to reconfigure the HDMI settings to lower the display resolution. I tried editing the "max_framebuffer_height" setting from 1920 to 800, but that didn't help. I also tried commenting out just the hdmi_timings line, but then the pi didn't boot anymore.

simonkuehling, 2022-01-14:
> @simonkuehling I'm not exactly sure how to reconfigure the HDMI settings to lower the display resolution. I tried editing the "max_framebuffer_height" setting from 1920 to 800, but that didn't help. I also tried commenting out just the hdmi_timings line, but then the pi didn't boot anymore.
> 
> With the framebuffer set to 800, the problem was still reproducible (with spimaxfrequency=1000000)

Re-reading your full summary and finally realizing that you are not enabling/disabling HDMI entirely, but rather adding/removing those touchscreen-specific configurations (my fault, you already said exactly that earlier as well), I would revise my comment on RF interference issue - that would not make sense as a cause here... 
For that reason, there is no need to try changing the resolution anymore, i think. Also I am not sure which resolution options that touchscreen would accept and if it has internal scaling at all.

But since you do seem to have a workaround in changing the SPI frequency on hand - shall we close the issue with that?

mrwilby, 2022-01-23:
Yes, I think we can close this. I think it's more likely that in my case, it's a HW/SW issue external to klipper.

simonkuehling, 2022-01-23:
Ok, thanks for the feedback - closing this now 👍

closed 2022-01-23 by simonkuehling