Klipper3d/klipper #4864: There was a problem updating!

pangxie0, 2021-11-03:
Option 'default_parameter_e' is not valid in section 'gcode_macro pause'
[image]

feadot, 2021-11-03:
https://github.com/Klipper3d/klipper/blob/master/docs/Config_Changes.md

`20211102: Several deprecated features have been removed. The stepper step_distance option has been removed (deprecated on 20201222). The rpi_temperature sensor alias has been removed (deprecated on 20210219). The mcu pin_map option has been removed (deprecated on 20210325). The gcode_macro default_parameter_<name> and macro access to command parameters other than via the params pseudo-variable has been removed (deprecated on 20210503). The heater pid_integral_max option has been removed (deprecated on 20210612). `

really what could have gone wrong...

sleepingAwake, 2021-11-03:
Could someone explain to me like a child how I am supposed to modify this code so it will work with the new syntax.
I have several macros calling default_parameters_*
Heres one. I've taken a look at the Klipper config examples but cant figure it out. What variable name do I use now to replace default_parameter??? Params doesnt work.

[gcode_macro M600]
default_parameter_X: 0
default_parameter_Y: 0
default_parameter_Z: 50
gcode:
    SAVE_GCODE_STATE NAME=M600_state
    PAUSE
    G91 ;relative
    G1 E-.2 F2700
    G1 Z{Z} ;raise nozzle
    G90 ;absolute
    G1 X{X} Y{Y} F3000
    G91 ;relative
    M300 P1000 ;beep
    FILAMENT_UNLOAD
    M300 P3000 ;beep
    G90 ;absolute
    RESTORE_GCODE_STATE NAME=M600_state
    
    What a great way to wake up. Probably hundreds of people woke up this morning to a non-working printer.
    While the change was announced months back I havent seen an updated sample config on git. Cmon guys.

pandel, 2021-11-03:
Cmon ;-) This is not that hard... did you try this?

```
[gcode_macro M600]
gcode:
  {% set X = params.X|default(0)|float %}
  {% set Y = params.Y|default(0)|float %}
  {% set Z = params.Z|default(50)|float %}

  SAVE_GCODE_STATE NAME=M600_state
  PAUSE
  G91 ;relative
  G1 E-.2 F2700
  G1 Z{Z} ;raise nozzle
  G90 ;absolute
  G1 X{X} Y{Y} F3000
  G91 ;relative
  M300 P1000 ;beep
  FILAMENT_UNLOAD
  M300 P3000 ;beep
  G90 ;absolute
  RESTORE_GCODE_STATE NAME=M600_state
```

sleepingAwake, 2021-11-03:
> Cmon ;-) This is not that hard... did you try this?
>
> ```
> [gcode_macro M600]
> gcode:
>   {% set X = params.X|default(0)|float %}
>   {% set Y = params.Y|default(0)|float %}
>   {% set Z = [...]

Cmon this is not that hard says the guy with decades of coding experience.
Rebuilding a Subaru engine from the bearings up isnt that hard for me. So it must be pretty easy for you too right?

Anyway thank you for the help. I will try that.

pandel, 2021-11-03:
@sleepingAwake This could be very helpful in the future maybe: https://klipper.discourse.group/t/macro-creation-tutorial/30

sleepingAwake, 2021-11-03:
> @sleepingAwake Sorry that I have to say this, but everything is in the manual: https://www.klipper3d.org/Command_Templates.html#macro-parameters This could be very helpful in the future maybe too: https://klipper.discourse.group/t/macro-creation-tutorial/30

I looked at the manual before I posted and tried a couple things before asking for help.
But something like the discourse link you posted is far more useful and where I need to start because as I hinted at before not everyone has the same level of programming knowledge.

stephbellemare, 2021-11-04:
Strangely enough, i'm getting the same error after updating my macros to remove the depreciated commands
Error: Option '{% set bed_temp' is not valid in section 'gcode_macro start_print'

{% set BED_TEMP = params.BED_TEMP|default(40)|float %}
{% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(200)|float %}

This happened after updating as well.
Maybe i'm missing the obvious?

feadot, 2021-11-04:
> {% set BED_TEMP = params.BED_TEMP|default(40)|float %} {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(200)|float %}

    {% set bed_temp = params.BED_TEMP|default(60) %}
    {% set extruder_temp = params.EXTRUDER_TEMP|default(200) %}

pangxie0, 2021-11-04:
Thank you. I'll just delete this code!

stephbellemare, 2021-11-04:
> {% set bed_temp = params.BED_TEMP|default(60) %}
> {% set extruder_temp = params.EXTRUDER_TEMP|default(200) %}

Still a no go for me.
                   Option '{% set bed_temp' is not valid in section 'gcode_macro start_print'

i'm really scratching my head now....

simonkuehling, 2021-11-04:
@stephbellemare, could you post your full config, or at least the full `[gcode_macro start_print]`
Maybe the code is not indented correctly?

stephbellemare, 2021-11-04:
> @stephbellemare, could you post your full config, or at least the full `[gcode_macro start_print]` Maybe the code is not indented correctly?

@simonkuehling here goes, thanks for looking into it.

[gcode_macro START_PRINT]
{% set bed_temp = params.BED_TEMP|default(60) %}
{% set extruder_temp = params.EXTRUDER_TEMP|default(200) %}
gcode:
    # Load bed mesh --- comment if loading default
    #BED_MESH_PROFILE LOAD={BED_TEMP}
    # Set and wait for bed to reach temperature
    M190 S{bed_temp}
    # Pre-heat extruder to 160
    M104 S160
    # Set GCODE offset
    SET_GCODE_OFFSET Z=0
    # Home
    G28
    # Move Z up
    G1 Z10.0 F3000
    # Reset extruder
    G92 E0
    # Set extruder temperature
    M104 S{bed_temp}
    # Move Z axis up a little to prevent scratching heat bed
    G1 Z2.0 F3000
    # Move to start position
    G1 X0.1 Y20 Z2.0 F5000.0
    # Wait for Extruder temperature
    M109 S{EXTRUDER_TEMP}
    # Move to start for purge line
    G1 X3.1 Y20 Z0.3 F5000.0
    # Draw 1st line
    G1 X3.1 Y150.0 Z0.3 F1500.0 E15
    # Move to side a little
    G1 X3.4 Y150.0 Z0.3 F5000.0
    # Draw 2nd line
    G1 X3.4 Y20 Z0.3 F1500.0 E30
    # Reset extruder
    G92 E0
    # Move Z up a little
    G1 Z5.0 F3000

Also have the same issue with my M600 macro when i comment out the start print one
[gcode_macro M600]
{% set X = params.X|default(50) %}
{% set Y = params.Y|default(220) %}
{% set Z = params.Z|default(10) %}
gcode:
   SAVE_GCODE_STATE NAME=M600_state
   PAUSE
   G91
   G1 E-.8 F2700
   G1 Z{Z}
   G90
   G1 X{X} Y{Y} F3000
   G91
   G1 E-90 F1000
   RESTORE_GCODE_STATE NAME=M600_state

pandel, 2021-11-04:
@stephbellemare 

It has to be
```
[gcode_macro START_PRINT]
gcode:
  {% set bed_temp = params.BED_TEMP|default(60) %}
  {% set extruder_temp = params.EXTRUDER_TEMP|default(200) %}
  # Load bed mesh --- comment if loading default# Load bed mesh --- comment if loading default
```

and **not** (you see the difference?)

```
[gcode_macro START_PRINT]
{% set bed_temp = params.BED_TEMP|default(60) %}
{% set extruder_temp = params.EXTRUDER_TEMP|default(200) %}
gcode:
  # Load bed mesh --- comment if loading default
```

stephbellemare, 2021-11-04:
> @stephbellemare
>
> It has to be
>
> ```
> [gcode_macro START_PRINT]
> gcode:
>   {% set bed_temp = params.BED_TEMP|default(60) %}
>   {% set extruder_temp = params.EXTRUDER_TEMP|default(200) %}
>   # Load bed mesh [...]

@pandel god do i feel dumb now. :-/
Thanks for taking the time.   Problem solved :)

martinux, 2021-12-04:
@pandel With respect, it is unnecessarily hard. I've been using Marlin for a few years now and I tried to install klipper from the official documentation two days ago. Are you aware that the official installation documentation (https://www.klipper3d.org/Installation.html) makes no mention of running the install script? I had to figure that out myself.

 I then had to go digging through issue reports here to find out that one can have a permissions error with /tmp/printer. I had to figure that out myself.

 I then went looking for a sample config for a printer identical to my own and ran into the issue described by @pangxie0 because there is no meaningful error message. There should have been an explicit error message that described the feature as being depreciated.  

Not what I'd consider a straightforward process.

simonkuehling, 2021-12-04:
@martinux may I ask which install script you did not see mentioned in the linked installation manual? If there is room for improvement, I‘m sure there are plenty of open ears to pick up on that and forge a pull request out of it. This is a community driven project that is always just as good as all the contributions have made it yet!

jakep82, 2021-12-04:
> @pandel With respect, it is unnecessarily hard. I've been using Marlin for a few years now and I tried to install klipper from the official documentation two days ago. Are you aware that the [...]

Running the install script is the second line in the first code block. Perhaps it should be separated into a separate code block to make it more obvious, but it is in the instructions.

martinux, 2021-12-04:
@simonkuehling @jakep82 Sorry, I should have proof-read that. What I meant to say was "makes no mention of an error when running the install script?"

```

./install-octopi.sh 

###### Running apt-get update...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Reading package lists... Done        
E: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'oldstable'
```

I had to figure how to fix it (sudo apt-get update --allow-releaseinfo-change).

Apologies again for the confusion.

Sineos, 2021-12-30:
Issue seems concluded. If you need support, we will be happy to assist on https://klipper.discourse.group

closed 2021-12-30 by Sineos