Klipper3d/klipper #4866: Mcu shutdown timer too close

benjjyman, 2021-11-03:
Have got log at moment, will later.
Just updated to today's version, 3 layers into a 10hour print received message :

Mcu shut down: timer too close.

Will post log later.

Thanks

samwells2000, 2021-11-03:
having the same issue here's my log/cf
[klippy.log](https://github.com/Klipper3d/klipper/files/7469081/klippy.log)
[printer.txt](https://github.com/Klipper3d/klipper/files/7469102/printer.txt)

this happens anytime I try to home the printer which moves a stepper, everything else works fine

Deneteus, 2021-11-03:
Related? #4679

benjjyman, 2021-11-03:
@Deneteus  Thanks for the lead, it may well be something to do with usb & ftdi , may be even ADXL345 as Ive just rewired the accelerometer, and attached more than one webcam (which keeps changing /dev/video0 to /dev/video3 which i havent beed able to fault find ) the problem for me atleast is  very intermittent as I have only had this happen twice in over 100 hours of printing

Eifel-Joe, 2021-11-06:
I got same problem on my Voron 2.4 with Octopus Board. After a long time and much coffee i came to the conclusion, that maybe i made a mistake with compiling the new firmware for the Octopus. For the Octopus you have to set the clock frequency. So if i made a mistake on that part, it could be related to the timer error.
So i recompiled it with the correct settings in make menuconfig and reflashed the board. And that's it. Printer is up and running again.

plloppii, 2021-11-07:
I'm seeing this issue as well.
Attaching my klippy.log that is relevant. I am seeing the issue on one of my machines and not on the other. 
I have my serial set to 115200, and I changed out the usb cable once. Any other tips in resolving this mcu timeout issue? 
[klippy.log](https://github.com/Klipper3d/klipper/files/7491794/klippy.log)

dinosauria123, 2021-11-08:
I have met the same issue. But it was fixed.
I found rotation_distance in stepper_x, _y, _z was set as less than 1.
I am confused with step_distance as rotation_distance. (step_distance = 1/rotation_distance)
Please check your value of rotation_distance is less than 1 or not.

benjjyman, 2021-11-08:
I personally found having to much stress on USB ports ( multiple webcam.txt, some of which were not plugged in plus, adxl345), when I cleared these the problem resolved. Running on armbian buster on orangepi 4 .

widewing, 2021-11-19:
I have this issue as well, I run klipper in docker in my x86 server (Lenovo T420, precisely). It randomly stop when doing print. What can be the reason for this particular failure? I checked server load, memory uses, swap uses, they all look fine. I have other applications running in this server though, but they are not causing a lot usage. Attached [klipper.log](https://github.com/Klipper3d/klipper/files/7572578/klipper.log)

benjjyman, 2021-11-20:
Widewing most people find it's due to USB connection errors, either faulty chip or config.
You might have better luck in the discord server though, as many people have faced similar issues with recent updates.

widewing, 2021-11-20:
@benjjyman thanks for the hint! I do use USB hub for connection. Will try direct connect.

plloppii, 2021-11-22:
> @benjjyman thanks for the hint! I do use USB hub for connection. Will try direct connect.

I've found improved performance by wiring a direct connection from the raspberry pi to the host MCU with UART.

benjjyman, 2021-11-22:
> > @benjjyman thanks for the hint! I do use USB hub for connection. Will try direct connect.
> 
> I've found improved performance by wiring a direct connection from the raspberry pi to the host MCU with UART.

Cool just got my octopus board today will try uart see if that eliminates problem, thanks

Sineos, 2021-12-30:
If this is still an issue please open a new post on https://klipper.discourse.group and we will be happy to assist

closed 2021-12-30 by Sineos